modelica/ModelicaStandardLibrary #107: Improve Modelica Association PR by a quarterly newsletter?

modelica-trac-importer, 2017-01-13:
**Reported by dietmarw on 6 Oct 2008 10:25 UTC**
<sub>I created this issue from the mailing list topic (same subject [link to Google Groups](http://is.gd/3Aqh)) to keep better track:</sub>

Martin's suggestion:
  I think this is a good idea. In order that this looks professional, it would be probably good to send the mail both in html (default) and in pure text form. The html form should have a good layout and should include the Modelica logo. We could have a deadline, where everyone can send proposals for the newsletter. We need probably a small process how this is performed, e.g.:
 * Deadline for newsletter (the info is sent to Modelica-design, so everyone is informed what is going on).
 * You collect the information and build a first version
 * The board reviews it and makes improvement suggestions
 * The newsletter is sent out.

 We can have a short discussion on this topic at the next Modelica design meeting.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/107

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 13 Dec 2008 19:39 UTC**
_From the discussion at the design meeting (taken from the minutes of the 59th Design Meeting):_

  Discussion, how to improve the operation of the Modelica Association (quarterly newsletter, ...) Quarterly Newsletter – presentation by Dietmar Dietmar: Propose to send one page about infos after every design meeting (so newsletter coupled  with design meetings). Hilding: Send pdf-file as newsletter (2-columns with pictures). 
  Kilian: Does not like pdf-file at all for newsletters 
  Martin: Have the newsletter (with images on the Web site) and have an email with just the topics and have a link to the web page.

  Poll:
   * Have only email in textual format: 4
   * Have only email in html format: 7
   * Have email with pdf file: 1
   * Have it on the web and an email with just headings and link to the web page: 17
   * Have it on the web and an email with just headings and link to a pdf-file on the web: 7
   * No newsletter: 1

  Francesco: Need professional for writing the newsletter.  
  Who is the editor. Hilding proposed that the secretary is the editor of the newsletter (Michael Tiller). 
  Send proposals (html-text with pictures) to Michael until Dec. 3 for the first newsletter. Editor asks people to provide articles. 
  Hilding: A column from the chairman. We make a first version until mid of December and send the link for discussion to the group.  
  Final decisions about content and sending is made by the board. 
  Newly: Francesco, Fillipo
  In favor: 17 Against: 0 Abstain: 2 The proposal is accepted.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 15 Jan 2009 10:13 UTC**
Hiding this ticket for a short period of time to test if the spammers would let go of it if it's not accessible anymore (currently about 20 blocked attacks per hour on only this ticket).

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 20 Dec 2016 14:01 UTC**
Unhide for GitHub transition.

beutlich, 2017-02-02:
@mtiller There was 1 newletter in 2015 and none in 2016. Is the newletter discontinued?

xogeny, 2017-02-02:
@beutlich This issue has been discussed at length in the board.  I do not have the time to work on the newsletter anymore.  I am just too busy with MA stuff, NA-MUG and business stuff.

I would say the board position on this issue is "accepting PRs" (slight pun intended).  If somebody wants to volunteer to do it, I think we would welcome it.  I have to say that the current newsletter tool we developed actually works quite well for collaborating on a newsletter.  However, the software infrastructure is pretty old, so I'm not sure how long it will last. :-(

P.S. - On GitHub, I'm always @xogeny (not @mtiller).  The @mtiller identity is only associated with my commits.  I know it is confusing.  Feel free to use @mtiller (I get notified), but don't be surprised when the response comes from @xogeny.

beutlich, 2017-02-03:
@xogeny Thanks for your honest reply. I feared to read your confirmation that the Modelica newsletter slowly died. Did you/the board consider to start a public call asking for volunteers?

beutlich, 2018-10-28:
Thanks to @filip-jezek https://newsletter.modelica.org was revived again in 2018.

closed 2018-10-28 by beutlich