modelica/ModelicaStandardLibrary #115: Separation of model code and parameter data

modelica-trac-importer, 2017-01-13:
**Reported by AHaumer on 4 Nov 2008 11:32 UTC**
In many cases it is desired to separate parameter data from model code, to be able to apply a ready-developed application model on different parameter sets defined by different projects.
If the parameters are all non-structural, i.e.not changing the count of unknowns and equations, the application model could be applied without re-translation.
Regarding the parameters, there are two types of parameters:
 * property parameters of componets (e.g. the resistance of a resistor)
 * boundary conditions resp. simulation parameters (e.g. the amplitude of the driving voltage source)
Reading both types of parameters from file would be convenient; currently there are 3 options to experiment with:
 * Modelica.Utilities (keyword-oriented)
 * external C-functions
 * XML-library (c) arsenal rsearch
We should test the different options, defining a convenient and tool-independet way to separata data from code.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/115

modelica-trac-importer, 2017-01-13:
**Comment by mtiller on 4 Nov 2008 15:32 UTC**
A few comments.  First, I want to point out that the three options mentioned are "pull" options.  What I mean by that is that the Modelica code needs to be such that all of the parameters in question explicitly "pull" their values from another source.  The important point here is that the models need to be instrumented in advance to pull this information.

The alternative approach, which is realized to some extent in using modifications, is to push changes on a class.  For example, when we extend from one class and then make a whole series of modifications in the extend clause we are pushing modifications on the original model.

The problem I see with modifications as they stand now is that they are not programmatic.  I can't write algorithmic code to decide how to configure something.  Peter and Sebastien (I think) have talked in the past about the ability to write code the actually manipulates models in such ways.  This might address the main issue I see with "pull" approaches which is that you have to pre-instrument all these models to pull their values.  You could "push" modifications onto the model that then "pull" the values from sources (e.g. model A extends B(x=getFromFile("x"))) but it might be more scalable to implement something that could walk the model, identify candidate parameters and apply such modifications in a more programmatic way.

This doesn't really address what Anton is proposing but I just wanted to add some comments about the "big picture".

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 20 Feb 2009 15:50 UTC**
closing milestone:Design60 and postponing this discussion to the next post 3.1 Design Meeting

modelica-trac-importer, 2017-01-13:
**Comment by hansolsson on 30 Mar 2016 14:03 UTC**
Correcting "Component" - the different parameterizations is primarily a library issue and version is already "Standard Library 3.0".

modelica-trac-importer, 2017-01-13:
**Changelog removed by hansolsson on 30 Mar 2016 14:03 UTC**

modelica-trac-importer, 2017-01-13:
**Modified by ahaumer on 20 Dec 2016 16:47 UTC**

beutlich, 2017-01-16:
@AHaumer There is a free library [ExternData](https://github.com/tbeu/ExternData) available that provides paramerization from external parameter sources.

beutlich, 2017-11-21:
Closing as worksforme after 9 years.

closed 2017-11-21 by beutlich