modelica/ModelicaStandardLibrary #143: 3-phase transformer transforms DC-signals

modelica-trac-importer, 2017-01-13:
**Reported by AHaumer on 23 Jan 2009 13:51 UTC**
Modelica.Electrical.Machines.BasicMachines.Transformers transform also DC-signals; no need for testing: the equations are equivalent to Modelica.Electrical.Analog.Basic.Transformer (see Ticket #142).
I do not know a solution for an ideal transformer (without magnetization current) to avoid this. At least a warning should be noted in the documentation.
However, I plan to redesign the technical transformers Modelica.Electrical.Machines.BasicMachines.Transformers, including a good estimation of the magnetizing current. To do this, parameterization of these models has to be changed - unfortunately not backward-compatible, and I'm unsure how to handle this in a conversion script.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/143

modelica-trac-importer, 2017-01-13:
**Modified by ahaumer on 2 Mar 2012 09:04 UTC**

modelica-trac-importer, 2017-01-13:
**Comment by otter on 30 Mar 2013 19:27 UTC**
If you do not know how to automatically convert, introduce a new model and keep the old one.

modelica-trac-importer, 2017-01-13:
**Changelog removed by otter on 30 Mar 2013 19:27 UTC**

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 18 Jul 2013 12:14 UTC**
Why exactly is this "wontfix" when the library officer put this in as an future feature he plans to integrate. I would expect the library officer to decide on this not anybody else really.

modelica-trac-importer, 2017-01-13:
**Comment by otter on 18 Jul 2013 12:48 UTC**
Replying to [comment:4 dietmarw]:
> Why exactly is this "wontfix" when the library officer put this in as an future feature he plans to integrate. I would expect the library officer to decide on this not anybody else really.

The question was: 

_To do this, parameterization of these models has to be changed - unfortunately not backward-compatible, and I'm unsure how to handle this in a conversion script._

The answer was:

_If you do not know how to automatically convert, introduce a new model and keep the old one._

So, the question was answered.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 19 Jul 2013 06:54 UTC**
Sorry I don't agree. Your answer is for some part of the ticket but the ticket is about the introduction of a "3-phase transformer transforms DC-signals" which still is planned and this ticket really is about this enhancement which still might happen in combination with the restructuring of the transformer models. So a "wontfix" just does not apply at all here unless the reporter and library officer decides willingly to **not** create such a model.

modelica-trac-importer, 2017-01-13:
**Modified by ahaumer on 20 Dec 2016 16:39 UTC**

beutlich, 2019-07-15:
@AHaumer More than 10 years later: Is is still relevant? If yes, do you plan to add new models in Modelica.Electrical.Machines.BasicMachines.Transformers or convert the existing ones?

AHaumer, 2019-07-15:
Yes please keep this ticket opne, it is still relevant. 
To summarize the problem:
Best solution would be to add magnetizing current (i.e. a main inductance) to the models. 
Drawback: Transformers have a very small manetizing current, i.e. a very high main inductance. 
It takes a rather long time until initial transients vanish - this is the case in real life. 
In simulations this is in many cases not desired. 
I don't have a quick solution a hand, but it's still on the to do - list.

christiankral, 2019-07-16:
Yes, I agree, we shall keep this ticket open.

beutlich, 2019-10-05:
> If yes, do you plan to add new models in Modelica.Electrical.Machines.BasicMachines.Transformers or convert the existing ones?

Is this feasible for MSL v4.0.0?