modelica/ModelicaStandardLibrary #194: Change greek parameter name "mue" in MSL

modelica-trac-importer, 2017-01-13:
**Reported by christiankral on 23 Jul 2009 15:07 UTC**
In the MSL the following packages are using a parameter called `mue`:

 * `Modelica.Mechanics.MultiBody`
 * `Modelica.Mechanics.Rotational` 
 * `Modelica.Mechanics.Translational`
 * `Modelica.Constants`

The intention was to represent the Greek letter for m (µ), but it is unfortunately spelled incorrectly. The correct spelling is "mu", not "mue". Therefore, these parameter names should be changed once the next conversion script is scheduled.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/194

modelica-trac-importer, 2017-01-13:
**Comment by thorade on 14 Aug 2009 07:42 UTC**
In Modelica.Thermal.FluidHeatFlow.Media there is in all four media a variable "lamda" which I think should be "lambda"

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 21 May 2010 11:10 UTC**
Milestone MSL3.3 deleted

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2017-10-18:
> Comment by thorade on 14 Aug 2009 07:42 UTC
In Modelica.Thermal.FluidHeatFlow.Media there is in all four media a variable "lamda" which I think should be "lambda"

@thorade This is covered by #2170.

beutlich, 2019-04-04:
Also nue should be renamed to nu then. Will also update #2170 for the conversion.

PR #2864 (2019-04-04): Rename mue to mu
close #194
commit 451b1dc6 (2019-04-04): Change 'mue' to 'mu' in Mechanics

close #194

# Conflicts:
#	Modelica/Resources/Scripts/Conversion/ConvertModelica_from_3.2.3_to_4.0.0.mos
PR #2985 (2019-06-21): Rename nue to nu
Close #194.
closed 2019-06-21 by beutlich