modelica/ModelicaStandardLibrary #199: move partial transformer to interfaces

modelica-trac-importer, 2017-01-13:
**Reported by AHaumer on 26 Aug 2009 06:12 UTC**
Whenever a conversion script is allowed, move partial BaseTransformer from Modelica.Electrical.Machines.Components to Modelica.Electrical.Machines.Interfaces.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/199

modelica-trac-importer, 2017-01-13:
**Modified by AHaumer on 26 Mar 2010 07:51 UTC**

modelica-trac-importer, 2017-01-13:
**Modified by ahaumer on 2 Mar 2012 09:04 UTC**

modelica-trac-importer, 2017-01-13:
**Modified by ahaumer on 20 Dec 2016 16:40 UTC**

modelica-trac-importer, 2017-01-13:
**Changelog removed by ahaumer on 20 Dec 2016 16:40 UTC**

AHaumer, 2019-07-15:
Seems to be already done - I can't remember who, I can't remember when. 
If nobody contraditcs, we should close this ticket.

beutlich, 2019-07-15:
Resolved by #2929.

closed 2019-07-15 by beutlich