modelica/ModelicaStandardLibrary #205: Bug in Modelica.Fluid.Examples.IncompressibleFluidNetwork

modelica-trac-importer, 2017-01-13:
**Reported by dietmarw on 9 Sep 2009 16:27 UTC**
When runnung the fluid example `Modelica.Fluid.Examples.IncompressibleFluidNetwork` the simulation will fail if one simple leaves the "valve1" closed.

When one sets valveOpening1(height=-1) the simulation will terminate
with an error message.

The tool shows:
  ```
... Error message from dymosim
At time T = 3.000001e+00 and stepsize
H = 1.164153e-13 the corrector could not
converge because IRES was equal to minus one.
Integration will be terminated.
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/205

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 2 Aug 2012 13:08 UTC**
Just tested this again and for me it works. So it looks like it was perhaps a tool issue.

modelica-trac-importer, 2017-01-13:
**Modified by dietmarw on 2 Aug 2012 13:33 UTC**

closed 2017-01-13 by modelica-trac-importer

Abantika-2409, 2022-02-09:
I am running a simulation on a building (two classrooms) with a VAV system with Indirect evaporative cooling. and the building also has night cooling with controllable windows. I get the same error during the simulation. I  run the simulation from (1st June to 31st August). When I run the simulation from day 58 (July 28th ) to day 68 (August 7th) the simulation runs. However from Day 0 if I start the simulation, I get this same error on day 13. How did you solve this error in your model?