modelica/ModelicaStandardLibrary #237: Mapping Modelica types to the target machine

modelica-trac-importer, 2017-01-13:
**Reported by dietmarw on 19 Nov 2009 12:24 UTC**
There wasn't a ticket on this yet...
## From [milestone:Design63]
Mapping Modelica types to the target machine
(vendor specific modifiers on annotations)
Details see [source:ModelicaDesign/trunk/MeetingMinutesMaterial/min63/Slides-and-Documents/Modelica_Proposal_EmbeddedSystems2.doc Modelica_Proposal_EmbeddedSystems2.doc]

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/237

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 15 Aug 2012 08:14 UTC**
Left-over tickets from an older Design meeting. Rescheduling to next design meeting to be reviewed.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 17 Oct 2012 16:23 UTC**
Since not handled at the last meeting rescheduling to the next one.

modelica-trac-importer, 2017-01-13:
**Comment by hansolsson on 5 Dec 2012 15:18 UTC**
No embedded group at this meeting.

modelica-trac-importer, 2017-01-13:
**Modified by otter on 20 Sep 2013 10:22 UTC**

modelica-trac-importer, 2017-01-13:
**Comment by hansolsson on 23 Oct 2013 13:38 UTC**
Shouldn't we close these as wontfix?
The entire embedding chapter has been removed; so we need to restart this in another way I guess.

modelica-trac-importer, 2017-01-13:
**Comment by hansolsson on 23 Oct 2013 13:46 UTC**
[Missed one previously.]
Shouldn't we close these as wontfix?
The entire embedding chapter has been removed; so we need to restart this in another way I guess.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 15 Nov 2013 13:30 UTC**
80th Design Meeting is long over ... closing that milestone and still open tickets set back to \*undecided\*.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 20 Dec 2016 14:01 UTC**
Unhide for GitHub transition.

HansOlsson, 2017-07-10:
Closing this. It was almost 4 years since I suggested to close it as wontfix, since it was an improvement for the embedding-chapter of Modelica which has been removed from the Modelica specification - and there has been no additional activity.

It might be that we need something similar not connected to embedding-chapter, possibly related to FMI; but I think that should start fresh.

closed 2017-07-10 by HansOlsson