modelica/ModelicaStandardLibrary #238: Defining the Download of File based Data to Targets without File System

modelica-trac-importer, 2017-01-13:
**Reported by dietmarw on 19 Nov 2009 12:25 UTC**
There wasn't a ticket on this yet...
## From [milestone:Design63]
Defining the Download of File based Data to Targets without File System 
(preEvaluateForTargetWithoutFileSystem annotation).

Michael Tiller: Introduce a new type “URI-Type” as subtype of String or an annotation. The system can then identify the URI and knows that this resource must be used.
Details see [source:ModelicaDesign/trunk/MeetingMinutesMaterial/min63/Slides-and-Documents/Modelica_Proposal_EmbeddedSystems2.doc Modelica_Proposal_EmbeddedSystems2.doc]

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/238

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 15 Aug 2012 08:14 UTC**
Left-over tickets from an older Design meeting. Rescheduling to next design meeting to be reviewed.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 17 Oct 2012 16:23 UTC**
Since not handled at the last meeting rescheduling to the next one.

modelica-trac-importer, 2017-01-13:
**Comment by hansolsson on 5 Dec 2012 15:18 UTC**
No embedded group at this meeting.

modelica-trac-importer, 2017-01-13:
**Modified by otter on 20 Sep 2013 10:22 UTC**

modelica-trac-importer, 2017-01-13:
**Comment by hansolsson on 23 Oct 2013 13:46 UTC**
[Missed one previously.]
Shouldn't we close these as wontfix?
The entire embedding chapter has been removed; so we need to restart this in another way I guess.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 15 Nov 2013 13:30 UTC**
80th Design Meeting is long over ... closing that milestone and still open tickets set back to \*undecided\*.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 20 Dec 2016 14:01 UTC**
Unhide for GitHub transition.

HansOlsson, 2017-02-22:
More than 3 years ago I proposed to close as wontfix, and I haven't seen any new reactions.
The reason for closing this proposal as wontfix are:

- Embedded chapter that would use it is removed.
- The proposal doesn't handle e.g. the tables in MSL, where the external files cannot just be "pre-evaluated".
 - The loadResource function is related to the underlying problem of knowing what is a resource (even for embedding in FMU - however, it is not clear how to use that without a file-system).

Obviously we should still try to handle the underlying case of easy code-generation for embedded targets without file-system.

HansOlsson, 2017-07-10:
Closing this. It was almost 4 years since I suggested to close it as wontfix, since it was primarily an improvement for the embedding-chapter of Modelica which has been removed from the Modelica specification - and the only new activity have been a list of reasons for closing it as wontfix.

It might be that we need something similar not connected to embedding-chapter - or that loadResource already solves the issue - together with a clever implementation of routines that use files.

closed 2017-07-10 by HansOlsson