modelica/ModelicaStandardLibrary #56: Would like to have the functionality of a timer

modelica-trac-importer, 2017-01-13:
**Reported by dietmarw on 10 Mar 2008 11:27 UTC**
Dear Martin,
I retrieved this issue from the an old issue tracker on our website and moved the issue here before deleting that obsolete issue tracker.
Please feel free to put this on "invalid" if this is not an issue anymore.

 * Gerd Kurzbach proposed to introduce a timer for embedded systems
   (missing "improvement suggestion")

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/56

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 29 Jul 2008 16:22 UTC**
Replying to [ticket:56 dietmarw]:
> Dear Martin,
> I retrieved this issue from the an old issue tracker on our website and moved the issue here before deleting that obsolete issue tracker.
> Please feel free to put this on "invalid" if [...]

Gerd, is this still a feature you think would be useful to have?

modelica-trac-importer, 2017-01-13:
**Comment by kurzbach on 30 Jul 2008 05:31 UTC**
Replying to [comment:1 dietmarw]:

> > \* Gerd Kurzbach proposed to introduce a timer for embedded systems (missing "improvement suggestion")
> Gerd, is this still a feature you think would be useful to have?

Yes, of course.

modelica-trac-importer, 2017-01-13:
**Modified by HansOlsson on 17 Nov 2008 13:13 UTC**

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 19 Nov 2009 12:14 UTC**
## From [milestone:Design63]
**Status:** Test implementation of Timer in Modelica ([source:ModelicaDesign/trunk/MeetingMinutesMaterial/min63/Slides-and-Documents/TestTimer.mo TestTimer.mo])

**Short Description:**
 Introduce Integer-based time in order that time events can be precisely described and handled, especially for embedded systems. Additionally, simplify definition of time events by introducing a built-in timer.
 Details see [source:ModelicaDesign/trunk/MeetingMinutesMaterial/min63/Slides-and-Documents/Modelica_Proposal_IntegerTime.doc Modelica_Proposal_IntegerTime.doc] (obsolete text should be removed) and [source:ModelicaDesign/trunk/MeetingMinutesMaterial/min63/Slides-and-Documents/Modelica_Proposal_Timer.doc Modelica_Proposal_Timer.doc].

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 15 Aug 2012 08:14 UTC**
Left-over tickets from an older Design meeting. Rescheduling to next design meeting to be reviewed.

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 17 Oct 2012 16:23 UTC**
Since not handled at the last meeting rescheduling to the next one.

modelica-trac-importer, 2017-01-13:
**Comment by hansolsson on 5 Dec 2012 15:20 UTC**
Is this handled by synchronous in Modelica 3.3? Or will it be added in 3.4?

modelica-trac-importer, 2017-01-13:
**Changelog removed by hansolsson on 5 Dec 2012 15:20 UTC**

modelica-trac-importer, 2017-01-13:
**Modified by otter on 20 Sep 2013 10:22 UTC**

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 15 Nov 2013 13:30 UTC**
80th Design Meeting is long over ... closing that milestone and still open tickets set back to \*undecided\*.

beutlich, 2017-11-22:
@gkurzbach Is this still relevant?

gkurzbach, 2017-11-23:
@beutlich You may close this ticket.

closed 2017-11-23 by beutlich