modelica/ModelicaStandardLibrary #85: Re-design and simplification of Modelica.Media

modelica-trac-importer, 2017-01-13:
**Reported by hubertus on 30 May 2008 07:44 UTC**
The discussion in the Fluid group and experiences at Modelon have shown a few shortcomings in the Media library. Discussions in the group have resulted in a suggestion for improvement that is documented in the attached file. Hubertus will start a branch, Siemens and TU-Braunschweig offered to help.  

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/85

modelica-trac-importer, 2017-01-13:
**Comment by hubertus on 30 May 2008 07:57 UTC**
Added a class diagram for the old Modelica.Media. It clearly shows that the structure is more complicated than necessary

modelica-trac-importer, 2017-01-13:
**Comment by hubertus on 30 May 2008 16:09 UTC**
Added the proposed class diagram of the new media. Note that I used UML "implements" for the partial classes even though this is extends in Modelica. The structure is much simpler and covers at the same time many more cases. Note that there is no distinction any more between mixtures and pure fluids. Pure fluids are mixtures with one component. Implementations have the default argument of {1.0} for the composition for pure fluids, so users don't have to type it.

modelica-trac-importer, 2017-01-13:
**Comment by otter on 22 Jul 2010 16:49 UTC**
This change needs conversion scripts and it is also too late for MSL3.2.

modelica-trac-importer, 2017-01-13:
**Comment by hubertus on 6 Mar 2013 22:39 UTC**
this is clearly an enhancement

modelica-trac-importer, 2017-01-13:
**Changelog removed by hubertus on 6 Mar 2013 22:39 UTC**

modelica-trac-importer, 2017-01-13:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2019-07-15:
@hubertus65 @thorade This is the oldest open issue targeted for MSL 4.0.0. Can it be covered by conversion? Is it still a feasible goal for MSL v4.0.0 being about 6 months before feature freeze?

thorade, 2019-07-19:
There are maybe some small modifications that could be done for MSL v4, but a complete redesign of the media interface would probably need a 3-year research project with input from all companies that offer commercial fluid properties libraries.

beutlich, 2019-07-19:
OK, removng milestone then.