modelica/ModelicaStandardLibrary #1153: Ideas for a new table call

modelica-trac-importer, 2017-01-14:
**Reported by otter on 1 Jun 2013 21:52 UTC**
The contract for the development of an open source implementation of the Table blocks was very successful, and it seems useful to continue in this fashion. Therefore, I plan to discuss on Wed. at the Modelica Design Meeting a new call for further improved tables and this ticket should collect ideas for such a new call. Please provide your input.

Here is a start:

 * Support more extrapolation options, see ticket !#477.
 * Support n-dimensional tables (linear or Akima-Spline interpolation in N dimensions). Several details need to be discussed, especially, whether ASCII format should be extended (or not available for ND tables) and in which form the interpolation problem should be formulated (there are several options).
 * Support new interpolation type for 1D tables: monotonic splines according to: Fritsch F.N. and Carlson R.E. (1980): Monotone piecewise cubic interpolation. SIAM J. Numerc. Anal., Vol. 17, No. 2, April 1980, pp. 238-246. These special cubic splines guarantee that if the ordinate points are monotonic, then all interpolated points are monotonic as well (so no "unwanted vibrations").

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1153

modelica-trac-importer, 2017-01-14:
**Comment by andreas&#46;heckmann on 3 Jun 2013 14:56 UTC**
Support the import/handling of MS Excel spreadsheet tables in a similar way as the xlsread function does for Matlab. This capability should enable a table look-up with respect to time or more general, with respect to another independent quantity.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 3 Jun 2013 18:24 UTC**
Replying to [comment:2 andreas.heckmann@…]:
> Support the import/handling of MS Excel spreadsheet tables in a similar way as the xlsread function does for Matlab. This capability should enable a table look-up with respect to time or more general, with respect to another independent quantity.

Probably very cumbersome in a C-only implementation. What about csv instead?

 * There also was a request of file caching, see #1110.
 * What about approximation?

modelica-trac-importer, 2017-01-14:
**Comment by andreas&#x2e;heckmann on 4 Jun 2013 08:00 UTC**
Replying to [comment:3 beutlich]:
> Replying to [comment:2 andreas.heckmann@…]:
> Probably very cumbersome in a C-only implementation. What about csv instead?
>
>  \* There also was a request of file caching, see #1110.
>  \* What about [...]

csv is water, xls is wine, may-be water is more reasonable

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 14 Jun 2013 20:56 UTC**
There is a C library with BSD license (http://sourceforge.net/projects/libxls/) that could be used for support of Excel (xls only, not xlsx) files.

modelica-trac-importer, 2017-01-14:
**Comment by ahaumer on 17 Jul 2013 13:42 UTC**
Support direct reading and writing of values from external files (e.g. Matlab) without using the externalObject (holding the values in external allocated memory). 
The use case is reading parameters from file at compile time to provide FMUs with fixed parameters, e.g. for other platforms / HIL, similar to (Dymola-specific) library DataFiles. 
Most convenient solution: possibility to switch between reading at compile time and reading at execution time; first solution to provide FMUs with fixed parameters, second solution to provide FMUs that read their parameters from file at execution time.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 6 Aug 2013 06:51 UTC**
Replying to [comment:6 ahaumer]:
> Support direct reading and writing of values from external files (e.g. Matlab) without using the externalObject (holding the values in external allocated memory).

I have two questions.
 1. What is writing needed here for?
 1. I am not sure if I understand it correctly. If you want to read at run-time from file you need any object to hold the values. This either can be a Modelica matrix with dimensions known at compile-time or an external object.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 6 Aug 2013 06:55 UTC**
Replying to [comment:2 andreas.heckmann@…]:
> Support the import/handling of MS Excel spreadsheet tables in a similar way as the xlsread function does for Matlab.

I have a working prototype that reads Excel files using libxls. Can you please provide a typical Excel file (with the formatting you usually apply). Thank you.
>This capability should enable a table look-up with respect to time or more general, with respect to another independent quantity.

You already can use CombiTable1D with input time or any other independent variable.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 3 Sep 2013 15:18 UTC**
Toni, Dr. Heckmann: Any feedback?

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 3 Sep 2013 15:22 UTC**
One more idea/fix: Support table text files with Mac line endings (= CR). Does neither work with MSL 3.2 nor MSL 3.2.1.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 18 Sep 2013 20:20 UTC**
 * It would be very few effort to add bicubic spline interpolation for CombiTable2D.

modelica-trac-importer, 2017-01-14:
**Comment by lionel&#x2e;belmon on 18 Nov 2013 10:44 UTC**
I wish to have possibility for using irregular / unstructured tables. The combiTables support very rigidly structured tables. Importing irregular measurements data needs usually to re-interpolate through another software to create "regular" tables.

First simple cases:
N series of 1D values are used to define a 2D table :
row x=0
y ![0,100,200,300,400]
z ![0,0,0,1,2]
row x=10
y ![10,50,150,300,390,450]
z ![1,0,0,1,2,4]

This is typical from measurements on a test bench... you fix one variable (speed for instance) and measure some other variables but cannot control the intervals, so you end up with an irregular map.

Second, fully unstructured data points:
the user just input a series of points (x,y,z)
0,0,10
0,10.5,100
15,100,200
17,120,210
...
This is much more complex and requires to have Delaunay triangulation etc...

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 19 Nov 2013 08:10 UTC**
Replying to [comment:15 lionel.belmon@…]:
> I wish to have possibility for using irregular / unstructured tables.

I guess you ask for something like [griddata](http://docs.scipy.org/doc/scipy/reference/tutorial/interpolate.html#multivariate-data-interpolation-griddata) available in SciPy.
Note that there are also meshless methods (like radial basis functions) available.

modelica-trac-importer, 2017-01-14:
**Comment by lionel&#x2e;belmon on 21 Nov 2013 03:39 UTC**
Thomas, to keep things simple: N series of 1D tables to define a 2D table would already be much helpful. And this is rather straightforward to implement, you just need to use the 1D table functions and add a linear interpolation between the 2 1D table.

Luxury would be a fully unstructured table :

About "griddata" : This is the correct functionality, but the implementation is very strange. I don't see why the user should provide a regular "grid". This is strange. The rather common way to evaluate to interpolate into the irregular series of points, it's to create a delaunay triangulation, interpolate on the plane defined by the triangles.
a quick google search gave me this :
http://doc.astro-wise.org/matplotlib.delaunay.interpolate.html
Getting smooth interpolant on a triangle mesh is something difficult I think, so linear interpolation would already be good. 
Be careful also that evaluation time can be long for big mesh (triangle search).

Matlab has similar functions with scatteredInterpolant...

modelica-trac-importer, 2017-01-14:
**Comment by anonymous on 21 Nov 2013 08:57 UTC**
Replying to [comment:17 lionel.belmon@…]:
> About "griddata" : This is the correct functionality, but the implementation is very strange. I don't see why the user should provide a regular "grid".

There is no need to provide a regular grid. It was just used as an example of where to evaluate the interpolant.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 6 Feb 2014 14:16 UTC**
There is a request from a SimulationX customer to make table lookup available as function call, i.e. support variable table data from the model itself. Table update from file is already kind of supported by function _readTableData_ with argument _forceRead=true_. But it still requires the external table object.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 6 Feb 2014 14:41 UTC**
Not sure if there was a discussion at Design81.

modelica-trac-importer, 2017-01-14:
**Comment by christiankral on 19 Feb 2014 19:27 UTC**
I have a physical issue where I model a quantity B as a function of position s. In order to determine the derivative of B with respect to s, 
```
dB/ds,
``` 
I could use the relationship
```
dB/ds * der(s) = der(B)
```
If der(s) is equal to zero, dB/ds cannot be derived correctly during initialization. And exactly this is the problem in model. Therefore, it were very useful for me to have the derivative of the output with respect the input available -- according to the actual interpolation / approximation. 

I could certainly also store the derivative dB/ds in my table data, but this is error prone and the data have to match the actual interpolation / approximation in order to have a consistent model. 

Particularly, it were also interesting to use a function instead of a table in my model, since all the equations are stated textually.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Dec 2014 07:01 UTC**
Ticket retargeted after milestone closed

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 12 May 2015 09:51 UTC**
 * ~~Support new interpolation type for 1D tables: monotonic splines according to: Fritsch F.N. and Carlson R.E. (1980): Monotone piecewise cubic interpolation. SIAM J. Numerc. Anal., Vol. 17, No. 2, April 1980, pp. 238-246. These special cubic splines guarantee that if the ordinate points are monotonic, then all interpolated points are monotonic as well (so no "unwanted vibrations").~~
Resolved by 6c5de73996a56837ab3c4cbcb7d86100c5d45f68 for #1717.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 16 Nov 2015 20:20 UTC**
Replying to [comment:10 beutlich]:
> One more idea/fix: Support table text files with Mac line endings (= CR). Does neither work with MSL 3.2 nor MSL 3.2.1.

I guess this idea is obsolet.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 7 Mar 2016 18:54 UTC**
 * #1820: Print verbose message when a spline interpolation degrades to linear interpolation due to lack of sample points
 * #1839: Custom extrapolation for 1D and 2D table blocks
 * #1899: Read table already in constructor
 * #1627: Add new smoothness kind "LinearSegmentsWithTimeEvents" with time events for every interval boundary

modelica-trac-importer, 2017-01-14:
**Comment by otter on 7 Mar 2016 21:28 UTC**
We should try to make a concrete plan at this design meeting for a next  table call. I will schedule a planning session on Wednesday

modelica-trac-importer, 2017-01-14:
**Comment by otter on 7 Mar 2016 21:33 UTC**
I think in the next call at least the following should be included, besides https://trac.modelica.org/Modelica/ticket/1153#comment:27:

 * Linear interpolation of ND-Tables
 * Excel read/write

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 7 Mar 2016 22:01 UTC**
Replying to [comment:29 otter]:
>  \* Excel read/write

What is Excel write needed for within the context of tables? Or do you have writeRealMatrix (to Excel) in mind?

modelica-trac-importer, 2017-01-14:
**Comment by otter on 7 Mar 2016 22:13 UTC**
Replying to [comment:30 beutlich]:

> Replying to [comment:29 otter]:
> 
> > \* Excel read/write
> > 
> What is Excel write needed for within the context of tables? Or do you have writeRealMatrix (to Excel) in mind?

Yes (so extending readRealMatrix and writeRealMatrix with additional support for Excel-Files).

modelica-trac-importer, 2017-01-14:
**Comment by otter on 7 Mar 2016 22:14 UTC**
Sorry, I wss too quick: One needs readRealMatrix/writeRealMatrix, CombiTableX, CombiTimeTableX with additional Excel-File Format support

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 10 Mar 2016 09:55 UTC**
Another issue that would be good to handle is to ensure (and document) that the table-blocks are re-entrant.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 10 Mar 2016 12:24 UTC**
Replying to [comment:33 hansolsson]:
> Another issue that would be good to handle is to ensure (and document) that the table-blocks are re-entrant.

Where do you see the danger of a table-block being non-re-entrant?

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 11 Mar 2016 15:52 UTC**
Replying to [comment:34 beutlich]:
> Replying to [comment:33 hansolsson]:
> > Another issue that would be good to handle is to ensure (and document) that the table-blocks are re-entrant.
> Where do you see the danger of a table-block being non-re-entrant?

Someone told me there might be a problem, and I hadn't investigated it. 

Looking at the code one possibility would be multiple callers using getValue for the same object - but that seems like bad code generation. (I also looked at strtok - but it uses one buffer per thread.)

However, simultaneous call of getValue and getDerValue for the same object might also cause problems; and that isn't clearly an error. We could state that each external object may only be used as argument to one function call at a time. It is not ideal - but acceptable as long as it is clearly documented.

I haven't investigated if it is actually a problem - I just see that it might be.

modelica-trac-importer, 2017-01-14:
**Comment by anonymous on 11 Mar 2016 20:03 UTC**
Correct. Both getValue and getDerValue vhange the last used interval member of the external object. This could be made thread-safe by another object-specific mutex/critical section.

modelica-trac-importer, 2017-01-14:
**Comment by gottelt on 23 Mar 2016 08:18 UTC**
I don't know if this is clear anyway but I think defining n-dimensional arrays as pure Modelica code shall be possible, too.

modelica-trac-importer, 2017-01-14:
**Comment by anonymous on 23 Mar 2016 08:25 UTC**
Replying to [comment:37 gottelt@…]:
> I don't know if this is clear anyway but I think defining n-dimensional arrays as pure Modelica code shall be possible, too.

This only would be useful if the table data comes from the model and is not read from file. And having the table data within the model also means the number of dimensions must be known. Thus if you have the use case with a small table array from within the model and a fixed dimensions N this may be considered.

modelica-trac-importer, 2017-01-14:
**Comment by gottelt on 23 Mar 2016 08:46 UTC**
Replying to [comment:38 anonymous]:
> Replying to [comment:37 gottelt@…]:
> This only would be useful if the table data comes from the model and is not read from file. And having the table data within the model also means the number of [...]

Exactly: The idea is to locally define an n-dimensional array within a model or function.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 23 Mar 2016 15:35 UTC**
The discussion of the proposed table ideas stopped w/o results at the last MDM in Linköping. Therefore, I started a poll about finding a date for a webmeeting (just after Easter) discussing this topic. If you like to contribute please select the date(s) where you are able to attend: http://doodle.com/poll/893n3aubvhzzdv4t

Thank you for your contribution.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 5 Apr 2016 09:07 UTC**
Replying to [comment:40 beutlich]:
> The discussion of the proposed table ideas stopped w/o results at the last MDM in Linköping. Therefore, I started a poll about finding a date for a webmeeting (just after Easter) discussing this [...]

The web meeting will take place April 6th 2016.

modelica-trac-importer, 2017-01-14:
**Comment by otter on 6 Apr 2016 14:27 UTC**
From DLR colleagues, I got the following feedback regarding tables:
Many real-time systems do not have a file systems. If a Modelica model is used on such a real-time system it is currently very awkward to utilize tables in a Modelica model. The Blocks.Tables.xxx provide the option that the table is within a C-code that is linked. However, this is too inconvenient to use. Best would be:
- Table defined on file (1D, 2D, the newly planned ND).
- Optionally, a program generates a C-code where the table is included.
- This C-code is utilized by the Modelica model.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 8 Apr 2016 14:03 UTC**
Please find the minuting minutes attached attachment:160406_Web_meeting_MSL_Tables_development.pdf

beutlich, 2018-02-26:
> **Comment by @lionelbelmon on 18 Nov 2013 10:44 UTC**
I wish to have possibility for using irregular / unstructured tables. The combiTables support very rigidly structured tables. Importing irregular measurements data needs usually to re-interpolate through another software to create "regular" tables.

> First simple cases:
N series of 1D values are used to define a 2D table :
row x=0
y [0,100,200,300,400]
z [0,0,0,1,2]
row x=10
y [10,50,150,300,390,450]
z [1,0,0,1,2,4]

> This is typical from measurements on a test bench... you fix one variable (speed for instance) and measure some other variables but cannot control the intervals, so you end up with an irregular map.

> Second, fully unstructured data points:
the user just input a series of points (x,y,z)
0,0,10
0,10.5,100
15,100,200
17,120,210
...
This is much more complex and requires to have Delaunay triangulation etc...

@lionelbelmon Let's take your example from the first case for now. It also maps to unstructured/scattered data points as stated by your second case. I can think of two possibilities to move ahead.

1. If you want to stick with interpolation you need to transform the unstructured/scattered data points to a regular/rectangular 2D grid table to be used with [CombiTable2D](http://doc.modelica.org/om/Modelica.Blocks.Tables.CombiTable2D.html). The following MATLAB script does the job

```matlab
x=[0,0,0,0,0,10,10,10,10,10,10];
y=[0,100,200,300,400,10,50,150,300,390,450];
z=[0,0,0,1,2,1,0,0,1,2,4];
xi=[0,10];
yi=[0;10;50;100;150;200;300;390;400;450]; % this extrapolates at [0;450] and [10;0]
yi=[  10;50;100;150;200;300;390;400]; % no extrapolation
zi=griddata(x,y,z,xi,yi);
table=[0,xi;yi,zi]'; % need to transpose for CombiTable2D if xi is for u1 and yi for u2
save table.mat table
```
The saved table is the input for CombiTable2D at simulation run time.
```mo
Modelica.Blocks.Tables.CombiTable2D combiTable2D(
  tableOnFile=true,
  tableName="table",
  fileName="table.mat");
```

2. If you think of approximation you can utilize the [ApproxSpline](https://github.com/modelica-3rdparty/ApproxSpline) library and directly input the scattered data.
```mo
ApproxSpline.Blocks.Surf2d surf2D(
  rectangular=false, // -> scattered = true
  data=[0,0,0;0,100,0;0,200,0;0,300,1;0,400,2;10,10,1;10,50,0;10,150,0;10,300,1;10,390,2;10,450,4],
  s=1,
  kx=1,
  ky=1);
```
Determining the smoothing factor and optional weights might be non-trivial still.

After all, are these possibilities acceptable workarounds for you?

lionelbelmon, 2018-02-27:
Thomas,
the case (1) is an interpolation in a series of 1D curves. If we only apply linear interpolation, it is rather straightforward to solve. It could be built with some simple extension of the existing msl tables. You interpolate between the 2 nearest 1D curve, then interpolate between the 2 resulting values. I think it brings a lot of comfort to users and is rather 'simple' to develop. Typical applications include "engine torque maps" or "engine fuel consumption maps". 
The point here is : it's rather easy to do based on existing MSL, it would help many users. So this is a good candidate for going into MSL.

the case (2) is much more general and complex. I did not try the library you mention though. This probably does not need to go in MSL.

beutlich, 2018-02-27:
Regarding the first case. You are right it can be solved based on bilinear interpolation (of the four adjacent data points) and thus without triangulation. Still, it can be generalized to the second case of scattered data points. Using [griddata](https://www.mathworks.com/help/matlab/ref/griddata.html) with (default) [method](https://de.mathworks.com/help/matlab/ref/griddata.html#input_argument_d119e381781) `'linear'` does (based on triangulation) exactly the job you/we want.

Sure, [ApproxSpline](https://github.com/modelica-3rdparty/ApproxSpline) should not go to MSL. It was decided once to avoid dependencies on [third-party C libraries](https://github.com/modelica/Modelica/issues/2253) for the sake of maintenance effort. Having it as an extra library is sufficient.

thorade, 2019-04-05:
> * It would be very few effort to add bicubic spline interpolation for CombiTable2D.

Sorry for posting on this already very long thread, I did not follow all discussion, so I am unsure about the current status for bicubic interpolation in 2 dimensions. This is something that would be very helpful for me, and I would be happy to assist with testing etc. With some help and quickstart I could also try to contribute to implementation.

dietmarw, 2020-12-15:
Stumbling across this old thread. When contemplating which format to use as data source for time series log data I was surprised that loading (no need for importing) of the two most commonly found formats is still not in option in Modelica:

 * xlsx (no need for the outdated and restricted xls format)
 * csv

Being able to store data on one of those two formats  (and then read them from CmobiTimeTable) also has the benefit of easy manipulation and processing outside Modelica with standard spreadsheet tools or text editors.
Is it very complicated to add that support to the existing CombiTimeTable?

sjoelund, 2020-12-17:
@dietmarw you mean in Modelica or in MSL? https://github.com/modelica-3rdparty/ExternData/blob/master/ExternData/Examples/package.mo#L35

dietmarw, 2020-12-17:
I'm aware of that library but mean that this functionality should be available in the MSL without having to rely on 3rd party libraries.

beutlich, 2020-12-17:
I am currently working on some table data input improvements as required by @dietmarw and as extension to the MSL blocks.

beutlich, 2020-12-19:
[ModelicaTableAdditions v1.0.0](https://github.com/tbeu/ModelicaTableAdditions/releases/tag/v1.0.0) (for MSL 3.2.3) is available which adds CSV support to CombiTable blocks. The blocks of ModelicaTableAdditions can be used as replacement of table blocks of MSL table blocks.

beutlich, 2024-04-01:
> Sorry for posting on this already very long thread, I did not follow all discussion, so I am unsure about the current status for bicubic interpolation in 2 dimensions. This is something that would be very helpful for me, being able to have continuous value as well as continuous first and second derivatives (C2 smoothness).

With cubic/bicubic table interpolation being available via [ModelicaTableAdditions v3.0.0](https://github.com/tbeu/ModelicaTableAdditions/releases/tag/v3.0.0) I take the liberty to close this stale issue.

closed 2024-04-01 by beutlich