modelica/ModelicaStandardLibrary #1260: Complex.scalarProduct not a scalar product

modelica-trac-importer, 2017-01-14:
**Reported by petar on 26 Aug 2013 06:57 UTC**
The scalar product for complex vectors is actually not a scalar product. Currently, the scalar product for complex vectors is defined as:

```

    function scalarProduct "Scalar product c1*c2 of two complex vectors"
      import Complex;
      input Complex c1[:] "Vector of Complex numbers 1";
      input Complex c2[size(c1, 1)] "Vector of Complex numbers 2";
      output Complex c3 "= c1*c2";
    algorithm
      c3:=Complex(0);
      for i in 1:size(c1, 1) loop
              c3:=c3 + c1[i] * c2[i];
      end for;
      /*
       c3 :=Complex(c3.re + c1[i].re*c2[i].re - c1[i].im*c2[i].im,
                    c3.im + c1[i].re*c2[i].im + c1[i].im*c2[i].re);
       */
    end scalarProduct;
```

which is just the scalar product for real vectors translated to complex vectors. An example of why this is not a proper scalar product is that using this scalar product you can get v \* v = 0 instead of the expected v \* v = norm(v)^2. (A concrete example where this happens is v = (1 + i, 1 - i)). For more information, see Wikipedia; http://en.wikipedia.org/wiki/Scalar_product#Complex_vectors.

To get a mathematically sound scalar product for complex numbers you need to apply conjugation on one of the vectors (which one differs between fields; quantum physicists mostly conjugate the left vector and mathematicians the right vector). I.e., the code should be:

```

      for i in 1:size(c1, 1) loop
              c3:=c3 + ComplexConjugate(c1[i]) * c2[i];
      end for;

```

or

```

      for i in 1:size(c1, 1) loop
              c3:=c3 + c1[i] * ComplexConjugate(c2[i]);
      end for;
```

where ComplexConjugate is the function mapping a + bi to a - bi.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1260

beutlich, 2020-11-01:
I'd prefer to move (and fix) scalarProduct from Complex to Modelica.ComplexMath since we only have there conjugation (function Modelica.ComplexMath.conj).

HansOlsson, 2020-11-02:
I would say I agree with this comment:
> I'd prefer to move (and fix) scalarProduct from Complex to Modelica.ComplexMath since we only have there conjugation (function Modelica.ComplexMath.conj).

but was a bit surprised that I didn't see it here.

beutlich, 2020-11-02:
I deleted my yesterday comment since we do not have conversion of operator functions in the Modelica language specification. But thinking again on it, I believe it might still be best to remove Complex.'*'.scalarProduct and add Modelica.ComplexMath.Vectors.scalarProduct (with a Real output) without any conversion. I do not believe that Complex.'*'.scalarProduct is ued anywhere, but I have no evidence.

christiankral, 2020-11-06:
I wonder why I wasn't aware of this ticket before...

I am very much in favor of fixing this issue. I do prefer variant 1 of the proposed implementations

```modelica
for i in 1:size(c1, 1) loop
    c3:=c3 + Modelica.ComplexMath.conj(c1[i]) * c2[i];
end for;
```

as this is the same way as it is implemented in Matlab and Julia. 

@beutlich Shall a PR go to the master branch?

Moving ```Complex.scalarProduct``` to ```Modelica.ComplexMath.Vectors.scalarProduct``` is a non backwards compatible change, isn't? So moving the function `scalarProduct` requires a major release which includes conversion scripts.

beutlich, 2020-11-06:
It is compatibility breaking in any case since the return value changes from Complex to Real.

christiankral, 2020-11-06:
The return type is only Real in case of ```scalarProduct(a,a)```, not in a general case:

[image]

beutlich, 2020-11-06:
Right. This makes the fix easier.

beutlich, 2020-11-09:
> I am very much in favor of fixing this issue. I do prefer variant 1 of the proposed implementations
> 
> ```modelica
> for i in 1:size(c1, 1) loop
>     c3:=c3 + Modelica.ComplexMath.conj(c1[i]) * c2[i];
> end for;
> ```
> 
> as this is the same way as it is implemented in Matlab and Julia.

I'd prefer to not let Complex depend on Modelica (as is now). Either we remove the operator function or we introduce conjugate also to Complex.

> @beutlich Shall a PR go to the master branch?

Yes.

closed 2021-05-14 by christiankral