modelica/ModelicaStandardLibrary #1299: Table test data of ModelicaTest should not be stored in the MSL

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 1 Oct 2013 10:57 UTC**
The `ModelicaTest` examples link to test data files `*.mat` which are stored inside 

`Modelica/Resources/Data/Tables`

Since these are test data files from the `ModelicaTest` library these really ought to be moved to 
`ModelicaTest/Resources/Data/Tables`

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1299

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 25 Oct 2013 21:27 UTC**
FIXED in a3041258138620322ff5719d82c53f6b3d3f78d3:  Moving test table data files into ModelicaTest

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 28 Oct 2013 09:42 UTC**
Change is non-backward compatible.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 28 Oct 2013 09:43 UTC**
Change is reverted in 86c27c9bb7cc8f7ebe64d198ded7c56a52473137 and can not really be fixed currently. See also #1339 for a follow up.

modelica-trac-importer, 2017-01-14:
**Modified by beutlich on 6 Oct 2015 09:24 UTC**

closed 2017-01-14 by modelica-trac-importer

HansOlsson, 2019-12-02:
To me there are two parts here:

1. The general need for renaming resources: I agree that this will need something more for conversion scripts.
2. This specific case: Since the resources are test-resources, I would argue that they can be changed without the need for conversion scripts - similarly as modifying the examples without conversions.