modelica/ModelicaStandardLibrary #1339: How to handle changes to resource data shipped with the MSL

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 28 Oct 2013 09:40 UTC**
Currently when doing changes to classes within Modelica classes we can use (hopefully soon standardised) conversion script to automatically upgrade user models from one version of the MSL to the next one. 

Now when it comes to resource data we don't actually have a procedure in place to update user models/libraries to reflect changes to resource data they use and have changed (e.g., name, location). This issue came up when trying to fix #1299, which I simply could not. Not even for a new MSL version with conversion script. I wonder how this can (if at all possible) be solved. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1339

beutlich, 2019-08-09:
For me this looks like a specification issue of the conversion scripts: With Modelica spec. 3.4 resource files cannot be renamed nor moved w/o breaking compatibility.

Changing milestone then.

dietmarw, 2019-11-29:
@HansOlsson probably good to add this to the list for specification related tickets