modelica/ModelicaStandardLibrary #1365: C-code used by Modelica.Utilities

modelica-trac-importer, 2017-01-14:
**Reported by ahaumer on 27 Nov 2013 13:17 UTC**
Is the C-code used by Modelica.Utilities delivered together with the MSL (i.e. tool independent) or is the implementation up to the tool vendors? 
The question came up when using Modelica.Utilities.Examples.readRealParameter. In Dymola, the parameters are read during translation. This is fine if I want to do conditional instantiation depending on parameters read from file. I've attached a small test lib with an alternative "readRealParameter" which allows to read the parameters during execution (after translation). Of course, conditional instantiation is not possbile. But you may exchange the parameter file after translation.
We need a clarification about the desired behaviour, and the same behaviour independent of the tool.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1365

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Dec 2014 07:01 UTC**
Ticket retargeted after milestone closed

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 8 Apr 2016 14:06 UTC**
This was discussed at the [attachment:160406_Web_meeting_MSL_Tables_development.pdf:ticket:1153 web meeting] with the following result:
Not sure how to proceed with it. One option is to keep the 3rd-party C-code (e.g., zlib or any future Excel/XML libraries) outside the MSL and let the tool vendors take care of it. This is currently done for the libraries LAPack and for HDF5 (as option for MAT-files).

modelica-trac-importer, 2017-01-14:
**Modified by beutlich on 8 Apr 2016 14:07 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by sjoelund&#46;se on 8 Apr 2016 14:09 UTC**
Replying to [comment:3 beutlich]:
> One option is to keep the 3rd-party C-code (e.g., zlib or any future Excel/XML libraries) outside the MSL and let the tool vendors take care of it. This is currently done for the libraries LAPack and for HDF5 (as option for MAT-files).

This is a bit problematic since MSL should be a reference library. And I don't think it is wise to just suggest that all library developers should communicate with all tools to get dependencies installed...

modelica-trac-importer, 2017-01-14:
**Modified by tsommer on 8 Apr 2016 14:14 UTC**

beutlich, 2020-05-08:
Closing, since not sure what to do or if it can be considered a tool-issue.

closed 2020-05-08 by beutlich