modelica/ModelicaStandardLibrary #1400: Mathematical mean functions

modelica-trac-importer, 2017-01-14:
**Reported by kdavies on 17 Jan 2014 00:39 UTC**
This ticket is to add a package of mathematical mean functions to Modelica.Math.  They were necessary for a model library I created and maybe they are useful to others.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1400

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 17 Jan 2014 09:20 UTC**
I think we need to consider these in more detail: first and foremost exactly what we need in MSL (do we need mean-functions at all? other similar functions? the median? as blocks as well?), and then where to place these functions.

Looking at the proposed implementation I see two issues:
* No guard against unnecessary overflow. The product of all elements is likely to overflow for longer vectors (just think of a geometric mean of a vector of 100 values each between 50 and 100)- there are ways of handling that, but it makes the algorithm longer - especially if you want to avoid any unnecessary rounding errors as well.
* Too many ifs. Having a special case for size=2 is not needed, and I found it more confusing for the harmonic mean.

modelica-trac-importer, 2017-01-14:
**Comment by Kevin Davies on 17 Jan 2014 15:56 UTC**
I understand.  I was hesitant myself (re: if and where they should be included in MSL), but I thought I may as well share them.  You can close this if you want.  If you decide to take it, then I'll improve the implementation.

tbeu, 2017-01-18:
Point to @kdavies4.

beutlich, 2017-02-27:
@kdavies4 @HansOlsson Can this issue be closed now that there is Mean, RectifiedMean, ContinuousMean, RootMeanSquare, Variance and StandardDeviation in Modelica.Blocks.Math of MSL v3.2.2?

HansOlsson, 2017-02-27:
I don't think we can close this just because there are blocks for Mean etc; since these functions could be used to perform analysis inside algorithms.