modelica/ModelicaStandardLibrary #1410: Remove or reduce to a minimum the usage of cardinality in MSL

modelica-trac-importer, 2017-01-14:
**Reported by hubertus on 5 Feb 2014 18:13 UTC**
Cardinality has been deprecated for a while, but is used at too many places in the MSL. This should be fixed asap, because we should not give bad examples and use deprecated features. I will post a list as found by Dymola (as image, does not copy text). 

The affected libraries are only Stategraph, MultiBody, and Fluid. This should be doable. I'll make Martin the first owner for MultiBody. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1410

modelica-trac-importer, 2017-01-14:
**Comment by msasena on 6 Feb 2014 19:03 UTC**
This issue was raised in ticket #1265.  Please consider closing one or the other to avoid duplication.

modelica-trac-importer, 2017-01-14:
**Comment by hubertus on 6 Feb 2014 19:34 UTC**
The same issue had already been raised in !#1265 which has been closed as duplicate. Here is the original ticket description, for completeness: 

The specification v3.2.1 describes cardinality by saying "This is a  deprecated operator. It should no longer be used, since it will be  removed in one of the next Modelica releases."  However, in looking  through MSL v3.2.1, I found:

 * 9 occurences of cardinality in the Fluids library
 * 58 occurences of cardinality in the Multibody library
 * 22 occurences of cardinality in the Stategraph library

Granted, a few of these instances are commented out within the model,  but I still think there are far too many occurences of an operator which  is deprecated.  I don't know how much other tools / libraries depend on  this operator, but it seems like including a deprecated operator in the  MSL is generally a bad idea.

Can those libraries be modified to remove all uses of cardinality, or is  cardinality going to be forever "deprecated", but never "obsolete" /  removed?

modelica-trac-importer, 2017-01-14:
**Comment by leo&#46;gall on 10 Feb 2014 10:15 UTC**
For MultiBody, see Martin's analysis in ticket:1409#comment:33.

modelica-trac-importer, 2017-01-14:
**Comment by rfranke on 12 Jan 2015 08:03 UTC**
Fluid uses the cardinality operator for diagnostics. All 9 occurences are of the form:
```
within Modelica.Fluid.Sources.BaseClasses;
partial model PartialSource
  parameter Integer nPorts = 0 "Number of ports" annotation(Dialog(connectorSizing = true));
  Interfaces.FluidPorts_b ports[nPorts];
equation
  for i in 1:nPorts loop
    assert(cardinality(ports[i]) <= 1, "
      each ports[i] of boundary shall at most be connected to one component.
      If two or more connections are present, ideal mixing takes
      place with these connections, which is usually not the intention
      of the modeller. Increase nPorts to add an additional port.
    ");
  end for;  
end PartialSource;
```
It would be a pity if these cardinality-asserts and the given diagnostic messages would just be removed, leaving a modeler with more sources of errors and less info.

The Modelica spec says in section 3.7.2.3:

//[The cardinality operator is deprecated for the following reasons and will be removed in a future release:
- Reflective operators may make early type checking more difficult.
- Almost always abused in strange ways
- Not used for Bond graphs even though it was originally introduced for that purpose.
]//

Which of these reasons applies to the use of cardinality for diagnostics? Wasn't it even introduced for such cardinality-asserts -- see also #63? I think it was deprecated because an operator that may appear in any model equation is too powerful and tricky. 

Before we can remove `cardinality` from MSL, we need a less powerful, i.e. more specific way to provide the intended diagnostic info. Opposed to MulitBody (see ticket:1409#comment:33) we don't need `Modelica_MustBeConnected`, but a kind of `each MustBeConnectedAtMostOnce`.

Btw. a more specific info, like an annotation for the connector definition, could also be propagated to exported models, e.g. if FMI will be extended with flow and stream connectors.

modelica-trac-importer, 2017-01-14:
**Comment by ahaumer on 24 Sep 2015 10:18 UTC**
Ticket re-assigned to Language group:
We need a less powerful operator in the next Modelica Spec. replacing cardinality.
Maybe an annotation useful for FMUs.
Afterwards reassign this ticket to Library group and re-schedule this ticket for a MSL version following the new Modelica Spec.

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 24 Sep 2015 11:34 UTC**
I have tried to look at the existing uses.
In Modelica.StateGraph the assertions should not be needed (the connectors contain input and output - they should thus be connected exactly once).
In Modelica.Fluid there is the diagnostics for multiple connections. The model would translate in the bad cases - it is just that it is unlikely to be the intention of the modeler (and it is possible to construct the same problem without the diagnostics; and I believe a temperature sensor would be sort of ok).
In Modelica.Mechanics.MultiBody there are various diagnostics - not all are neeeed.

 * FixedRotation gives diagnostics if neither side is connected; but the overdetermined connector graph lacks a root in that case. Thus we could just remove the diagnostics.
 * PointMass gives an error if selected as root and connected (very important and unlikely).
 * PartialCutForceSensor has broken error diagnostics. Modelica.Mechanics.MultiBody.Sensors.CutForce connected on neither side gives error - unrelated to assertion - connected on one side the assertion is not triggered (zero flow is measured)
* LineForceWithMass (etc) uses it for more than assertions. Need investigation

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 25 Sep 2015 10:32 UTC**
The LineForceWithMass is really problematic.

To me it seems similar to the "bearing" (now "support") connector used MSL 2 in Modelica.Mechanics.Rotational; that logic was simplified by having an explicit parameter 'useSupport' instead of using the cardinality (there are two separate connectors in this case).

Something similar could be done for these classes, but:
* It will require conversions of user models
* It used a special conversion-variant:

` convertModifiers("...", {"cardinality(bearing)=1"}, {"useSupport=true"}); `

That is not included in the current conversion-proposal (it was seen as one of those strange special cases that we always need).

If we can remove this the rest could be handled by one of the following:
* Only allowing cardinality in asserts.
* Having specific annotations for cardinality.
The advantage with having annotations is that we can evaluate them early (e.g. even if unconnected root). 
The minor disadvantage is that PointMass is really an error, and an annotation generating a blocking error seems odd.

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 23 Mar 2016 15:58 UTC**
The conversion MCP contains the possibility to replace cardinality in LineForceWithMass with separate models/parameters.

If we accept that cardinality can be used in asserts it is then only a matter of updating MSL (once the MCP is added to the spec).

beutlich, 2019-03-26:
@HansOlsson Can you confirm that we are able to convert Modelica.Mechanics.MultiBody.Forces.{LineForceWithMass, LineForceWithTwoMasses} as you proposed above?

beutlich, 2019-11-02:
Not feasible for MSL v4.0.0 -> removing milestone.