modelica/ModelicaStandardLibrary #1425: Conditionally uninitialized local variable dbrgdb in Modelica.Electrical.Spice3.Internal.Mos2.drainCurRevised used

modelica-trac-importer, 2017-01-14:
**Reported by jfrenkel on 17 Feb 2014 13:47 UTC**
In the function Modelica.Electrical.Spice3.Internal.Mos2.drainCurRevised is the local variable dbsrdb only initialised if the input vds <> 0.0. Otherwise the local variable dbsrdb is used if vgs <= out_c.m_von and vds > out_c.m_vdsat is. These seems to be true in the most cases but it would be good to have a defined value in all cases.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1425

modelica-trac-importer, 2017-01-14:
**Comment by jfrenkel on 17 Feb 2014 13:50 UTC**
The same issue is in the function Modelica.Electrical.Spice3.Internal.Mos2.drainCur.

modelica-trac-importer, 2017-01-14:
**Comment by jfrenkel on 17 Feb 2014 13:58 UTC**
The same issue in the functions Modelica.Electrical.Spice3.Internal.Mos2.drainCurRevised and Modelica.Electrical.Spice3.Internal.Mos2.drainCur is there for the local variable x4. It is only initialised if delta4 >= 0.

beutlich, 2017-04-19:
@FrenkelJ Resolved in master.

closed 2017-04-19 by beutlich