modelica/ModelicaStandardLibrary #1473: (Combi)TimeTable with different time base

modelica-trac-importer, 2017-01-14:
**Reported by beutlich on 30 Apr 2014 11:29 UTC**
The blocks Modelica.Blocks.Sources.TimeTable and Modelica.Blocks.Sources.CombiTimeTable require the first column of the table matrix/array to be in the SI unit Seconds. Unfortunately there is no unit handling (possible) for array data but Seconds is somehow unsuitable for weather data (where units can be in Hours, Days etc.) and leads to nicely tuned tables like seen in Buildings.Examples.Tutorial.Boiler.System1
```modelica
  Modelica.Blocks.Sources.CombiTimeTable timTab(
      extrapolation=Modelica.Blocks.Types.Extrapolation.Periodic,
      table=[      0, 0;
              8*3600, 0;
              8*3600, QRooInt_flow;
             18*3600, QRooInt_flow;
             18*3600, 0;
             24*3600, 0]) "Time table for internal heat gain"
    annotation (Placement(transformation(extent={{-20,70},{0,90}})));
```

I propopse to add a new, non-zero parameter timeBase of type Modelica.SIunits.Time that is used to scale the input time of the time table. See the demonstrative model.

```modelica
  model A 
    parameter Modelica.SIunits.Time timeBase(min=Modelica.Constants.eps)=1 "Time base of first table column";
    Real y "Scaled time";
  equation 
    y = time/timeBase;
  end A;
```

The milestone is MSL3.2.1 since this tiny enhancement can be done in a backward compatible way.

If there is no contribution to this enhancement within the next two weeks (till 14th of May 2014) I consider this as agreement and will perform the above proposed changes. Nevertheless I hope to see some contribution from members of ModelicaTablesList and other readers.

I know there was an [article](https://modelica.org/events/modelica2014/proceedings/html/submissions/ECP14096885_AndresDiehlSommer.pdf) about general unit and type safety for table data at the last MoCo but this is compeletly out of scope here.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1473

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 30 Apr 2014 14:59 UTC**
Update: I actually prefer to have the Evaluate=true annotation set at timeBase.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 16 May 2014 11:40 UTC**
70f01e3f87338629f5cffc7e739785d8226ae178 introduces a new parameter timeScale for TimeTable and CombiTimeTable. Since it was not a single line change I decided to only add to the trunk (i.e. MSL > 3.2.1).

The fix is physically correct w.r.t. to quantities and units. The table is unitless, especially is the first column that previously was interpreted as time axis. Thus if a unitless, scaled timeScaled=time/timeScale is used the table look-up does no longer depend on units.

There is one small issue in CombiTimeTable though. Parameters t_min and t_max are of type SIunits.Time and previously returned the min/max abscissa values of the table. I decided to keep the units for t_min/t_max therefore they now return the scaled min/max abscissa values (which breaks backward compatibility in case timeScale is not equal to one). The original min/max values of the table are returned by new parameters t_minScaled/t_maxScaled, respectively.

modelica-trac-importer, 2017-01-14:
**Changelog modified by beutlich on 16 May 2014 11:40 UTC**
Added new parameter timeScale to Modelica.Blocks.Sources.TimeTable and Modelica.Blocks.Sources.CombiTimeTable to explicitly scale the first table column

closed 2017-01-14 by modelica-trac-importer

Rahaf1975, 2019-12-15:
> **Reported by beutlich on 30 Apr 2014 11:29 UTC**
> The blocks Modelica.Blocks.Sources.TimeTable and Modelica.Blocks.Sources.CombiTimeTable require the first column of the table matrix/array to be in [...]

Could you please tell me how I can upload a *.mat file for weather data into Dymola 2018 exactly in details. I am new in using Dymola and I have the weather data for the location I want as mat file. I read that I can use CombiTable but could you please tell me how exactly.
your help is much appreciated