modelica/ModelicaStandardLibrary #1478: Declare readTableData as impure

modelica-trac-importer, 2017-01-14:
**Reported by hansolsson on 9 May 2014 12:48 UTC**
The functions readTableData in
Modelica.Blocks.Sources.CombiTimeTable
Modelica.Blocks.Tables.CombiTable1D
Modelica.Blocks.Tables.CombiTable1Ds
Modelica.Blocks.Tables.CombiTable2D
should be marked as impure (since it has an important side-effect) when MSL is based on Modelica Specification 3.3 (or later).

Note that it should not be done for MSL 3.2.x.

Milestone should probably be MSL 3.3; but it does not exist.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1478

modelica-trac-importer, 2017-01-14:
**Comment by anonymous on 9 May 2014 14:06 UTC**
See also comment 3 of #1259.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 14 Aug 2014 11:09 UTC**
Further impure functions are
 * Modelica.Utilities.System.getWorkDirectory
 * Modelica.Utilities.System.getEnvironmentVariable
 * Modelica.Utilities.System.command
 * Many functions from Modelica.Utilities.Files.*

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 28 Oct 2014 08:03 UTC**
SimulationX supports the vendor-specific `annotation(__iti_Impure=true);`. Do you think we should add such annotations to the MSL?

modelica-trac-importer, 2017-01-14:
**Comment by sjoelund&#46;se on 28 Oct 2014 08:08 UTC**
Vendor-specific annotations are forbidden in the MSL. Otherwise, the MSL would use `__OpenModelica_Impure=true`, `__iti_Impure=true`, `..._Impure=true`, ...

For MSL 3.3 you have the impure keyword which can be used without requiring annotations.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 28 Oct 2014 08:16 UTC**
I would not have asked if there are no vendor-specific annotations in the current MSL.

Also, there is no MSL 3.3 yet.

modelica-trac-importer, 2017-01-14:
**Comment by sjoelund&#46;se on 28 Oct 2014 08:25 UTC**
Replying to [comment:5 beutlich]:
> I would not have asked if there are no vendor-specific annotations in the current MSL.

There are a total of 3 plus some `__Dymola_commands`. 2 of them can be fixed (using choicesAllMatching=true).

> Also, there is no MSL 3.3 yet.

Bring this up during the design meeting. The 3.2.1 maintenance release has been taking its time. And there was talk to make 3.2.2 the next release. You could make a point for making 3.3.0 the next release instead (since you want to use the impure keyword).

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Dec 2014 07:01 UTC**
Ticket retargeted after milestone closed

beutlich, 2017-06-20:
@HansOlsson FYI: readTableData was removed due to #1899 (for MSL > v3.2.2).

beutlich, 2017-08-26:
@HansOlsson FYI: readTableData was reintroduced by 4b5055c87478dd0babaf58acda15237715ef91d0 in order to ensure backward compatibility with older models, see e.g., https://stackoverflow.com/a/45895857/8520615.

PR #2858 (2020-01-21): Declare external functions as pure/impure
Fix #1478.

@HansOlsson @MartinOtter Please review thoroughly.

For me, one open issue is the "dummy" table hack as for instance documented in https://github.com/modelica/ModelicaStandardLibrary/blob/ec70d1f10887371f11f14cb1abb696b60ee8cd6f/Modelica/Blocks/Tables.mo#L1320-L1330. This is a pure external function, but must not removed during optimization due to the intended side-effect to include the usertab header file. Is the recommendation to mark it as impure then?
closed 2020-01-21 by beutlich