modelica/ModelicaStandardLibrary #1480: Duplication of newlines for ModelicaMessage etc. in FMUs

modelica-trac-importer, 2017-01-14:
**Reported by beutlich on 13 May 2014 08:15 UTC**
In [FMI#158](https://trac.fmi-standard.org/ticket/158) it is said that ModelicaUtility functions (ModelicaMessage, ModelicaFormatMessage, ModelicaVFormatMessage, ModelicaError, ModelicaFormatError and ModelicaVFormatError) duplicate the newlines if logged as FMU. This is due the case
 * that every ModelicaUtility function (of the MSL) has a trailing newline and
 * the FMU logger adds a trailing newline.
So I wonder why every ModelicaUtility function (of the MSL) is called with a message string that ends with a newline. As jakesson pointed out this is not specified everywhere. Should it be added to the MLS or removed from the MSL?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1480

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Dec 2014 07:01 UTC**
Ticket retargeted after milestone closed

beutlich, 2020-05-04:
Clsosing as discussion did not happen.

closed 2020-05-04 by beutlich