modelica/ModelicaStandardLibrary #1489: Requirements and Review of cndidates for MSL

modelica-trac-importer, 2017-01-14:
**Reported by ahaumer on 14 May 2014 16:06 UTC**
As discussed during today's MAP-Lib meeting, I want to start a discussion how to have a more formal handling of candidates for inclusion to MSL:

1. We need a check-list of requirements a candidate has to fulfill
Discussion and suggestions are welcome - further discussion at te next Design Meeting in Dresden.

2. I suggest a formal review process:
2.1 A new sub-library shall be stored at svn (Modelica\projects\Modelica\branches\development.
2.2 Members from MAP-Lib which are experts in the area the library covers have to test the library and fill in a review form (see attachment)  within one month. 
2.3 Based on the reviews, MAP-Lib decides (in a web / phone conference) whether the library is ready to me moved to Modelica\trunk or some further work is necessary or the library is not suitable for MSL. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1489

modelica-trac-importer, 2017-01-14:
**Comment by pharman on 14 May 2014 16:31 UTC**
Regarding 2.2: I don't see that the reviewers need to be from MAP-Lib (or in fact that it is feasible to find experts for every domain already in MAP-Lib).

Regarding the evaluation form: one thing not mentioned is what tool(s) the reviewer has used. It should be a requirement that any library being adopted should work in multiple tools and not depend on language features that are still subject to discussions here.

modelica-trac-importer, 2017-01-14:
**Comment by hubertus on 14 May 2014 19:38 UTC**
I agree with Pete on 2.2. 

Also, it would be much better if the evaluation form would be in an editable format, or maybe even a shared google docs document to get input from the group.

modelica-trac-importer, 2017-01-14:
**Comment by msasena on 14 May 2014 19:50 UTC**
A few thought starters for a check-list of requirements:
 - Conformance to some (to be specified) style guideline.  This could include some simple, but useful guidelines like a specification for class / instance name capitalization, proper use of commenting, etc.
 - At least one simulatable model to test each component.  Example models, as per the Modelica Test and Compliance Test Suite, should include experiments annotations.
 - A set of reference results along with documentation on the test environment (Modelica tool / version used, solver selection / settings, etc.)
 - Inclusion of a User's Guide and / or Documentation package

modelica-trac-importer, 2017-01-14:
**Comment by wischhusen on 15 May 2014 07:15 UTC**
There are already some guidelines detailed in the MSL package:

Modelica.UsersGuide.Conventions

Furthermore, one need to check for tool-dependent source code (e.g. in annotations .... or certain functions), which should be avoided.

modelica-trac-importer, 2017-01-14:
**Modified by ahaumer on 18 Jun 2014 08:35 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Dec 2014 07:01 UTC**
Ticket retargeted after milestone closed

beutlich, 2019-08-04:
There are already some relevant points here. @dietmarw Should we create wiki for it, finally?

beutlich, 2019-08-09:
> @dietmarw Should we create wiki for it, finally?

Thanks. There now is a bullet list available in the [wiki](https://github.com/modelica/ModelicaStandardLibrary/wiki/Requirements-and-Review-of-candidates-for-MSL).

closed 2019-08-09 by beutlich