modelica/ModelicaStandardLibrary #1490: Fixed release cycle for MSL

modelica-trac-importer, 2017-01-14:
**Reported by ahaumer on 14 May 2014 16:10 UTC**
As mentioned during today's MAP-Lib meeting, I want to start a discussion about pros and cons of a fixed release cycle. If we agree, we could try to fix the dates (spring and autumn?).
Please give your comments.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1490

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 18 Jun 2014 15:28 UTC**
I do not have a strong opinion if there should be a maintenance/feature release every 6, 12 or 18 months as long as it is somehow predicatable which it is currently is not. I even thought about having hot fixes for severe bugs but this does not seem to be realistic. Therefore I could imagine a maintenance release twice a year and new-feature releases whenever it is ready or required. Reasoning from the past it does not seem to be realistic to have a fixed release cycle for feature releases since it takes many iterations to get the new feature to a mature state.

modelica-trac-importer, 2017-01-14:
**Comment by sjoelund&#x2e;se on 19 Jun 2014 06:55 UTC**
For OpenModelica, the release schedule is not that important for the maintenance branches. We already pull the latest fixes from the maintenance branches and update them nightly. We also ship a version called `Modelica trunk` (it is the same as the svn trunk, except the version string is set to `trunk` to avoid users setting a uses annotation to an unreleased MSL version by accident).

I would say a regular 0.0.1 or 0.1 release (depending on advances in the language spec) would be nice to add new models to the library at a regular pace. That would allow for the library officers to make some planning before the release.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Dec 2014 07:01 UTC**
Ticket retargeted after milestone closed

beutlich, 2018-03-23:
Resolved by the new project rules.

closed 2018-03-23 by beutlich