modelica/ModelicaStandardLibrary #1491: Strategic MSL development

modelica-trac-importer, 2017-01-14:
**Reported by ahaumer on 14 May 2014 16:12 UTC**
This could be a long-term ticket for discussions about the strategic direction of future enhancements and extensions of the MSL, not driven by actual developments, but driven by strategic considerations.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1491

modelica-trac-importer, 2017-01-14:
**Modified by ahaumer on 18 Jun 2014 08:36 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by ahaumer on 18 Jun 2014 08:50 UTC**
New contributions (models and examples as well as sub-libraries) are mostly driven by personal ideas, recent projects or daily work. A strategic point of view could ask: Which new models, sub-libraries and models do we need to increase interest of new users in usig Modelica? I just want to name a few examples:
- optical domain
- acoustic domain
- chemistry (mol flow)
- economic models
- opto-electronics
- thermo-electric
- photovoltaic
- solar-thermal (depends on weather data!)
- wind energy (depends on weather data!)

On the other hand, the Modelica Standard Library should contain models that are commonly / often used and the models have to be maintained over the years. We could also offer free additional libraries: It would be nice to have a list of such libraries (e.g. in the library browser of a tool), and the user could decide these libraries to try them (i.e. download them from w.g. github) by pressing a button.

modelica-trac-importer, 2017-01-14:
**Modified by ahaumer on 18 Jun 2014 08:51 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Jun 2014 12:43 UTC**
Before getting ahead of us here the MA should first really look into solving the underlying problem of library management **before** deciding on adding more and more things to the MSL which could just as well be a (MA maintained) library outside the MSL but easy to use. Just like Toni says with click of a button. 

This is not a new problem and we really ought to put some energy into solving this **first**. Points to start with are: #89 (6 years old !!!), #132, and #573. Due to apparent lack of interest from the MA side Mike Tiller and I already started providing a solution that is not perfect yet but already works for standard conform 3rd-party libraries: https://github.com/xogeny/impact presented at the last Modelica'2014 conference (http://dx.doi.org/10.3384/ecp14096543).

So for me it does not make sense to add loads of stuff to the MSL now and then later find a solution for dependency management only to then realise that we can not simply remove things from the MSL again that should not have been placed there in the first place. It really ought to be the other way round.

modelica-trac-importer, 2017-01-14:
**Comment by otter on 23 Jun 2014 12:24 UTC**
Replying to [comment:4 dietmarw]:
> Before getting ahead of us here the MA should first really look into solving the underlying problem of library management **before** deciding on adding more and more things to the MSL which could [...]

These are orthogonal issues.

Its very good that Mike and you started with a concrete Modelica package manager. I will put this on the agenda of the next design meeting to further discuss it. Especially, there are open issues such as:
- Version numbering: In the Modelica specification there is a different version numbering scheme as you use it (e.g. the patch level is defined by a separate annotation and is not part of the version number; beta and alpha releases are defined with separate keywords, MSL is using 4 numbers and not 3 for good reasons). Before going on, one has to get a common agreement on the version numbering and this might mean to improve the Modelica specification so that automated processing of version numbers becomes simpler.
- What happens if two libraries A and B use a third library C and both A and B are downloaded? Most likely, C is downloaded twice? Obviously, one has to standardize where downloaded packages are stored and the package manager needs to know which packages are already installed on the machine where the libraries are downloaded.
- The tool vendors need to support the scheme, in order that the standard user gets a graphical user interface. So, the question is how to get the tool vendors into it? Tool vendors that are not based on the Python environment most likely need to re-implement the package manager in their own environment (because they cannot expect that a particular Python environment is installed on the users machine).

You are right that one needs to be careful what to include in to the Modelica Standard Library (MSL) and what not. Note, the MSL has the big advantage that it is continuously maintained from one organization (the Modelica Association). Such a stability is impossible to achieve with a distributed solution where a package manager is needed (and in fact, my own experience with Python shows that package managers are a nice feature, but the whole process is not really robust).

To my understanding, the current plan for new features in MSL 3.2.2 is anyway useful, and it makes no sense to have the new functionality as separate libraries outside of MSL (e.g. Noise is basically one block, that is naturally in Modelica.Blocks; there are extensions to the FluxTubes library, PlanarMechanics is naturally a 2D mechanical version as an extension to the already available 3D and 1D sublibraries). So, it would be not useful to mix this with a package manager discussion.

Yet another orthogonal issue is the release cycle of MSL. As proposed, it might improve the situation with MSL, if every 3 or 6 months an official maintenance version of MSL is released by MA.

modelica-trac-importer, 2017-01-14:
**Comment by leo&#x2e;gall on 23 Jun 2014 13:35 UTC**
Martin's comment gives an good overview of the current topics of MSL planning.
I collected a list of open tickets, so that we don't need to spoil this ticket with discussions about some of the orthogonal issues: 

Currently discussed MSL additions:
- DLR Noise Lib for inclusion into MSL #1514
- New ND-table based on SDF to be integrated in MSL #1516
- PlanarMultiBody ought to be included into MSL 3.3 #464
- New power converters library to be integrated in MSL #1416
- New quasistatic flux tubes library to be integrated in MSL #1515
- Extensions to FluxTubes by Thomas Bödrich and colleagues (magnetic hysteresis, from CleanSky project, no ticket yet)

Review and release process:
- Requirements and Review of candidates for MSL #1489
- Fixed release cycle for MSL #1490

modelica-trac-importer, 2017-01-14:
**Comment by mtiller on 23 Jun 2014 13:37 UTC**
Martin, let me address some of your points.

Regarding version numbering, we have chosen to use Semantic Versions.  This allows to use an "off the shelf" set of semantics for the version numbers which makes life much easier.  Note that semantic versioning allows for "extra" information (e.g. pre-release information like "beta1" or build information).  I don't recall all the details of the MSL versioning off the top of my head, but I think it would be very useful to adopt the semantic versioning scheme (or at least identify why it cannot be used and perhaps address that).

Our package manager does not download two copies of a library that is a dependency of other libraries.  Instead, it identifies the set of versions that are consistent with all dependencies or throws an error if it cannot identify such a common version.  Because of the current versioning scheme, this calculation is trivial.  But if/when Modelica adopts a more flexible version annotation that allows the specification of multiple compatible versions, this logic will need to change.  But in principle, this is how we felt the package manager should work.

You say that tool vendors are not using Python.  But we don't require them to use Python.  Python is used by the tool that builds the library index files and by the command line tool.  But **all information** regarding libraries, versions, compatibility, \*etc\*. is provided in a single index file that a tool can easily download.  So the point is that we go to the trouble of building the complete index of packages, versions, download links, etc.  All they need to do is read it and implement some very simple algorithms for identifying cross compatibility.

modelica-trac-importer, 2017-01-14:
**Comment by dag on 24 Jun 2014 07:49 UTC**
Important topic, but I think we need to fix the issue with multiple compatible versions at the same time. Otherwise we would have to re-issue the package manager specification.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Dec 2014 07:01 UTC**
Ticket retargeted after milestone closed

beutlich, 2019-08-04:
Closing, since brain-stormin stopped about 4.5 years ago.

closed 2019-08-04 by beutlich