modelica/ModelicaStandardLibrary #1493: Mechanics.Rotational.Interfaces.PartialFriction.w_small

modelica-trac-importer, 2017-01-14:
**Reported by ahaumer on 15 May 2014 15:27 UTC**
Default value for w_small=1e10 doesn't seem to be really small ...

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/1493

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 15 May 2014 19:14 UTC**
I guess this is a duplicate of #701.

modelica-trac-importer, 2017-01-14:
**Comment by otter on 20 Jun 2014 14:35 UTC**
Replying to [comment:1 beutlich]:
> I guess this is a duplicate of #701.

Yes. Here is a copy of the essential description of #701:

 w_small was introduced due to a bug report, where at zero velocity a reinit was triggered and then the friction handling logic is no longer correct. This was a hack to at least give the possibility to fix this manually for a specific model. On the other hand, introducing w_small in general gave other problems for connected friction elements. Therefore, it was decided to introduce w_small but with a large value. If the special case with reinit occurs, then the user has to set w_small to a small value.

 This is of course not nice, but there is currently no better idea how to handle it (this would require to remove the hack "reinit" and introduce proper handling of Dirac impulses).

 The description text of variable w_small seems to be also correct ("set to low value only if such impulses can occur").

modelica-trac-importer, 2017-01-14:
**Modified by beutlich on 6 Jan 2015 21:35 UTC**

AHaumer, 2019-12-08:
@tobolar 
There's a rather old ticket regarding the issue with w_small.
I seems to be a wrong parameter description and incorrect or missing documentation. 
Would you please talk to Martin Otter and take care of that?

tobolar, 2019-12-09:
@AHaumer I can do this. But is there any new/particular update regarding https://github.com/modelica/ModelicaStandardLibrary/issues/1493#issuecomment-272661458?

MartinOtter, 2019-12-10:
> 
> 
> @tobolar
> There's a rather old ticket regarding the issue with w_small.
> I seems to be a wrong parameter description and incorrect or missing documentation.
> Would you please talk to Martin Otter and take care of that?

Nothing has changed. Please, read my previous comment. Please, do not make a change because this might break existing models.

AHaumer, 2019-12-10:
Then PLEASE adapt the documentation - w_small=1E10 is total misleading!

MartinOtter, 2019-12-11:
parameter w_small is in the Advanced tab and has the following description text:
```
parameter SI.AngularVelocity w_small=1.0e10
    "Relative angular velocity near to zero if jumps due to a reinit(..) of the velocity can occur 
    (set to low value only if such impulses can occur)"
```
Seems to me that this descriptiontext is fine. Alternatively, my longer description from above could be copied into the documentation of PartialFriction

tobolar, 2019-12-13:
@MartinOtter 
> Alternatively, my longer description from above could be copied into the documentation of PartialFriction

I would prepare a text proposal.

PR #3633 (2020-10-29): Describe the effect of w_small in friction elements 
closes #1493
closed 2020-10-29 by beutlich