modelica/ModelicaStandardLibrary #282: Migrate FluidHeatFlow to stream-concept

modelica-trac-importer, 2017-01-14:
**Reported by AHaumer on 12 Feb 2010 14:31 UTC**
What are the consequences?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/282

modelica-trac-importer, 2017-01-14:
**Comment by otter on 16 Jul 2010 11:51 UTC**
It is important to change this soon. This is a non-backward compatible change that cannot be handled with conversion scripts, since too many changes. The only meaningful was is:

 1. Move FluidHeatFlow to library ObsoleteModelica.
 2. Provide a conversion script that changes user models from
    Modelica.Thermal.FluidHeatFlow to ObsoleteModelica.Thermal.FluidHeatFlow. 
 3. Provide a new FluidHeatFlow where the stream operator is used

modelica-trac-importer, 2017-01-14:
**Modified by ahaumer on 2 Mar 2012 09:04 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2019-07-15:
@AHaumer This is of low priority though it was considered to be dealt with rather soon. I also doubt that you intend to move the current FluidHeatFlow to ObsoleteModelica. Thus, after all, is this still a feasible goal for MSL v4.0.0 with about 6 months before feature freeze?

AHaumer, 2019-07-15:
In the meantime I changed my mind: FluidHeatFlow should keep using semiLinear instead of the stream concept. Therefore closing the ticket.

closed 2019-07-15 by AHaumer