modelica/ModelicaStandardLibrary #283: Improve/fix documentation of the whole ModelicaStandardLibrary

modelica-trac-importer, 2017-01-14:
**Reported by AHaumer on 12 Feb 2010 14:34 UTC**
When a conversion script is allowed, some naming styles should be adapated:
Naming of functions (starting with a lower case letter)
Naming of connectors withput '_' (e.g. Flange_a -> FlangeA)

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/283

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 12 Feb 2010 15:59 UTC**
Actually names like `Flange_a` are perfectly legal and this is now also properly documented (wasn't so before) since c804483fa05cbaaf33f6a4f14065ee9185570d4c
So the underscores could stay there. In general given the more detailed rule from the current version of the User's Guide:
 1.  Class and instance names are usually written in upper and lower case letters, e.g., "ElectricCurrent". An underscore may be used in names. However, it has to be taken into account that the last underscore in a name might indicate that the following characters are rendered as a subscript. Example: "pin_a" may be rendered as "pin<sub>a</sub>".

modelica-trac-importer, 2017-01-14:
**Comment by otter on 30 Mar 2013 14:02 UTC**
Unclear what should be performed. Seems this ticket is no longer valid.

modelica-trac-importer, 2017-01-14:
**Changelog removed by otter on 30 Mar 2013 14:02 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Apr 2013 12:32 UTC**
How can this be invalid. The Reporter made this a ticket with a specific task and also a Milestone that is not yet due. Martin, just because it is unclear to **you** what should be performed does not mean this ticket is invalid. This is something that can be decided by the MSL group once we come to the next MSL version with conversion script.

modelica-trac-importer, 2017-01-14:
**Comment by otter on 2 Apr 2013 09:30 UTC**
Replying to [comment:4 dietmarw]:

> How can this be invalid. The Reporter made this a ticket with a specific task and also a Milestone that is not yet due. Martin, just because it is unclear to **you** what should be performed does [...]

The given examples for a conversion script are wrong:

 * All functions that I am  aware off start with a lower case letter. If there are functions with an  upper case letter, it should be listed and then one can decide whether  it makes sense to keep them uppercase or to change to lower case.
 * As already stated in a comment the "flange_a" should be kept and it is according to the naming convention (it should not be flangeA).

It is much too dangerous to make an automatic conversion according to style guidelines, because for most style guides there are also often exceptions. The only tool support hat can be given is to list classes that do not follow a guideline and then the user decides whether this is o.k. or not.

To summarize, the current ticket description is not useful and therefore the ticket is closed. Please, re-open the ticket only, if there is a new, meaningful ticket description (a generic ticket description: "use a conversion script to enforce guidelines" is not meaningful as explained).

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 2 Apr 2013 10:41 UTC**
Still the issue is a valid one, and yes some more specific description of what is needed is in order but that's why we have tickets. So we can enhance and discuss what needs to be done. No-one is going to look at any closed tickets again in one ore two years time. Nobody said something about a dangerous automatic conversion.

So I'm re-opening this tickets on the grounds that this is not forgotten and can be investigated in more detail and the MAP LIB **group** can decide what to do about it. So please let this open until the MAP LIB group had the time to look into this once the next MSL with conversion script is due. 

BTW, I've improved the title so it reflects what this ticket is really about.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2017-02-13:
List of MSL 3.2.2 functions that start with an uppercase letter:

| File(line) | Function |
| -- | -- |
| Modelica\Icons.mo(489) | Function |
| Modelica\SIunits.mo(239) | Conversion |
| Modelica\SIunits.mo(934) | ConversionIcon |
| Modelica\Blocks\Continuous.mo(2088) | CriticalDamping |
| Modelica\Blocks\Continuous.mo(2124) | Bessel |
| Modelica\Blocks\Continuous.mo(2165) | Butterworth |
| Modelica\Blocks\Continuous.mo(2219) | ChebyshevI |
| Modelica\Blocks\Continuous.mo(2989) | BesselBaseCoefficients |
| Modelica\Electrical\Machines.mo(10809) | ToSpacePhasor |
| Modelica\Electrical\Machines.mo(10831) | FromSpacePhasor |
| Modelica\Electrical\Machines.mo(10854) | Rotator |
| Modelica\Electrical\Machines.mo(10878) | ToPolar |
| Modelica\Electrical\Machines.mo(10904) | FromPolar |
| Modelica\Electrical\Analog\Basic.mo(1402) | FCNiout_limit |
| Modelica\Electrical\Analog\Basic.mo(1425) | FCNq_sum_limit |
| Modelica\Fluid\Dissipation.mo(10139) | TwoPhaseMultiplierFriedel |
| Modelica\Fluid\Dissipation.mo(10240) | TwoPhaseMultiplierChisholm |
| Modelica\Fluid\Dissipation.mo(10326) | TwoPhaseDensity |
| Modelica\Fluid\Dissipation.mo(10413) | VoidFraction |
| Modelica\Fluid\Dissipation.mo(10461) | SlipRatio |
| Modelica\Fluid\Dissipation.mo(10741) | CubicInterpolation_DP |
| Modelica\Fluid\Dissipation.mo(10780) | CubicInterpolation_MFLOW |
| Modelica\Fluid\Dissipation.mo(10817) | LambertW |
| Modelica\Fluid\Dissipation.mo(10863) | LambertWIter |
| Modelica\Fluid\Dissipation.mo(10922) | PrandtlNumber |
| Modelica\Fluid\Dissipation.mo(10939) | ReynoldsNumber |
| Modelica\Fluid\Dissipation.mo(10965) | SmoothPower |
| Modelica\Fluid\Dissipation.mo(11022) | SmoothPower_der |
| Modelica\Fluid\Dissipation.mo(11051) | Stepsmoother |
| Modelica\Fluid\Dissipation.mo(11106) | Stepsmoother_der |
| Modelica\Fluid\Pipes.mo(1705) | ReynoldsNumber |
| Modelica\Fluid\Pipes.mo(1745) | ReynoldsNumber_m_flow |
| Modelica\Fluid\Pipes.mo(1770) | NusseltNumber |
| Modelica\Fluid\Valves.mo(103) | FlCharacteristic |
| Modelica\Math\package.mo(1616) | LU |
| Modelica\Math\package.mo(1713) | LU_solve |
| Modelica\Math\package.mo(1799) | LU_solve2 |
| Modelica\Math\package.mo(2100) | QR |
| Modelica\Math\package.mo(11824) | AxisLeft |
| Modelica\Math\package.mo(11864) | AxisCenter |
| Modelica\Mechanics\Translational.mo(1699) | GenerateStribeckFrictionTable |
| Modelica\Media\Incompressible.mo(477) | T_ph |
| Modelica\Media\Incompressible.mo(503) | T_ps |
| Modelica\Media\package.mo(5497) | T_ph |
| Modelica\Media\package.mo(5507) | T_ps |
| Modelica\Media\package.mo(8075) | TwoPhaseToProps_ph |
| Modelica\Media\package.mo(8105) | TwoPhaseToProps_dT |
| Modelica\Media\package.mo(8740) | Helmholtz_ph |
| Modelica\Media\package.mo(8758) | Helmholtz_pT |
| Modelica\Media\package.mo(8770) | Helmholtz_ps |
| Modelica\Media\package.mo(8878) | Gibbs2_ph |
| Modelica\Media\package.mo(8897) | Gibbs2_dT |
| Modelica\Media\package.mo(8910) | Gibbs2_ps |
| Modelica\Media\R134a.mo(75) | CubicSplineDerEval |
| Modelica\Media\R134a.mo(86) | CubicSplineEval |
| Modelica\Media\R134a.mo(129) | FindInterval |
| Modelica\Media\R134a.mo(2295) | R134a_liqofdT |
| Modelica\Media\R134a.mo(2320) | R134a_vapofdT |
| Modelica\Media\R134a.mo(2392) | T_ph_der |
| Modelica\Media\R134a.mo(2416) | T_props_ph |
| Modelica\Media\Air\MoistAir.mo(202) | Xsaturation |
| Modelica\Media\Air\MoistAir.mo(561) | HeatCapacityOfWater |
| Modelica\Media\Air\MoistAir.mo(721) | T_phX |
| Modelica\Media\Air\MoistAir.mo(1285) | T_psX |
| Modelica\Media\Air\ReferenceAir.mo(596) | Helmholtz |
| Modelica\Media\Air\ReferenceAir.mo(1047) | T_props_ps |
[... 32 lines cut ...]
| Modelica\Media\Air\ReferenceMoistAir.mo(1666) | V2_der |
| Modelica\Media\Air\ReferenceMoistAir.mo(1676) | V3_der |
| Modelica\Media\Air\ReferenceMoistAir.mo(1685) | V4_der |
| Modelica\Media\Air\ReferenceMoistAir.mo(1695) | V5_der |
| Modelica\Media\Air\ReferenceMoistAir.mo(1704) | V6_der |
| Modelica\Media\Air\ReferenceMoistAir.mo(2318) | Tsat |
| Modelica\Media\Air\ReferenceMoistAir.mo(2383) | Tsat_der |
| Modelica\Media\Air\ReferenceMoistAir.mo(2540) | Gibbs |
| Modelica\Media\Air\ReferenceMoistAir.mo(2747) | Tsub |
| Modelica\Media\Air\ReferenceMoistAir.mo(2753) | Tsub_res |
| Modelica\Media\IdealGases\Common\package.mo(353) | T_h |
| Modelica\Media\IdealGases\Common\package.mo(382) | T_ps |
| Modelica\Media\IdealGases\Common\package.mo(848) | MixEntropy |
| Modelica\Media\IdealGases\Common\package.mo(1311) | T_hX |
| Modelica\Media\IdealGases\Common\package.mo(1348) | T_psX |
| Modelica\Media\Water\IF97_Utilities.mo(3315) | T3a_ph |
| Modelica\Media\Water\IF97_Utilities.mo(3357) | T3b_ph |
| Modelica\Media\Water\IF97_Utilities.mo(3487) | T3a_ps |
| Modelica\Media\Water\IF97_Utilities.mo(3532) | T3b_ps |
| Modelica\Media\Water\IF97_Utilities.mo(6617) | T_props_ps |
| Modelica\Media\Water\IF97_Utilities.mo(6629) | T_ps |
| Modelica\Media\Water\IF97_Utilities.mo(6771) | T_props_ph |
| Modelica\Media\Water\IF97_Utilities.mo(6786) | T_ph |
| Modelica\Media\Water\IF97_Utilities.mo(6806) | T_ph_der |
| Modelica\Utilities\Internal.mo(98) | PartialLoadResource |

HansOlsson, 2017-02-13:
To me it seems clear that we should weaken the guideline. 
We should not rename `T_ps` just because the guidelines says so, and not change `LU `to `lU `(perhaps it should be `lu `- but I don't see a major issue with keeping `LU`).

Specifically we should weaken the guideline:

1. Class and instance names are usually written in upper and lower case letters, e.g., "ElectricCurrent". An underscore may be used in names. However, it has to be taken into account that the last underscore in a name might indicate that the following characters are rendered as a subscript. Example: "pin_a" may be rendered as "pina". 
2. Class names start always with an upper case letter, with the exception of functions, that start with a lower case letter. 
3. Instance names, i.e., names of component instances and of variables (with the exception of constants), start usually with a lower case letter with only a few exceptions if this is common sense (such as T for a temperature variable). 

by changing (2) to at least say "Class names start always with an upper case letter, with the exception of functions, that usually start with a lower case letter." We could even add the same exceptions as in (3).

Additionally if we have multiple words in the name, like `PartialLoadResource`, it to me looks odd if the first name isn't capitalized: `partialLoadResource`. Using underscore to separate these words does not make sense if we use the sub-scripting logic for underscores.

beutlich, 2017-02-13:
I do not see the the issue with lower camel case, e.g., `partialLoadResource` is still easily readable. Same holds for `cubicSplineEval`, `generateStribeckFrictionTable` or `toPolar` or other functions. Function `LU` could be renamed to `solveLU`, and `T_h` to `getT_h`. Generally, functions perform some actions, and if the verb is used as initial part of the compound name like `getFileName`, `evalCubicSpline` or `displayErrorMessage` it would be pretty consistent.

HansOlsson, 2017-02-14:
I believe we need to consider it more; if we want to have a convention that multi-word function names shall be `verbNoun` we should document that and use it consistently. Renaming  `h_pTX` to `getT_phX`  in order to follow some convention, and keep `h_pTX` and `pressure` will just be confusing - renaming all seems excessive. We also need to see how these look in the actual code.

Note that we have an exception in the conventions for variables allowing `T` and `T1` as temperature variables - to me it looks logical to extend that to functions as well.

BTW: Regarding `CubicSplineEval` or `evalCubicSpline` I would prefer that evaluate is spelled out fully.

beutlich, 2017-02-14:
> "There are only two hard problems in Computer Science:
   cache invalidation and naming things."
-- Phil Karlton

HansOlsson, 2017-02-15:
Additionally I thought more about `PartialLoadResource` since the name looked wrong in some way - but I didn't realize why at first. The reason is that it sounds to me as though it is a "partial load", and thus I would have preferred something like `loadResource_interface` - focusing first on the fact that it is for loading a resource.

dietmarw, 2017-02-15:
Yes in general I also find it awkward having to repeat information stored in a model in its name. `partial` being one example. I rather like to have all `partial` models extend from a easily identifiable icon then to have those clumsy names. Calling it "interface" or "frame" or "base", etc. is probably of more use.

beutlich, 2019-10-05:
Not feasible for MSL v4.0.0 and of low prio -> removing milestone.