modelica/ModelicaStandardLibrary #285: Improve/fix documentation of Modelica.Mechanics.Multibody

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 12 Feb 2010 17:09 UTC**
See attached [raw-attachment:Modelica.Mechanics.MultiBody_StyleCheckLog.html StyleCheckLog.html] for indicators what needs to be improved/fixed.

Mind warnings about underscores can be ignored because of c804483fa05cbaaf33f6a4f14065ee9185570d4c.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/285

modelica-trac-importer, 2017-01-14:
**Modified by otter on 5 Aug 2010 21:38 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 6 Sep 2012 11:57 UTC**
These tickets don't rely on conversion scripts, hence rescheduling for the next major release

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 29 Apr 2014 06:32 UTC**
@dietmarw: Should Milestone be set back to MSL_withConversionScript? Additionally, the attached file is missing.

modelica-trac-importer, 2017-01-14:
**Changelog removed by beutlich on 29 Apr 2014 06:32 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 29 Apr 2014 06:50 UTC**
Well part of it can already be fixed without conversion script. Also recovered the attachment.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 29 Apr 2014 07:28 UTC**
Can you regenerate that style check log file on current trunk? Most of the "Component description string beginning with whitespace for" should be gone by now.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 29 Apr 2014 07:51 UTC**
Sorry **I** can't since I no longer have access to the model management option of Dymola. Somebody else might be able to run it though.

modelica-trac-importer, 2017-01-14:
**Comment by otter on 13 Dec 2015 12:53 UTC**
Partially fixed in 9f761f8b6cd0289dd3f9043ba18bf685e124e467 by providing missing documentation and description strings.

Do not have more time to work on this now. Therefore, shifted the milestone.

modelica-trac-importer, 2017-01-14:
**Comment by anonymous on 26 Dec 2016 22:36 UTC**
FYI: The Model Management package is available license-free in Dymola 2017 FD 01.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 2 Jan 2017 14:35 UTC**
Replying to [comment:8 anonymous]:
> FYI: The Model Management package is available license-free in Dymola 2017 FD 01.

Recreated [Modelica.Mechanics.MultiBody_StyleCheckLog.html](https://cdn.rawgit.com/tbeu/0c4e89e5c86a1e6b813cd08a8ed8bc7b/raw/4b63e918e2a433fee096428c5754f1552df40bcd/Modelica.Mechanics.MultiBody_StyleCheckLog.html) based on MSL 3.2.2+build.3.

tobolar, 2018-01-11:
Partially solved by f9fdb3b.

tobolar, 2018-01-17:
I have finalized missing documentation now. Anybody wants to check it?

beutlich, 2018-01-17:
Recreated [Modelica.Mechanics.MultiBody_StyleCheckLog.html](https://cdn.rawgit.com/tbeu/0c4e89e5c86a1e6b813cd08a8ed8bc7b/raw/4b63e918e2a433fee096428c5754f1552df40bcd/Modelica.Mechanics.MultiBody_StyleCheckLog.html) based on 3fe8775fdc3ca4eeec4d5dfdba8d89a98acb017b.

beutlich, 2018-01-17:
Edit: Updated link above.

tobolar, 2018-01-18:
@beutlich Thank's.
The documentation is complete now and we can close this issue. I' going to open another one concerning "Bad class names".

beutlich, 2018-02-26:
Is it really necessary to have a duplicated html comment in 

https://github.com/modelica/Modelica/blob/cf964a06a7665927c64392c95b95dfd5e4caa5f9/Modelica/Mechanics/MultiBody/Sensors.mo#L3567-L3568

and

https://github.com/modelica/Modelica/blob/cf964a06a7665927c64392c95b95dfd5e4caa5f9/Modelica/Mechanics/MultiBody/Sensors.mo#L3579-L3580

? Furthermore, the typo domentation -> documentation should be fixed.

commit 9f761f8b (2015-12-13): Partially fixes #285 (Improve/fix documentation of Modelica.Mechanics.Multibody)
- Added missing documentation / description strings to MultiBody library

git-svn-id: https://svn.modelica.org/projects/Modelica/trunk@8734 7ce873d0-865f-4ce7-a662-4bb36ea78beb
commit 2bf17f02 (2018-02-27): Deleted class documentation placeholders defined twice by mistake

Close #285
closed 2018-02-27 by tobolar