modelica/ModelicaStandardLibrary #287: Refine temperature dependency modelling

modelica-trac-importer, 2017-01-14:
**Reported by AHaumer on 12 Feb 2010 18:29 UTC**
The following can only be done when a conversion script is allowed, and holds not only for Electrical but also for Modelica.Thermal.HeatTransfer.Sources.Fixed/PrescribedHeatFlow:

1. We should choose a consistent default / start value for all reference temperatures of 20 degC = 293.15 K (27 degC = 300.15 K has no meaning to me).

2. We should define the LinearTemperatureCoefficient alpha consistently at 20 degC = 293.15 K (not at reference temperature).
Such a material coefficient can be found in material databases defined at 20 degC, not at other temperatures. I have attached a document that shows how alpha can be converted from 20 degC to an other reference temperature.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/287

modelica-trac-importer, 2017-01-14:
**Modified by AHaumer on 12 Feb 2010 18:32 UTC**

modelica-trac-importer, 2017-01-14:
**Modified by ahaumer on 20 Dec 2016 16:49 UTC**

modelica-trac-importer, 2017-01-14:
**Changelog removed by ahaumer on 20 Dec 2016 16:49 UTC**

beutlich, 2017-01-30:
Restored attachment LinearTemperatureDependency.pdf.

christophclauss, 2019-08-04:
The remaining time to MSL4.0.0 seems to be too short for me. I'd like postpone this issue.