modelica/ModelicaStandardLibrary #289: Machines airGap

modelica-trac-importer, 2017-01-14:
**Reported by AHaumer on 14 Feb 2010 10:25 UTC**
When a conversion script is allowed, rename the airGaps in the machine models without a trailing character to have the chance to move the airGap to the partial machine model and make it replaceable.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/289

modelica-trac-importer, 2017-01-14:
**Modified by ahaumer on 2 Mar 2012 09:05 UTC**

modelica-trac-importer, 2017-01-14:
**Modified by ahaumer on 20 Dec 2016 16:41 UTC**

modelica-trac-importer, 2017-01-14:
**Changelog removed by ahaumer on 20 Dec 2016 16:41 UTC**

AHaumer, 2019-07-15:
After PR #3045 (thanks to @beutlich ) is merged, we plan to move Machines.BasicMachines.Components.{PartialAirGap, PartialAirGapDC, PartialCore} to Machines.Interfaces.
@christiankral should we have a sub-library BaseClasses, keeping in Interfaces only {SpacePhasor, FlangeSupport, ThermalPort$, PowerBalance$}?

PR #3045 (2019-07-16): Rename airGapS/airGapR components
Close #289 (since it only is on renaming and not yet on restructuring).
closed 2019-07-16 by beutlich