modelica/ModelicaStandardLibrary #291: Better  naming conventions for Electrical, Magnetic, Mechanics

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 15 Feb 2010 13:48 UTC**
As a non-native speaker it kind of strikes me that the current naming convention in the MSL is:
 * `Electrical`
 * `Magnetic`
 * `Mechanics`

For me it sounds far more stringent to call those "domain" libaries:
 1. "cal"-variant:
  * `Electrical`
  * `Magnetical`
  * `Mechanical`
or
 2. "ics"-variant:
  * `Electrics`
  * `Magnetics`
  * `Mechanics`
or
 3. "ic"-variant:
  * `Electric`
  * `Magnetic`
  * `Mechanic`

But the current mixture sounds odd to me. Maybe there is a pattern/system that I don't see if not I'd suggest to change the sub-libraries to a more stringent naming convention (3. - the "ic"-variant - being my favourite).

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/291

modelica-trac-importer, 2017-01-14:
**Comment by Stefan Vorkoetter on 16 Feb 2010 14:02 UTC**
I agree there's an inconsistency here, but that's due to English being inconsistent, not due to a poor choice of naming convention. The problems with your proposals are:

- I don't thing there's such a word as "magnetical", but if there is, I've never heard it used.

- "Mechanic" doesn't mean the same as "mechanical". It means "a person who repairs cars" (which makes a "celestial mechanic" a "person who repairs starships"?).

The "ics" proposal is probably the most consistent, and I've heard the words "magnetics" and "mechanics" used, but the word "electrics" is very rarely used, and thus sounds awkward.

Probably the closest one can get to consistent yet sounding "right" would be:

- Electrical
- Magentics
- Mechanical

In other words, keep all but "Mechanics" the same, and maybe add an "s" to "Magnetic".

modelica-trac-importer, 2017-01-14:
**Comment by Stefan Vorkoetter on 16 Feb 2010 14:04 UTC**
Further to my previous post, I don't think I'd add the "s" to "Magnetic".

Consider how each word would sound with "system" appended to it: "electrical system", "magnetic system", "mechanical system". Each of the three words works "correctly" in that context, and thus it is consistent.

modelica-trac-importer, 2017-01-14:
**Comment by mtiller on 23 Feb 2010 14:34 UTC**
I completely agree with Stefan.  This is definite a problem with English (blame Pete :-).

I also agree with his (latest) suggestion regarding naming (if we want to make a change at all).

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 23 Feb 2010 15:14 UTC**
Well I guess in that case I don't think it's worth the hassle. I leave this open for a bit though since people (BE where are you?) can have a say.

modelica-trac-importer, 2017-01-14:
**Comment by Stefan Vorkoetter on 23 Feb 2010 15:46 UTC**
For what it's worth, in MapleSim (where we have our own palette hierarchy on top of the underlying MSL hierarchy), we've always used the names "Electrical" and "Mechanical". I'd never paid much attention to that until this ticket appeared, but clearly someone here also felt there was an inconsistency.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2017-01-30:
@dietmarw What about `Thermal`?

dietmarw, 2017-01-30:
What about it?

tbeu, 2017-01-30:
Ah, there is no Thermical, Thermics or Thermic. So, Thermal is fine.

beutlich, 2018-10-28:
To summarize: Proposal is to rename Mechanics to Mechanical and keep Electrical and Magnetic as is.

tobolar, 2018-10-29:
To be really consequent, what about
- Fluid -> Hydraulical and
- Math -> Mathematical?

... and we should also think about Blocks...

dietmarw, 2018-10-29:
- Block -> Signals 
perhaps?

svorkoetter, 2018-10-29:
I think the Blocks one is orthogonal. It's called Blocks because it contains Modelica blocks (i.e. causal components).

For the others, we need to decide what type of name we want to use? The name of the field? The adjective describing objects used in the field? Something else?

If we wanted to use the former, we'd have Mechanics, Magnetics, Electrics (awkward, but I think technically correct), Mathematics, Fluidics (?)

For the second, it would be Mechanical, Magnetic, Electrical, Mathematical, Fluid (? perhaps something else, but Hydraulic is too narrow).

svorkoetter, 2018-10-29:
PS. "Math" is a common abbreviation for "Mathematics" in North American English (orders of magnitude more common than the full word). I believe the British call it "Maths".

HansOlsson, 2018-10-30:
> PS. "Math" is a common abbreviation for "Mathematics" in North American English (orders of magnitude more common than the full word). I believe the British call it "Maths".

Agreed, but we should also be careful since Math(s) is also a school subject and it seems some unrelated to that (e.g. plus, minus, division aren't in that library).

It might be that something other than 'Math' more clearly captures the contents. 

However, I'm also a bit hesitant about all of these changes, since there is a risk we change some of them for the sake of change.

beutlich, 2018-10-30:
I'd actually only go for the renaming Mechanics -> Mechanical and nothing else - for the sake of consistency.

beutlich, 2019-07-13:
I'll care for a PR renaming Mechanics -> Mechanical (but not now).

tobolar, 2019-08-30:
@MartinOtter Any concerns abot that renaming of Mechnics?

MartinOtter, 2019-09-05:
Its not really clear to me why package Mechanics should be renamed to package Mechanical.

"Mechanics" is the **field** to which the package provides components. For example, if you look for books, it is typical to find titles such as "An Introduction to Mechanics", "Principles of Quantum Mechanics", "The Variational Principles of Mechanics". It is much less common to use "mechanical" in the titel of a book about  mechanics (e.g. "Mechanical Engineering Handbook").

Other packages also use the "field" name as package name, e.g. Blocks, Fluid, Media, Math, Constants, Icons, Utilities. In general, in the MSL sometimes nouns (Machines, Vessles, Pipes, ...) and sometimes adjectives (Electrical, Rotational, ...) are used as package names. 

Note, there seems to be no established single noun for the field of Electrical Systems. In German, one would say "Elektrotechnik", but the englisch word "electrotechnology" sounds strange in the context of MSL and there seems nothing similar.

For me there must be a really strong reason to change the name of a top-level subpackage of MSL and I do not see yet a strong argument to change from Mechanics to Mechanical (don't misunderstand me: if the package would be named "Mechanical" from the beginning and it would now be proposed to change this to "Mechanics", I would have the same objection).

dietmarw, 2019-09-05:
The why is pretty much explained in https://github.com/modelica/ModelicaStandardLibrary/issues/291#issuecomment-272586999 and following. All by *native* speakers. 

The strong reason is really "consistency" which also reflects on the quality of the content.

svorkoetter, 2019-09-05:
While we native speakers agree there's an inconsistency, from rereading the comments (including my own) it also seems that we feel the inconsistency is in the English language, not in the naming convention. I think Martin makes a strong case for NOT changing it, and I now feel that changing it would not be a good idea. The inconsistency is not serious enough to reflect on the quality IMHO.

Aside on "Fluid": In speech when talking about the library, I always say "Fluids", and I often type "Fluids" by mistake when writing Modelica code involving the Fluid library. That one is probably more inconsistent than Mechanics is. I'm not proposing changing it though.

beutlich, 2019-09-05:
Clsoing it then as wontfix. No more need to argue here. Let's please concentrate on the more relevant issues.

closed 2019-09-05 by beutlich