modelica/ModelicaStandardLibrary #340: Rename and remove icons from Modelica.Icons

modelica-trac-importer, 2017-01-14:
**Reported by christiankral on 26 Mar 2010 06:27 UTC**
At the 66th design meeting in Hamburg Harburg it was decided to incorporate fully backwards compatible icons. With the next conversion script in MSL 4.0 the following changes shall be applied:

The following icons of `Modelica.Icons` shall be renamed
 * `Icons.RotationalSensor` -> `Icons.RoundSensor`
 * `Icons.TranslationalSensor` -> `Icons.RectangularSensor`

The following icons shall be removed (left side) and instead other icons (right side) shall be used
 * `Icons.Info` -> `Icons.Information`
 * `Icons.Library` -> `Icons.Package`
 * `Icons.Library2` -> `Icons.Package`

The following icons shall be indicated as deprecated
 * `Icons.MotorIcon`
 * `Icons.GearIcon`

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/340

modelica-trac-importer, 2017-01-14:
**Modified by christiankral on 26 Mar 2010 06:27 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by otter on 16 Jul 2010 09:50 UTC**
The proposal:

```
The following icons shall be indicated as deprecated

    * Icons.MotorIcon
    * Icons.GearIcon 
```

is not good. These icons are used in user models (it is therefore not possible to remove them) and it is useful to have these icons. However, these icons should be moved to the corresponding domain-specific library (Rotational).

modelica-trac-importer, 2017-01-14:
**Changelog removed by otter on 18 Jul 2013 11:53 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

tobolar, 2018-02-05:
For MotorIcon, there exists `Modelica.Mechanics.MultiBody.Icons.MotorIcon`, and for GearIcon, there is 'Modelica.Mechanics.Rotational.Icons.Gearbox'. Thus, both Icons.MotorIcon and Icons.GearIcon could be marked as obsolete.

HansOlsson, 2018-02-05:
> For MotorIcon, there exists Modelica.Mechanics.MultiBody.Icons.MotorIcon, and for GearIcon, there is 'Modelica.Mechanics.Rotational.Icons.Gearbox'. Thus, both Icons.MotorIcon and Icons.GearIcon could be marked as obsolete.

Or in a new version with conversion scripts they can be removed, and user models converted to the appropriate ones.

Looking in some commercial libraries I note that they already use Modelica.Mechanics.Rotational.Icons.Gear instead of Modelica.Icons.GearIcon. (I did not find any use of either MotorIcon outside of MSL).

tobolar, 2018-02-05:
Additionally, icon `Modelica.Mechanics.MultiBody.Icons.MotorIcon` should be renamed to 'Motor' only.

christiankral, 2018-02-07:
I have two comments/suggestions here:

1. In electrical engineering we call it machine, not motor, since a machine can be operated as motor or generator. Example: If an electric machine in a robotic application brakes electrically, it is operated as generator.  I wonder if mechanical engineers use and understand the term machine in the same way. If yes, the icon could be called ```Machine``` instead of ```Motor```. However, if mechanical engineers use the term motor, it is an electric motor, isn't it? Or could it be a hydraulic motor that this particular icon is used for?

2. Unify icon design: I think we could even consider to make the Multibody motor icon look more like the ones in ```Modelica.Electrical.Machines.Icons```. One could copy one of these icons an make the housing color red to match the actual icon color.

tobolar, 2018-02-08:
@christiankral 
1. I think the common name in the mechanical engineering is _(electric) drive_ since in most cases it is used as a power delivering unit. The name is used even if the machine could recuperate the energy.

2. To use an icon from `Modelica.Electrical.Machines.Icons` would be reasonable, especially if there would be some kind of  "basic" icon for those purposes.
Even better, the robot example would use a machine directly from `Modelica.Electrical.Machines` package, i.e. we don't use any machine model designed only for that particular case (`Modelica.Mechanics.MultiBody.Examples.Systems.RobotR3.Components.Motor`).
On the contrary, a bearing friction model from `Modelica.Mechanics.Rotational.Components` should be used within the machines. (Optionally, we can move `Modelica.Electrical.Machines.Losses.Friction` to `Modelica.Mechanics.Rotational.Components` and modify it to be consistent with other components.)

christiankral, 2018-05-21:
@tobolar I created a generic machine and drive icon in fbb74f88e3ced7a797b8d24939c040bc37ec45c9:

- Modelica.Electrical.Machines.Icons.Machine (left)
- Modelica.Electrical.Machines.Icons.Drive (right)

[image]

The generic icons are gray which should match the rotational and multi body color scheme approximately.

The shape of the machine icon is fully in line with all other electric machine icons.

dietmarw, 2018-05-28:
@tobolar

dietmarw, 2018-06-05:
Is this issue now resolved. It looks like it to me?

tobolar, 2018-06-05:
I don't believe so. It is aimed for next major version at the earliest.

dietmarw, 2018-06-05:
Ah yes of course. I only saw the latest additions.

PR #3016 (2019-07-18): Rename Modelica.Icons.{RotationalSensor, TranslationalSensor}
Closes #340.
closed 2019-07-18 by beutlich