modelica/ModelicaStandardLibrary #343: Integrate HeatTransfer components from Dissipation library

modelica-trac-importer, 2017-01-14:
**Reported by kproelss on 28 Mar 2010 19:25 UTC**
Heat transfer coefficient correlations from the `Dissipation`library are going to be part of the MSL 3.2 under `Modelica.Fluid.Dissipation`. As decided at the 66th Design Meeting the existing corresponding component models are not included in that release, because their user interface has not been clearly defined yet.

Currently, variable modifications are required when using the component in addition to a heat port that needs to be connected. As a task for the next major release, the integration of such components into existing Fluid component models or their usage as top level system components needs to be discussed and a corresponding interface needs to be defined.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/343

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 21 May 2010 11:10 UTC**
Milestone MSL3.3 deleted

modelica-trac-importer, 2017-01-14:
**Modified by fcasella on 14 Dec 2015 22:41 UTC**

modelica-trac-importer, 2017-01-14:
**Changelog removed by fcasella on 14 Dec 2015 22:41 UTC**

beutlich, 2019-03-19:
@wischhusen @casella Almost 9 years later, is this still desirable for MSL 4.0.0?

wischhusen, 2019-03-27:
Honestly, the integration concept of the Heat´Transfer components (not functions) seems to be too complicated for regular users. So I would not insist to keep this issue open.

beutlich, 2019-03-27:
> So I would not insist to keep this issue open.

Closing then.

closed 2019-03-27 by beutlich