modelica/ModelicaStandardLibrary #359: Move conditional heat port models to Modelica.Thermal.HeatTransfer.Interfaces

modelica-trac-importer, 2017-01-14:
**Reported by christiankral on 21 Apr 2010 15:44 UTC**
The conditional heat port models `Modelica.Electrical.Analog.Interfaces.ConditionalHeatPort` and `Modelica.Electrical.MultiPhase.Interfaces.ConditionalHeatPort` are not specific for electrical systems. Since these conditional heat port models can be applied to any model with one or more conditional heatports, the location in the electrical domain is not correct. Therefore, both conditional heat port models shall be moved to `Modelica.Thermal.HeatTransfer.Interfaces` when the next conversion script is due.

**Example**: Conditional heat port models shall also be applied in `Modelica.Magnetic`. 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/359

modelica-trac-importer, 2017-01-14:
**Comment by AHaumer on 22 Apr 2010 10:07 UTC**
partly resolved by e34dfbd972ead6c71f1b738ef1df8f80654af251: 
PartialELementaryConditionalHeatPort and PartialConditionalHeatPort defined in Thermal.HeatTransfer.Interfaces - Electrical.Analog could use that / extend from that!
Electrical.MultiPhase has a special ConditionalHeatPort, adapted for the m phases - would leave this model in MultiPhase.
Used the ConditionalHeatPorts in Mechanics: 
Rotational.Damper, SpringDamper, ElastoBacklash, BearingFriction, Brake need some testing. 
Rotational.Clutch, OneWayClutch, LossyGear, GearBox: not yet implemented
Translational.Damper, SpringDamper, ElastoGap, SupportFriction, Brake need some testing.
Translational.MassWithStopAndFriction: not yet implemented.
Are there other components (MultiBody?) that would need a ConditionalHeatPort?

modelica-trac-importer, 2017-01-14:
**Comment by AHaumer on 22 Apr 2010 16:46 UTC**
Summary:
Done: Thermal.HeatTransfer.Interfaces.PartialElementaryConditionalHeatPort
Electrical.MultiPhase.Interfaces.ConditionalHeatPort remains in MultiPhase.
Electrical.Analog could use / extend from that.

modelica-trac-importer, 2017-01-14:
**Modified by AHaumer on 27 Apr 2010 19:33 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by clauss on 29 Apr 2010 12:47 UTC**
Electrical.Analog models changed: use now Thermal.HeatTransfer.Interfaces.PartialElementaryConditionalHeatPort.
Electrical.Analog.Interfaces.ConditionalHeatPort removed

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 29 Apr 2010 13:17 UTC**
Clauss please reintroduce the deleted `Electrical.Analog.Interfaces.ConditionalHeatPort` (2deef6fcabe46dc85f18fb9f8c1fc5cba40fb109) again as MSL3.2 is a **backward compatible** release. Deleting components from the library will break user models. When you done you can leave this ticket open as a reminder and when the next MSL with conversion script comes you can delete it but also have to provide a conversion script entry.

modelica-trac-importer, 2017-01-14:
**Comment by clauss on 30 Apr 2010 07:58 UTC**
Electrical.Analog.Interfaces.ConditionalHeatPort reintroduced, done.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 30 Apr 2010 08:35 UTC**
Electrical.Analog.Interfaces.ConditionalHeatPort reintroduced in d887b5c3e3600bd657a22d26ad92deb972cefeb6.

modelica-trac-importer, 2017-01-14:
**Comment by christiankral on 4 May 2010 13:37 UTC**
There is still an non backwards compatible change in the Analog package of the MSL 3.2. Even if the conditional heat port has been re-introduced in Electrical.Analog, the models are NOT using these heat ports. Instead the non compatible heat ports of `Modelica.Thermal.HeatTransfer.Interfaces.PartialElementaryConditionalHeatPort` are used. 

**Result:** Example `Modelica.Magnetic.FundamentalWave.Examples.Components.EddyCurrentLosses` is broken and other user specific models may not be working, too.

The partial **original implementation** of the heat ports **has to be established! **

modelica-trac-importer, 2017-01-14:
**Comment by clauss on 5 May 2010 07:08 UTC**
done, original implementation established, above mentioned example tested.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

christiankral, 2018-01-03:
There was the duplicate ticket #2312 introduced which summarizes some aspects on the duplicate conditional heat ports.

beutlich, 2019-09-20:
Removing milestone for reasons given in https://github.com/modelica/ModelicaStandardLibrary/pull/3097#issuecomment-522327170.