modelica/ModelicaStandardLibrary #386: Animation missing in MultiBody.Forces.SpringDamperSeries

modelica-trac-importer, 2017-01-14:
**Reported by otter on 22 Jul 2010 10:19 UTC**
The component MultiBody.Forces.SpringDamperSeries does not yet have animation 

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/386

modelica-trac-importer, 2017-01-14:
**Modified by otter on 5 Aug 2010 21:37 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 6 Sep 2012 11:57 UTC**
These tickets don't rely on conversion scripts, hence rescheduling for the next major release

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 29 Apr 2014 06:29 UTC**
@dietmarw: Should Milestone be set back to MSL_withConversionScript?

modelica-trac-importer, 2017-01-14:
**Changelog removed by beutlich on 29 Apr 2014 06:29 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by otter on 13 Dec 2015 08:05 UTC**
Do not yet have time to provide an animation. Feel free to propose an animation object and send a patch

beutlich, 2017-02-20:
> Feel free to propose an animation object and send a patch

@MartinOtter Done by #2183. Please check the proposed animation enhancements for  SpringDamperSeries and SpringDamperParallel.

beutlich, 2017-04-18:
Resolved in master by #2183.

closed 2017-04-18 by beutlich