modelica/ModelicaStandardLibrary #464: PlanarMultiBody ought to be included into MSL 3.3

modelica-trac-importer, 2017-01-14:
**Reported by Volker Beuter on 2 Dec 2010 16:26 UTC**
The PlanarMultiBody package developed at DLR-RM and already presented at the Modelica2008 conference was intended to be included into the MSL some day. This could be done with the 3.3 release. As soon as I have some Dymola 7 license again I may do some testing.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/464

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 29 Apr 2014 06:33 UTC**
@dietmarw: Should Milestone be set back to undecided?

modelica-trac-importer, 2017-01-14:
**Changelog removed by beutlich on 29 Apr 2014 06:33 UTC**

modelica-trac-importer, 2017-01-14:
**Modified by beutlich on 29 Apr 2014 06:33 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 29 Apr 2014 06:47 UTC**
It's not on me to decide this but the map-lib group should discuss this.

modelica-trac-importer, 2017-01-14:
**Comment by dzimmer on 11 Jun 2014 15:32 UTC**
The planar mechanical library is available on github:

https://github.com/dzimmer/PlanarMechanics

We support its inclusion into the MSL and welcome a discussion about it.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 11 Jun 2014 20:52 UTC**
Dirk, are we really speaking about the same library? The [PlanarMultiBody](http://elib.dlr.de/55893/1/otter2008-modelica-planar-multi-body.pdf) was presented 2008 in Bielefeld and e.g. contains contact elements and JointRR. Your library [PlanarMechanics](http://www.ep.liu.se/ecp/076/069/ecp12076069.pdf) was presented 2012 in Fürstenfeldbruck and was focused on teaching and simplification of MultiBody.

modelica-trac-importer, 2017-01-14:
**Comment by dzimmer on 12 Jun 2014 08:54 UTC**
Oh indeed, you are right.  My mistake. 

It is good that you mention that because we plan to integrate the contributions from the Bielefeld version into our own but sofar did not have time for it.

But this definitly should not be forgotten. For the moment however, I refrain from opening a new ticket since the goal of having a planar mechanical librariy in the MSL is the same.

modelica-trac-importer, 2017-01-14:
**Modified by otter on 7 Oct 2014 15:42 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by otter on 19 Nov 2014 17:36 UTC**
The latest version of the PlanarMechanics library is available from here:
https://github.com/DLR-SR/PlanarMechanics

modelica-trac-importer, 2017-01-14:
**Comment by otter on 19 Nov 2014 18:15 UTC**
I evaluated the PlanarMechanics library:
I like it and recommend to include this library in the next version of the Modelica Standard Library (MSL). There are two attachments to this ticket containing the evaluation (EvaluationModelicaLibraryCandidate_PlanarMechanics_otter.pdf) and the style check result (PlanarMechanics_StyleCheckLog.html).

**The following issues should be fixed before inclusion in the MSL**

- The name of the library, PlanarMechanics, has a different style as the other sub-libraries in Modelica.Mechanics. It is recommended to rename the library to "Planar" (then there are sub-libraries MultiBody, Planar, Rotational, Translational; note in all these cases one could also expand the name with Mechanics, like RotationalMechanics. However, this is not done in the MSL in order to not repease the same string, Mechanics, in the full path name).

- Style check with Dymola gives several (minor) issues with missing description string (PlanarMechanics_StyleCheckLog.html). This needs to be fixed.

- Check with Dymola "pedantic mode" gives 18 errors (all: "the annotation 'revisions' appeared multiple times."). This needs to be fixed.

- Issues with PlanarMechanics.Parts.Body:
Parameter "I" is unclear from the documentation. It is stated "Inertia of the Body". From the equations it is clear that the Inertia is with respect to "frame_a". Therefore the documentation string should be changed to "Inertia of the body with respect to the origin of frame_a along the z-axis of frame_a".
Parameter "g" (local gravity acting on the mass) is questionable. It seems not practical to define at every mass a separate gravity acceleration. If there is a good reason for this parameter, explain it in the documentation, but move it in an "Advanced" tab so that a user does not see it at once. If there is no good reason for this parameter, remove it.

- In the UsersGuide a link to the PlanarMechanics paper should be added (or even better to include this paper in the library and have a link to the local copy of the paper.)

**The following recommendation are a "matter of style" and it is up to the library authors whether to follow the suggestions below**

- The structure of the sub-libraries should be done in such a way that the most used packages/models are listed first and the least important at the end. This would suggest to move the Interfaces and Utilities sub-packages at the end (note, all sub-packages of MSL have Interfaces close to the end of the package list). 

- Why is the Types package outside of the Utilities package but the Icons package is inside the Utilities packages. This seems not logical. Either both should be outside or inside the Utilities package (the latter is better).

- The MultiBody package has "Parts" and "Forces" sub-libraries. In the PlanarMechanics library there is only one package "Parts", that contains both. Maybe it would be better to also use Parts and Forces sub-libraries in PlanarMechanics (I have no strong opinion, but just recognized this).

- The connector icons have a small "cross" in the middle of the icon. This does not look good and is anyway too small to be really visible. I would remove it. Furthermore, the frames seem to be a bit too large. Why not using a similar size as the MultiBody frame connectors (and just differen color)?

- In the MultiBody library, also the sensors have a visualization (to indicate visually what they are measuring, since it is so easy to make a mistake here). In the PlanarMechanics library such a visualization is not present.

- It would be helpful to add a screenshot of corresponding animation to the documentation of every example, in order to get a quick overview about the example.

modelica-trac-importer, 2017-01-14:
**Comment by johan on 21 Nov 2014 05:45 UTC**
I have evaluated the library and my conclusion is somewhat different than Martins. The discussion touches on more generic aspects of what should be included in MSL. There are two documents attached to this comment:

EvaluationModelicaLibraryCandidate_PlanarMechanics_Johan_Andreasson.pdf - formal evaluation document.

Feedback_Johan_Andreasson.pdf - My feedback to the library developers, including generic comments, detailed suggestions and some open questions.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 21 Nov 2014 08:20 UTC**
I can only repeat myself on the two different planar libraries:
> The [PlanarMultiBody](http://elib.dlr.de/55893/1/otter2008-modelica-planar-multi-body.pdf) was presented 2008 in Bielefeld and e.g. contains contact elements and JointRR. Dirk's library [PlanarMechanics](http://www.ep.liu.se/ecp/076/069/ecp12076069.pdf) was presented 2012 in Fürstenfeldbruck and was focused on teaching and simplification of MultiBody.

modelica-trac-importer, 2017-01-14:
**Comment by franciscus&#x2e;vanderlinden on 26 Nov 2014 11:55 UTC**
I have looked through the comments of otter:

Replying to [comment:12 otter]:
> - Style check with Dymola gives several (minor) issues with missing description string (PlanarMechanics_StyleCheckLog.html). This needs to be fixed.
> - Check with Dymola "pedantic mode" gives 18 errors (all: "the annotation 'revisions' appeared multiple times."). This needs to be fixed.

This is fixed by now by tbeu on [Github](https://github.com/dzimmer/PlanarMechanics/pull/59). 

Replying to [comment:12 otter]:
> - Issues with PlanarMechanics.Parts.Body:
Parameter "I" is unclear from the documentation. It is stated "Inertia of the Body". From the equations it is clear that the Inertia is with respect to "frame_a". Therefore the documentation string should be changed to "Inertia of the body with respect to the origin of frame_a along the z-axis of frame_a".

This is fixed by now by me [Github](https://github.com/dzimmer/PlanarMechanics/pull/60). 

Replying to [comment:12 otter]:
> Parameter "g" (local gravity acting on the mass) is questionable. It seems not practical to define at every mass a separate gravity acceleration. If there is a good reason for this parameter, [...]

I have moved this to the Advanced tab. However, I would like to keep the parameter to accomodate the possibility to include multiple Planar mechanism in a different orientation.

Replying to [comment:12 otter]:
> - The MultiBody package has "Parts" and "Forces" sub-libraries. In the PlanarMechanics library there is only one package "Parts", that contains both. Maybe it would be better to also use Parts and Forces sub-libraries in PlanarMechanics (I have no strong opinion, but just recognized this).

I would like to link to the discussion on GitHub: [Github](https://github.com/dzimmer/PlanarMechanics/issues/56). Summary: I would keep it this way, it is the same was as in Modelica.Rotational.

modelica-trac-importer, 2017-01-14:
**Comment by leo&#x2e;gall on 27 Nov 2014 09:31 UTC**
**General comments about inclusion**

Long story short: I think this is a great library, but I don't know if it needs to be in MSL.

- I don't see users having a big problem using only two of the three dimensions of MultiBody. Of course, PlanarMechanics is great for teaching. If you want to look into the components, with MultiBody it gets complicated very fast.

- Do we want to add and maintain several modeling levels of MSL? Lets consider a different domain of MSL: Do we want to add a simplified Fluid library which doesn't have multi-phase or doesn't allow flow reversal? We have #1491 and #573 to discuss these strategic and management issues.

**Comments on implementation**

(Review based on PlanarMechanics 1.3.0, download date 2014-11-19, used in Dymola 2015)

- Parameters and connectors should be as close as possible to MultiBody, if there is no good reason against. E.g. connectors use scalars fx, fy. What was the reason to not use f![2]? I didn't compare all parameters, yet.

- Would it make sense to add adapter components to MultiBody connectors?

- Packages for Parts/Forces/Sources: For me it is OK to have a different packaging, compared to MultiBody. But, if there are the same packages, they should be in the same order (e.g. Parts and Joints are swapped)

- PlanarMechanics.Utilities should be split and placed in the appropriate places in MSL.

- Icons (especially Revolute) look like "old" style from MSL 3.2. For a newly added library, I would propose to use the new style. Of course, there should be a clear visual distinction between components from Planar and MultiBody. But I think the current icons are to close to MultiBody from MSL 3.2.

- I propose to change the icon of MultiBody.World. It should have the z axis. This way, it should be easier to distinguish between Planar and MultiBody models from just looking at the model diagram.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Dec 2014 07:01 UTC**
Ticket retargeted after milestone closed

AHaumer, 2018-01-04:
I've also used the PlanarMechanics, it is definitvely usefull fpr certain applications.
I support the idea of including PlanarMechanics in the MSL.

AHaumer, 2018-05-21:
@dzimmer: If you create a pull request, it could be included within short.

HansOlsson, 2018-05-21:
Just to be clear: We are currently working on the 3.2.3 release of Modelica Standard Library, and the proposal is to include Planar in 3.3, right?