modelica/ModelicaStandardLibrary #477: Interpolation options

modelica-trac-importer, 2017-01-14:
**Reported by mtiller on 14 Jan 2011 17:54 UTC**
I've been working on a project for some time that uses model inversion extensively.  One of the issues we have seen is that the tables provided in the Modelica Standard Library assume extrapolation beyond the limits of the independent variables.  In many cases, this is not desirable. 

If the behavior of the model is only valid over a given table, there should be a way to reflect this.  If they want a different way of interpolating, that should also be possible.

Specifically, it should be possible for users to specify different approaches to interpolation beyond the edges.  I would propose at least the following options:
  * Fatal assertion - A fatal assertion is violated in such cases.
  * Warning assertion - A warning assertion (and presumably a warning message) is generated.
  * Limits - The inputs are "capped" by the upper and lower values of the independent variable.
  * Extrapolation - The current behavior
  * Periodic - The independent variable should "wrap" around when it goes beyond the edges of the table.

These are not arbitrary choices, we have seen use cases for all of these in our regular work.

Ideally, it should be possible to have a different approach for each independent variable in a given table.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/477

modelica-trac-importer, 2017-01-14:
**Comment by Volker&#x2e;Beuter on 24 Jan 2011 15:27 UTC**
In case you issue a warning when leaving the interpolation range but do not stop, you are left with the question how to extrapolate. So the second choice is not a choice of its own but something to combine possibly with one of the three last choices.
Besides the warning the four remaining cases (assertion, constant, linear and periodic extrapolation) is what we at Kämmerer implemented for extrapolation in our cubic spline package Spline. In two dimensions things get even more complicated: There are applications where you need linear extrapolation in x but periodic in y (e.g. a radius function for some shape in cylinder coordinates. So all possible combinations are implemented in our AreaSpline package.

modelica-trac-importer, 2017-01-14:
**Modified by hubertus on 23 Apr 2014 13:48 UTC**

modelica-trac-importer, 2017-01-14:
**Changelog removed by hubertus on 23 Apr 2014 13:48 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 25 Apr 2014 13:20 UTC**
See also #1153 where this ticket is referenced.

beutlich, 2017-08-29:
@mtiller Good news: After closing #1839, #2120, #2190 and #2244 I believe your issue is resolved.  :icecream:

closed 2017-08-29 by beutlich