modelica/ModelicaStandardLibrary #496: Inconsistent partial model naming [in Modelica.Magnetic]

modelica-trac-importer, 2017-01-14:
**Reported by christiankral on 4 Feb 2011 14:18 UTC**
The partial models 
 * `Interfaces.PartialTwoPortsElementary` and 
 * `Interfaces.PartialTwoPorts` 
are implemented differently in the packages 
 * `Modelica.Magnetic.FluxTubes` and 
 * `Modelica.Magnetic.FundamentalWave`.

# Current Implementation
## `FluxTubes.Interfaces.PartialTwoPortsElementary`

```
partial model PartialTwoPortsElementary 
  PositiveMagneticPort port_p;
  NegativeMagneticPort port_n;
end PartialTwoPortsElementary;

```

## `FluxTubes.Interfaces.PartialTwoPorts`

```
partial model PartialTwoPorts 
  extends FluxTubes.Interfaces.PartialTwoPortsElementary;
  SI.MagneticPotentialDifference V_m;
  SI.MagneticFlux Phi;

equation 
  V_m = port_p.V_m - port_n.V_m;
  Phi = port_p.Phi;
  0 = port_p.Phi + port_n.Phi;
end PartialTwoPorts;
```

## `FundamentalWave.Interfaces.PartialTwoPort`
```
partial model PartialTwoPort 
  PositiveMagneticPort port_p;
  NegativeMagneticPort port_n;
end PartialTwoPort;
```

## `FundamentalWave.Interfaces.PartialTwoPortElementary`
```
partial model PartialTwoPortElementary
  extends FundamentalWave.Interfaces.PartialTwoPort;
  Modelica.SIunits.ComplexMagneticPotentialDifference V_m;
  Modelica.SIunits.ComplexMagneticFlux Phi;

equation 
  V_m = port_p.V_m - port_n.V_m;
  port_p.Phi = Phi;
  port_p.Phi + port_n.Phi = Complex(0,0);
end PartialTwoPortElementary;
```

The naming of the partial models in the fundamental wave library is simply a bug. 

# Additional Issues

The term `port` used in the magnetic and thermal domain refers to a connector and it represents two pins or plugs in the electrical domains. We should come up with some clarification here. We should also unify whether or not we should used the term `Partial` in the class name. We have various different implementations in the MSL. 

The implementation of partial models is even entirely different throughout the package `Modelica.Electrical`: 

 * `Analog.Interfaces.TwoPin` is a partial model containing 
   * definition of voltage drop `v` across pins  
 * `MultiPhase.Interfaces.TwoPlug` is a partial model containing
   * definition of voltage drop `v` across plugs
   * definition of current `i` into positive plug
 * `Analog.Interfaces.OnePort` and `MultiPhase.Interfaces.OnePort` are partial models with two pins or plugs, respectively [the problem here is that we are using the term `port` as an abstraction for two pins and plugs, respectively, whereas this term is used differently, e.g, in the magnetic domain]; these partial models contain 
   * definition of voltage drop `v` across pins and plugs, respectively 
   * definition of current `i` into positive pin and plug, respectively
   * current balance of positive and negative pin (or plug)
 * `Analog.Interfaces.TwoPort` and `MultiPhase.Interfaces.TwoPort` are partial models with four pins or plugs, respectively; these partial models contain 
   * definition of voltage drop `v` across pins and plugs, respectively 
   * definition of current `i` into positive pin (or plug)
   * current balance of positive and negative pin (or plug)
 * `MultiPhase.Interfaces.FourPlug` contains
   * definition of voltage drop `v` across paris of plugs, respectively 
   * definition of current `i` into positive plugs

However, a unification of namings and implementations is highly desirable since all the differences are confusing to experienced and news users. I understand that we do need a conversion script for renaming partial models; so we do have wait for Modelica 4, I suppose.

# Proposal

So here we have lot very different namings with different implementations. On my opinion we need three different partial model types:
 * `TwoPinGeneric`, `TwoPlugGeneric`, etc., which contains only the connectors; these are the most generic components; they are also used for sources and sensors since it may be undesired in these cases to have a pre defined variable for either the potential difference or the flow variable
 * `TwoPinElementary`, `TwoPlugElementary`, etc., which contains the connectors and the definitions of the potential difference, e.g., `v`, and the flow variable into the positive connector, e.g., `i`; these models are mostly used for graphical programming  
 * `TwoPin`, `TwoPlug`, or `TwoPinBalanced`, `TwoPlugBalanced`, etc., containing the flow balances in addition to the `Elemenary` named partial models; these models are used for textual programming

Naming may be discussed.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/496

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 4 Feb 2011 15:38 UTC**
Christian,

if I understand your proposal correctly then the different classes would be defined as follows (taking `TwoPin` as an example):
 * 
```text/x-modelica
partial model TwoPin
 extends TwoPinElementary
...
```
 * 
```text/x-modelica
partial model TwoPinElementary
 extends TwoPinGeneric
...
```

 * 
```text/x-modelica
partial model TwoPinGeneric
...
```
Did I understand this correctly? I'm not quite sure why the last group in your proposal (`TwoPin`,...) is/should be especially suitable for textual programming. I mean they inherit everything from the other partial models right? So what would make them so special then?

modelica-trac-importer, 2017-01-14:
**Comment by christiankral on 4 Feb 2011 16:45 UTC**
Right, thats the way I meant it; the reason why TwoPin is suitable for textual programming, is because it already contains the flow balance:
```
partial model TwoPin
  extends TwoPinElementary;
  p.i + n.i = 0;
end TwoPin;
```
In this case I cannot make a graphical connection between the positive and negative pin when extending from this class.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 4 Feb 2011 21:01 UTC**
Right I see. Thanks for explaining. Otherwise I'm most definitely for removing **Partial** from the name. Since this is part of the model specifications it should be the job of the tool to highlight that if wanted and not done clumsily by putting the already present information also in the name :)

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

christiankral, 2019-06-22:
I will create a PR to remove Partial from the names and fix the naming issue with of the TwoPorts.

beutlich, 2019-07-16:
Reopen since #2998 did not address Magnetic-FluxTubes.

christiankral, 2020-02-14:
When going through #3341 I realized that we still have two different naming conventions in the magnetic domain: `TwoPort` vs `TwoPorts`. I suppose this shall be fixed before releasing v4.0.0. @dietmarw, @beutlich and @AHaumer do you agree?

The following list shows the difference w.r.t. the elementarty ports:

- `Modelica.Magnetic.FluxTubes.Interfaces.TwoPortsElementary`
- `Modelica.Magnetic.FundamentalWave.Interfaces.TwoPortElementary`
- `Modelica.Magnetic.QuasiStatic.FluxTubes.Interfaces.TwoPortsElementary`
- `Modelica.Magnetic.QuasiStatic.FundamentalWave.Interfaces.TwoPortElementary`

At the end one question remains: `TwoPorts` or `TwoPort`? Any suggestions?

Other domains use, for example:

- `Modelica.Electrical.Analog.Interfaces.TwoPin`
- `Modelica.Electrical.Analog.Interfaces.OnePort`
- `Modelica.Electrical.Polyphase.Interfaces.TwoPlug`
- `Modelica.Mechanics.Rotational.Interfaces.PartialTwoFlanges`
- `Modelica.Mechanics.Translational.Interfaces.PartialTwoFlanges`
- `Modelica.Mechanics.MultiBody.Interfaces.PartialTwoFrames`

dietmarw, 2020-02-14:
A quick search lead me to https://en.wikipedia.org/wiki/Two-port_network
So I'm for `TwoPort`.

christiankral, 2020-02-14:
OK then. So I will create a PR to fix the names of 

- `Modelica.Magnetic.FluxTubes`
- `Modelica.Magnetic.QuasiStatic.FluxTubes`

beutlich, 2020-02-22:
Closing again after merge of #3426.

PR #2998 (2019-07-16): Fix inconsistent partial model naming in Modelica.Magnetic
close #496
PR #3051 (2019-07-17): Refs #496: Fix inconsistent partial model naming of FluxTubes
close #496
closed 2020-02-22 by beutlich