modelica/ModelicaStandardLibrary #545: Molar units missing

modelica-trac-importer, 2017-01-14:
**Reported by wischhusen on 2 May 2011 11:21 UTC**
SIunits require also molar specific enthalpy:

```
type MolarEnthalpy = Real (final quantity="MolarEnthalpy", final unit=
        "J/mol", nominal=5e4);
```

Moreover, a very popular Non-SIunit for molar mass is required e.g. in Modelica.Fluid.Dissipation.HeatTransfer.StraightPipe.kc_twoPhase_overall_KC (see ticket #502):

```
type MolarMass_gpmol = Real (final quantity="MolarMass", final unit=
          "g/mol", min=0, nominal=1) "Molar mass in gramm per mol";
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/545

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 2 May 2011 11:55 UTC**
Replying to [ticket:545 wischhusen]:
> Moreover, a very popular Non-SIunit for molar mass is required e.g. in Modelica.Fluid.Dissipation.HeatTransfer.StraightPipe.kc_twoPhase_overall_KC (see ticket #502):
>
> ```
> type MolarMass_gpmol = Real [...]

Instead of introducing yet another non-SI unit which is simply a multiple of an existing unit I would rather suggest the use of `displayUnit`. E.g.,
```
parameter Modelica.SIunits.MolarMass GrammPerMol(displayUnit="g/mol");
```
This requires of course tool support (e.g., in Dymola you can add such conversions via script:
 ```
defineUnitConversion("kg/mol", "g/mol", 1e3, 0);
```
)

modelica-trac-importer, 2017-01-14:
**Comment by thorade on 14 Sep 2011 08:43 UTC**
I agree that it would be nice to have displayUnit g/mol for MolarMass.

Also it would be nice to have displayUnit mol/l for Modelica.SIunits.Concentration
The SI default unit for concentration is mol/m³ but mol/l is also frequently used.
Another molar quantity that would be nice to have is molar density (the inverse of molarVolume). It should have the same dimension as Concentration so it can be defined as an Alias of Concentration.

For Viscosity it would be nice to add displayUnit µPa·s
Default unit is Pa·s but that gives very low values.

modelica-trac-importer, 2017-01-14:
**Comment by otter on 1 Jan 2013 10:40 UTC**
Changed owner to Hubertus, since I do not have enough fluid knowledge to decide.

modelica-trac-importer, 2017-01-14:
**Changelog removed by otter on 1 Jan 2013 10:40 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by hubertus on 25 Mar 2013 21:56 UTC**
Proper support for Molar units, also in function calls, is a major change and upgrade of functionality. This should be done for the next major version of Modelica.Media, so I update the milestone to MSL 3.3.

modelica-trac-importer, 2017-01-14:
**Modified by dietmarw on 1 Aug 2015 21:38 UTC**

beutlich, 2019-07-06:
@thorade Do you think you could provide a PR for adding the missing units? Thanks!

thorade, 2019-07-09:
Yes, I can submit a PR, in two weeks, after vacation and a conference!

beutlich, 2019-07-30:
@thorade Have also a look at https://github.com/ORNL-Modelica/TRANSFORM-Library/blob/master/TRANSFORM/Units/MolePerMass.mo

beutlich, 2019-09-05:
> Yes, I can submit a PR, in two weeks, after vacation and a conference!

Friendly reminder. :bell:

thorade, 2019-09-06:
After re-reading this ticket, I suggest to add the following
new types:
```
type MolarEnthalpy = Real (final quantity="MolarEnthalpy", final unit="J/mol", nominal=5e4);
type MolarDensity = Real (final quantity="MolePerMass", final unit="mol/kg");
```

new display unit conversion scripts:
```
defineUnitConversion("kg/mol", "g/mol", 1e3, 0);
defineUnitConversion("mol/m3", "mol/l", 1e-3, 0);
defineUnitConversion("Pa.s", "µPa·s", 1e6,, 0);
```
But before actually submitting that as PR, I would like to have feedback on the names - is there a naming scheme for the types?
For the defineUnitConversion it seems these are not stored in this repo, so I cannot submit a PR for it. Are they stored anywhere or should they be sent by mail to the tool vendors? As displayUnit is used all over the MSL, it seems to be standardized!?

beutlich, 2019-09-06:
defineUnitConversion is Dymola specific. Not sure about the `µ` prefix either.

We already have (introduced by 53cce7f3e574ad179ac707bdbdc6effdb1949c7d)

```mo
type MolarEnergy = Real (final quantity="MolarEnergy", final unit="J/mol", nominal=2e4); 
type MolarEnthalpy = MolarEnergy; 
```

and (introduced by e2676d3ae248c1811a90ba0ffbc41d3d4b4f17f9)

```mo
type MolarDensity = Real (final quantity="MolarDensity", unit="mol/m3");
```

and since 7366c28985d642838fadbe8fae46305e9c62b98f

```mo
type AmountOfSubstance = Real (final quantity="AmountOfSubstance", final unit="mol", min=0); 
type IonicStrength = Real (final quantity="IonicStrength", final unit="mol/kg"); 
```

I guess we can close this issue then, except if you want to add a new short type definition for IonicStrength:

```mo
type MolePerMass = IonicStrength;
```

beutlich, 2019-09-06:
> **Comment by hubertus on 25 Mar 2013 21:56 UTC**
> Proper support for Molar units, also in function calls, is a major change and upgrade of functionality. This should be done for the next major version of Modelica.Media, so I update the milestone to MSL 3.3.

Otherwise, could be, that@hubertus65 meant proper support of molar units in Modelica.Media as this issue's actual topic.

HansOlsson, 2019-09-09:
> defineUnitConversion is Dymola specific. Not sure about the µ prefix either.

The specification says that you should use "u" instead of "µ".  (And I haven't checked whether the latter is micron-symbol or greek mu.)

However, tools are encouraged to use the proper symbol when displaying that  - as discussed in https://github.com/modelica/ModelicaSpecification/issues/2142

And since Dymola 2019 the need for
`defineUnitConversion("kg/mol", "g/mol", 1e3, 0);`, is reduced, as we can deduce it from the conversion between kg and g; so I wouldn't worry too much about adding that.

thorade, 2019-09-09:
OK, so unit conversion will not be handled here, instead conversions could be suggested to tool vendors.

I think we could have molar units more or less complete by adding molar entropy:
```
type MolarEntropy = Real (final quantity="MolarEntropy", final unit="J/mol.K"); 
```
and also the short type definition `type MolePerMass = IonicStrength;`

And then close the ticket.

beutlich, 2019-09-09:
> I think we could have molar units more or less complete by adding molar entropy:

It already is there. Have a look at

https://github.com/modelica/ModelicaStandardLibrary/blob/1897ca82d47bb48d33af5d8ac96efb1c4d73050d/Modelica/SIunits.mo#L1365-L1366

(different unit though).

HansOlsson, 2019-09-09:
Should we add MolePerMass or https://en.wikipedia.org/wiki/Molality ?

I would also propose to have that as the base-class (with definition) and IonicStrength equal to that; the effect is the same - but it makes more sense to me.

thorade, 2019-09-09:
> It already is there. Have a look at
> 
> https://github.com/modelica/ModelicaStandardLibrary/blob/1897ca82d47bb48d33af5d8ac96efb1c4d73050d/Modelica/SIunits.mo#L1365-L1366
> 
> (different unit though).

That is the correct unit, I forgot the parantheses.

I agree with Hans' proposal of adding molality and having IonicStrength as alias!

beutlich, 2019-10-05:
Nothing more to be expected.

closed 2019-10-05 by beutlich