modelica/ModelicaStandardLibrary #559: application of choices annotation on enumerations

modelica-trac-importer, 2017-01-14:
**Reported by eichler on 25 May 2011 08:45 UTC**
the idea: use a existing enumeration in a choicebox with a restricted number of choices.

this could avoid the definition of an additional enumeration type with almost the same options/choices.

example:

```
	type Dynamics=enumeration(DynamicFreeInitial "DynamicFreeInitial",
		FixedInitial "FixedInitial",
		SteadyStateInitial "SteadyStateInitial",
		SteadyState "SteadyState")"definition of balance equations";
	
	parameter Types.Dynamics energyDynamics=Dynamics.DynamicFreeInitial 
		// only two alternatives are allowed, so restrict the choices at the GUI to the both
		annotation(choices(
			choice= Dynamics.DynamicFreeInitial,
			choice= Dynamics.FixedInitial));

```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/559

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 28 Sep 2011 08:28 UTC**
The idea (using choices to restrict enumerations to avoid having too many enumerations types) looks promising, but to me this looks like changes in the library and not in the spec and thus changing milestone.

I assume that illegal settings (e.g. energyDynamics=Dynamics.SteadyStateInitial) will be handled by an assert in the library.

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 24 Feb 2012 14:06 UTC**
As previously indicated this seems like a library issue and not a a specification issue.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Mar 2012 14:39 UTC**
Feasible for MSL 3.2.1?

modelica-trac-importer, 2017-01-14:
**Comment by fcasella on 5 Mar 2013 22:54 UTC**
As of March 5 2013, the only case in Modelica.Fluid where this might apply is Types.ModelStructure and Types.ModelStructureReduced.

How this might be fixed will depend on how we eventually implement the specialized pipe components

modelica-trac-importer, 2017-01-14:
**Changelog removed by fcasella on 5 Mar 2013 22:54 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by otter on 19 Apr 2013 08:17 UTC**
Since this is a tiny issue and seems to appear only once in MSL, it seems to be not worth to spend effort on it now.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 19 Apr 2013 08:21 UTC**
Well but maybe we can implement it in the future.

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 19 Apr 2013 16:07 UTC**
There are three parts here:
* The general idea of avoiding new enumerations by reusing an existing one with choices-annotation to limit the intended choices.
* The specific case of using this in a specific place in MSL.
* Is this ok according to the specification?

In reverse order:

Yes, it is ok according to the spec as far as I can see.

The specific case cannot easily be added after MSL 3.2.1 since it seems it breaks backward compatibility (changing the type of a parameter).

The general idea can be done at any point and is not connected to any specific component of MSL and it can also be done multiple times etc - i.e. it is not suitable to track the general idea with a ticket.

modelica-trac-importer, 2017-01-14:
**Comment by otter on 13 Dec 2015 18:44 UTC**
Since this would be a non-backwards compatible changes, change the milestone

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 15 Dec 2015 10:55 UTC**
I would re-close this, since the current specification allows choices as a GUI-restriction - but changing an existing enumeration is not fully backwards compatible.
Each specific enumeration getting such a choices-annotation deserves its own ticket.

beutlich, 2019-08-18:
> I would re-close this

Ok.

closed 2019-08-18 by beutlich