modelica/ModelicaStandardLibrary #575: Check of `ControlledPump` fails

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 25 Jun 2011 12:32 UTC**
When "checking" the MSL I noticed that the model **`Modelica.Fluid.Machines.ControlledPump`** will fail with the following error message (generated by Dymola 2012):
```
Checking model Modelica.Fluid.Machines.ControlledPump:
Error: Component type specifier BaseProperties is a partial type.
  File: /home/dietmarw/.workspace/Modelica.git/Modelica/Fluid/Interfaces.mo, line 590
  Component context: Modelica.Fluid.Machines.BaseClasses.PartialPump.medium
  Component declared as Medium.BaseProperties medium in Modelica.Media.Interfaces.PartialMedium
Error: No match for overloaded binary operator Real '*' Modelica.Fluid.Machines.ControlledPump
Error: No match for overloaded binary operator Integer '*' Modelica.Fluid.Machines.ControlledPump
Error: No match for overloaded binary operator Real '*' Modelica.Fluid.Machines.ControlledPump
Error: No match for overloaded binary operator Integer '*' Modelica.Fluid.Machines.ControlledPump
Error: No match for overloaded binary operator Real '*' Modelica.Fluid.Machines.ControlledPump
Error: No match for overloaded binary operator Integer '*' Modelica.Fluid.Machines.ControlledPump
 
Error: No match for overloaded binary operatorNo match for overloaded binary operator
No match for overloaded binary operatorNo match for overloaded binary operator
No match for overloaded binary operatorNo match for overloaded binary operator
in the equation
   equation
if ( not checkValve) then 
  head = (N/N_nominal)^2*flowCharacteristic_Unique'"0x22397420"'(V_flow_single*(N_nominal/N), {0, , 1.5*}, {2*, , 0});
  s = 0;
elseif (s > 0) then 
  head = (N/N_nominal)^2*flowCharacteristic_Unique'"0x22397420"'(V_flow_single*(N_nominal/N), {0, , 1.5*}, {2*, , 0});
  V_flow_single = s*unitMassFlowRate/rho;
else
  head = (N/N_nominal)^2*flowCharacteristic_Unique'"0x22397420"'(0, {0, , 1.5*}, {2*, , 0})-s*unitHead;
  V_flow_single = 0;
end if;
 
Missing declarations.
Check aborted.
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/575

modelica-trac-importer, 2017-01-14:
**Comment by per&#x2e;ostlund on 9 Nov 2011 13:59 UTC**
The first error, that BaseProperties is partial, also affects Modelica.Fluid.Vessels.ClosedVolume and the pump models in Modelica.Fluid.Machines. This is because they try to extend PartialLumpedVolume without redeclaring Medium to something, so it defaults to PartialMedium.

modelica-trac-importer, 2017-01-14:
**Changelog removed by dietmarw on 17 Dec 2012 13:28 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by rfranke on 18 Dec 2012 08:46 UTC**
The check passes without error in Dymola 2013 FD01 with default settings. Which settings have been used to produce the error?

Dymola 2013 FD01 in pedandic mode raises 15 errors. They are all related to the fact that PartialMedium is a partial package. "Component type specifier xxx is from a partial package. That is not correct."

Of course a non-partial medium needs to be defined when using the component model. But what can be done in the component model itself offering a replaceable medium package?

modelica-trac-importer, 2017-01-14:
**Comment by hubertus on 18 Dec 2012 09:55 UTC**
This is probably connected to #924, please wait until that is fixed.

modelica-trac-importer, 2017-01-14:
**Modified by rfranke on 3 Jan 2013 09:44 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by rfranke on 3 Jan 2013 09:55 UTC**
This seemed to be a bug in Dymola 2012. The error message reports on missing binary operators '\*' for a call to flowCharacteristics with the arguments
```
V_flow_nominal = {0, , 1.5*},
head_nominal = {2*, , 0}
```
The model ControlledPump specifies the default flow characteristics:
```
V_flow_nominal = {0, V_flow_op, 1.5*V_flow_op},
head_nominal = {2*head_op, head_op, 0});
```

closed 2017-01-14 by modelica-trac-importer

joedenlys, 2019-03-29:
It's a hazard which you must want to solve, [itunes error 0xe80000a](https://itunessupport.org/blog/fix-itunes-error-0xe80000a/) helped me to solve this issue so far.