modelica/ModelicaStandardLibrary #576: Move `Rn` and `Rp` models into the `Utilities` package

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 27 Jun 2011 08:25 UTC**
Currently the example model `Modelica.Electrical.Analog.Examples.CauerLowPassSC` contains two submodels 
 * `Rn`
 * `Rp`

It's confusing for some users that a model can contain other models. when normally packages are used for this purpose. Since `Modelica.Electrical.Analog.Examples` already contains a `Utilities` subpackage I would recommend moving those two models `Rn` and `Rp` into the sub-package `Modelica.Electrical.Analog.Examples.CauerLowPassSC.Utilities` for the next non-backward compatible version of MSL and provide the proper conversion script.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/576

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

kristinmajetta, 2018-05-03:
It might be confusing (which I doubt), but it also could help to show the users the wide range of possibilities they have when creating a model. Therefore I would prefer to not move the submodels into a sub-package. Or is it a guideline of MSL?

dietmarw, 2018-05-03:
Models within models are always confusing. Especially since we teach people to put models into packages not into models. MSL being a role model here should not really advertise this.

HansOlsson, 2018-05-03:
If they are models within models instead of models in packages they cannot be used anywhere else.

I don't know if it makes sense to re-use these models in other contexts - but I think that could influence the decision on whether to move them: either encourage reuse by having them in Utilities - or prevent misuse by keeping them local to the model.

dietmarw, 2018-05-03:
My suggestion is to make an improved version by creating **one** component that is called `SwitchedCapacitor` and place that under `Utilities`. This new `SwitchedCapacitor` can then be used instead of `Rp` and `Rn` but accept a resistance value that can be negative and will hence change the nature of the `SwitchedCapacitor` from positive to negative resistance. This can also fix the connectors which are kind of wrong because they use "n" for both sides.

[image]

PR #2533 (2018-05-18): Fixes #576 by replacing internal models by one utility model
Instead of having the no-accessible models `rn` and `rp` inside the example I've created **one**  universally usable switched capacitor model instead. 

Note: When replacing `rn` and `rp` in `Modelica.Electrical.Analog.Examples.CauerLowPassSC` I noticed that this and the other examples are in dire need of an clean up since the coding style is sub fortunate (lots of unnecessary protected connectors with no real purpose). 

Close #576.
commit e41a9e20 (2018-05-18): Fixes #576 by replacing internal models by one utility model (#2533)

* Fixes #576 by replacing internal models by one utility model

Instead of having the no-accessible models `rn` and `rp` inside the example I've created **one**  universally usable switched capacitor model instead. 

Note: When replacing `rn` and `rp` in `Modelica.Electrical.Analog.Examples.CauerLowPassSC` I noticed that this and the other examples are in dire need of an clean up since the coding style is sub fortunate (lots of unnecessary protected connectors with no real purpose).

* Add helping constant in order to satisf [... cut]
closed 2018-05-18 by beutlich