modelica/ModelicaStandardLibrary #608: Remove unneeded function `bjtRenameParametersType`

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 13 Sep 2011 11:16 UTC**
In 33337aea75ee77ea9ada61a0dfaf4ef5db32bffc the function `bjtRenameParametersType` was removed. But since this removal  is a non-backward compatible change (user models might allready use that function) the removal was reverted in 00580861925fd409b177a5cf5c3e564d0a76dd82. 

The function should be removed for the next non-backward compatible version of the MSL which also needs to supply a conversion script.

<sub>Keep this ticket as a reminder of that task.</sub>

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/608

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2019-07-13:
Also addressed by #3017.

PR #3017 (2019-07-24): Remove all obsolete classes from Modelica.Electrical.Spice3.Internal
There is neither conversion provided nor possible. Instead, a conversion message is raised, if internal classes of Spice3 are to be converted.

There is one backward-compatibility breaking change in Modelica.Electrical.Spice3.Internal.MOS2, since the final parameter vp was declared as return value of the (obsolete and now removed) function Modelica.Electrical.Spice3.Internal.Mos2.mos2ModelLineParamsInitEquations, but mos2ModelLineParamsInitEquationsRevised has a different return type. In the current PR I simply decided to rem [... cut]
closed 2019-07-24 by beutlich