modelica/ModelicaStandardLibrary #609: Rename models that do not follow the naming convention

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 14 Sep 2011 14:05 UTC**
Looking at the MSL there exist several models that don't quite follow the naming convention as pointed out in `Modelica.UsersGuide.Conventions.ModelicaCode.Naming`

So far I found that the following models need to be corrected once a new version with conversion script will be released:
```
Modelica.Blocks.Interfaces.partial*
```

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/609

modelica-trac-importer, 2017-01-14:
**Comment by otter on 30 Mar 2013 14:39 UTC**
The attachment "classeswithclosescases.txt" is no longer present

modelica-trac-importer, 2017-01-14:
**Changelog removed by otter on 30 Mar 2013 14:39 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Apr 2013 12:51 UTC**
Attachment recovered.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

dietmarw, 2018-05-16:
Based on the discussion in #2478 it perhaps is a good idea to go away from the `partial` prefix all together and rather call those base classes with a `Base´ postfix.

HansOlsson, 2018-05-16:
> Based on the discussion in #2478, Icon style guide: missing information on libraries' colors it perhaps is a good idea to go away from the partial prefix all together and rather call those base classes with a `Base´ postfix.

I agree that the current "Partial" prefix is not ideal, but I am not sure if Base-prefix is always the correct solution - even if it often is an improvement. Basically "XBase" may sound as the base-class for "X"-functionality, whereas "PartialX" sounds as a partial "X" and they are not necessarily the same. However, I don't have a better alternative, and e.g. "XInterface" would be worse (since that would imply that it were only an interface). We might also skip some of the changes (in particular PartialMedium) if we view it as too impactful for users (even with conversion scripts) - compared to the gain.

BTW: We also have to give better name to Modelica.Mechanics.MultiBody.Sensors.Internal.PartialCutForceBaseSensor and Modelica.Mechanics.MultiBody.Sensors.Internal.PartialCutForceSensor

dietmarw, 2018-05-16:
That sounds like a clean solution if applied. I wonder how one then would distinct this from the `Modelica.Icons.InternalPackage` which also is often use to collect models that should not be used directly.

HansOlsson, 2018-05-16:
To me InternalPackage and BasePackage are different.
BasePackage contain partial models - they can be used by users directly as base-classes - but clearly not as components in simulation models.

 InternalPackage should not be used directly by users - at all; neither as base-classes nor as models for components.

Models from both packages can be used indirectly by users.

MartinOtter, 2018-05-17:
Hm. I do not really understand the discussion about "PartialXXX" and "XBase". These are different things. "PartialXXX" clearly indicates in its model name, that it is a partial class and the only useful operation on it is inheritance (and using it as a base class of a replaceable component). I find this very useful information: So it makes no sense to drag a "PartialXXX" from the package browser into the model, whereas all other components in the package browser can be dragged in a model!
XBase just means it is a building block to build higher level models. An "XBase" model needs not to be a partial class.

HansOlsson, 2018-05-18:
There are multiple variants being discussed here. 

One is naming "PartialX", "XBase" or ..., and another is whether the name of the class should contain this information at all - or whether it should be implicit from the package (and then it is a matter of the name of that package).

In many cases it is more implicit - in e.g. Modelica.Electrical.Analog.Interfaces we have partial models and non-partial connectors. Those model names don't contain "Partial" or "Base" - but it is still clear from context that they are partial models - and one could even guess this from names such as "TwoPin".

Modelica.Mechanics.MultiBody.Interfaces is different - all the partial models have "Partial" in their name, and there is also one non-partial model. However, even without "Partial" one would guess that "TwoFlanges" is partial.

Similarly Modelica.Mechanics.MultiBody.Sensors.Internal could be separated into a sub-package for the base-classes - and thus remove the need for "Partial" in "PartialAbsoluteSensor"

This also made me realize that we have both
Modelica.Mechanics.MultiBody.Interfaces.PartialAbsoluteSensor and Modelica.Mechanics.MultiBody.Sensors.Internal.PartialAbsoluteSensor - and it is the Internal-variant that is used in models, whereas the one intended to be used by users isn't used.

I believe the internal-one is cleaner and that we should replace 
Modelica.Mechanics.MultiBody.Interfaces.PartialAbsoluteSensor by Modelica.Mechanics.MultiBody.Sensors.Internal.PartialAbsoluteSensor; (and similarly for other sensors).

beutlich, 2019-11-04:
I believe we cannot get this sorted out for MSL v4.0.0. Therefore, removing milestone.