modelica/ModelicaStandardLibrary #665: Homotopy for Modelica.Fluid---Wall friction in pipe models

modelica-trac-importer, 2017-01-14:
**Reported by msielemann on 15 Dec 2011 14:13 UTC**
In order to implement homotopy in Modelica Fluid, the pipe models shall be modified on the branch 

https://svn.modelica.org/projects/Modelica/branches/development/Homotopy3.3-dev

This shall be done along the ideas presented in

https://www.modelica.org/events/modelica2011/Proceedings/pages/papers/04_2_ID_131_a_fv.pdf

Please make the modifications and implement two or more test models in ModelicaTest.Fluid

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/665

modelica-trac-importer, 2017-01-14:
**Comment by msielemann on 15 Dec 2011 14:16 UTC**
This ticket should cover both wall friction and heat transfer

modelica-trac-importer, 2017-01-14:
**Comment by kilian_link on 7 Mar 2012 09:56 UTC**
I checked in the following changes:

pressure loss:
- in Fluid.Pipes.BaseClasses.FlowModels.PartialGenericPipeFlow:
    used homotopy 4 times to compute the mass flow / pressure loss not with the corresponding exact function, but with given nominal values, e.g. `homotopy(..., m_flow_nominal/dp_nominal*(dps_fg - {g*dheights[i]*rho_nominal for i in 1:n-1}))`
- in Fluid.Pipes.BaseClasses.FlowModels.NominalLaminarFlow
    used homotopy 2 times to compute the static head pressure loss with the nominal density instead of the actual density, e.g. `homotopy(..., {g*dheights[i]*rho_nominal for i in 1:n-1})`

 
heattransfer: 
 - in Fluid.Pipes.BaseClasses.HeatTransfer.PartialPipeFlowHeatTransfer
  used homotopy to compute the Q_flows with a constant alpha value (alpha0) instead of the exact one, e.g. `homotopy(alphas[i], alpha0)`

I also added 2 testcases under Fluid.Examples.PipeHomotopy, ran them with and without homotopy (set the flags Advanced.OnlyUseHomotopyMethod and Advanced.SupportHomotopyFunction to true or false accordingly) and both of them only succeed when using homotopy.

modelica-trac-importer, 2017-01-14:
**Comment by kilian_link on 17 Apr 2012 11:58 UTC**
Since there were some problems with the two testcases (they could not be initialized at all), we figured out that it is very important that the flag Evaluate = true is set in order to succeed.

Additionally the second example (PipesWithHeattransfer) has been updated and now should work as stated in the comment above (initialization only possible if homotopy is used).

modelica-trac-importer, 2017-01-14:
**Comment by pjordan on 24 Apr 2012 09:41 UTC**
The second test case `PipesWithHeatTransfer` is still sensitive to increased heat flow rates at the boundaries. For example, it will fail if `heat1.Q_flow` = 2.5 \* 200e3. Convergence is better if the nominal values for pressure loss and mass flow rate in the `FlowModel` are set to the correct order of magnitude (e.g. `m_flow_nominal`=10, `dp_nominal`=0.05e5).

modelica-trac-importer, 2017-01-14:
**Comment by fcasella on 14 Dec 2015 23:00 UTC**
Accepted tickets

modelica-trac-importer, 2017-01-14:
**Comment by fcasella on 14 Dec 2015 23:08 UTC**
Not enough time to handle this for MSL 3.2.2 and test properly. Rescheduled to next minor revision

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 20 Dec 2016 19:50 UTC**
Unhide in agreement with msielemann and otter.

casella, 2019-03-20:
Adding homotopy may require some new parameters, e.g. for nominal mass flow rate, which were not originally present in the model. We could give a start value to them, with a warning like "This parameter was introduced in v. 4.0.0, please put an appropriate value to it to ensure proper operation of the model". In most cases the model will work anyway, and in all cases there's proper diagnostics to fix the issue.

The same applies to #666, #667, #668, #669