modelica/ModelicaStandardLibrary #671: Different icon sizes of Block connectors (input vs output)

modelica-trac-importer, 2017-01-14:
**Reported by petar on 20 Dec 2011 11:42 UTC**
The input icons in Blocks.Interfaces are much larger than the output icons. The reason for this according to Martin Otter is to resemble how it looks like in Matlab, where the output connector is removed when the connection is made.

Alternative solutions 
copied from email-conversation- Johan Andreasson): 
A bit off the original topic: I guess an alternative way to achieve Simulink look-alike would be to have small icons for both input and output, and have the connection line as arrow instead. This would make scaling when mixing smaller and larger blocks much nicer (especially when the block are not scaled proportionally in x and y). Maybe it could be similar to how (at least) Dymola let you affect the thickness and color of the connection line. The directions would then have to be handled.

(copied from email-conversation- Martin Otter):

 * A general annoying issue is that connector sizes are changed, if the icon size is changed. This holds for all types of connectors. This is a Modelica issue, and one would need to improve the graphical annotations, so that a point in the connector is specially marked and the relative relation to the unscaled coordinate system is defined. When the size of the icon is changed, then the size of the connector is not changed, and only the specially marked point in the connector is appropriately moved (and therefore the connector icon).
 * How to get a signal arrow? Whether the proposal from Johan works needs to be evaluated. Unfortunately, arrows are not precisely defined in the Modelica graphical annotations. It is therefore a tool issue how this looks like. When I tested this feature (having a line with an arrow) the last time in Dymola, it was still not practical to use, because depending on the environment, the arrow was either too small (in diagram layer) or too large (in generated html documentation). Therefore, I never use graphical annotation arrows. When needed, I construct them by myself with polylines. Even if this would work in Dymola, one cannot be sure that this works in other tools reasonably. Therefore, in order to improve this situation, one should also improve the graphical annotations (and define arrows precisely).

(copied from email-conversation - Mike Tiller)
For the first point, one could define two annotations for a connector.  One representing the unconnected appearance and the other representing its appearance if there is at least one connection made to it.  It seems like this could be generally useful (particularly for connectors that MUST be connected) and it would be mostly backward compatible as well.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/671

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 20 Dec 2011 11:48 UTC**
I am not clear about exactly what is meant with changing the icon size in the first issue.

Does this refer to:
1.	Changing the icon size for the class – which is sometimes done in order to fit more into the drawing area?
2.	Changing the icon size for the component.
For the first case I agree that connector sizes should not be changed; and an operation that increases the size of the drawing area without increasing the size of the connectors seems possible. However, I believe the existing origin-handling in Modelica could suffice for this, and it is mostly a tool issue - since there is a UI operation for changing the size.

For the second case it would work for minor changes in size – but when you shrink it more the connectors would overlap. Handling that becomes messy, and instead of manually changing sizes of icons it seems better to modify the default size of the icon. This is partially standarized in Modelica using the attribute initialScale. 

So to me the remaining part of standardization for that is to more fully specify initialScale and note that changing initialScale should be viewed as changing the drawing area of the icon (as discussed above).

modelica-trac-importer, 2017-01-14:
**Comment by leo&#x2e;gall on 20 Dec 2011 12:30 UTC**
I like Michael Tiller's proposal. Possibly it could be used to solve two issues I have with connector graphics:

1. **Visualization of vectorized connections**

   This could be done by an alternative connector icon, which appears for vectorized connections.

   Use case 1: connection to the vectorized heat port of a discretized pipe model. Currently, there is no indication on the diagram, that there is a vector of connects.

   Use case 2: connectors like Modelica.Fluid.Interfaces.FluidPorts_a. This connector could become obsolete and be replaced by a version of Modelica.Fluid.Interfaces.FluidPort_a which changes appearance.

2. **Visualization of connections which do not have graphical annotations**

   There are good reasons for mixing graphical and textual modeling. For example for-loops of connects. The annoying part is, that there is no graphical hint on connects which do not have graphical annotations. As far as I understood, this point was also on the list of requirements of japanese car manufactures presented at Design Meeting 73.

   One solution could be an annotation for connect equations like "markConnectionsOnDiagram". 
   If markConnectionsOnDiagram=true, the affected connectors could change to a representation making clear that there are connections happening.

modelica-trac-importer, 2017-01-14:
**Comment by mtiller on 20 Dec 2011 13:29 UTC**
I have a comment on my comment. :-)

I think we need to take a step back and look at our requirements here.  We already have the ability to have a different location in both the icon and diagram layer.  I proposed that we have the ability to specify appearance for both connected and unconnected connectors.  Leo points out two additional use cases, vectorized connectors and appearance related to non-graphical connections.  I would extend Leo's second use case to include things like "connections in layers" (to avoid having all connections crossing over a diagram at once) or "translucent" connections that are translucent normally but opaque if you mouse over them or even "from-goto" style connections.  These are in some ways all variations on a theme.

Is there a way to capture all these use cases in a consistent way.  Yes, we could define different annotations like "diagramWhenConnected", "diagramWhenNotConnected", "iconWhenConnected" "iconWhenNotConnected", "diagramWhenVectorConnected", "diagramWhenVectorNotConnected", etc.

But hopefully everyone sees the potential issue with that.  So here are my comments:

  1. I think for vectorized connectors it is simple enough to simply draw a "triple" of overlapping connectors (or some similar notation) to indicate the plurality of connectors.  In this case, its more a tool issue.  In other words, I'm not convinced that we need to change the appearance for this case.  Maybe I'm wrong?
  2. I think we need to think in terms of "minimum" annotations (icon only, all other representations duplicate the icon) and "maximum" annotations (all possible combinations, whatever they prove to be).  In other words, it shouldn't be necessary for people to specify ALL combinations.  In fact, it should be possible to get by with just one and assume all others look like the icon.  Then have the ability to selectively "override" different cases.
  3. At the moment, we think of Modelica models as having an "icon layer" and a "diagram layer".  But this points out that we might need something richer than this.  Sure, adding another annotation isn't a big deal in the language...but what is the UI going to be?
  4. This is all specific to connectors, right?  It's a little ugly to make this special annotations only apply to connectors but I think it is even uglier to abstract it to apply to records, models, functions, etc.
  5. I cannot help but think that a more general capability like CSS (cascading style sheets) is really what is required here (perhaps in Modelica 4?).  It opens up a huge range of possibilities to the library developer (I'm thinking of things like "a:hover" in CSS).

Perhaps the simplest thing for now is to have this be one new optional annotation for now and address the wide range of use cases with something like item 5 above in Modelica 4?

christiankral, 2019-02-10:
In the Modelica Standard Library (MSL) we have even cases of different sizes of input and output connectors:

[image]

- `integerPassThrough` connector size of most block in the MSL
- `triggeredAdd` one Integer input has a different size than the other Interger input and the Boolean inputs
- `rising` the input and the output connectors are bigger than the standard size

beutlich, 2019-10-23:
@HansOlsson Beside the inconsistent appearance in current MSL, there are so many valuable proposals and use-cases on connect lines and connector icons that this issue rather belongs to the Modelica language specification.

HansOlsson, 2019-10-30:
> @HansOlsson Beside the inconsistent appearance in current MSL, there are so many valuable proposals and use-cases on connect lines and connector icons that this issue rather belongs to the Modelica language specification.

Ok, but I don't see that we will be able to handle them quickly. Note that we also must consider the user-interaction in these cases; even if the output connector isn't drawn you should still be able to connect as if it were there (connecting to the existing connection line is a possibility in e.g. Dymola; but it is not as convenient).