modelica/ModelicaStandardLibrary #682: Modelica.Math.BooleanVectors.allTrue gives wrong result when size(b,1)=0

modelica-trac-importer, 2017-01-14:
**Reported by jriel on 12 Jan 2012 20:09 UTC**
When size(b,1) of allTrue is 0, meaning there are no inputs, the output should be true, not false.
That is, the function allTrue(), with no inputs, should return true.  One way to see this is that
allTrue(x1,x2,...,xn) = allTrue(allTrue(x1,...,xm), allTrue(xm+1,..,xn)).  So 
allTrue(x) = allTrue(allTrue(x),allTrue()) which returns x only if allTrue() = true.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/682

modelica-trac-importer, 2017-01-14:
**Comment by otter on 9 Dec 2012 15:28 UTC**
You are right that the semantics seems to be more useful if allTrue returns true for an empty vectors. Currently, this is not the case and this is also defined in this way in the documentation. Therefore, a backwards compatible change is not possible currently.

To summarize, a change of the semantics of this function requires a conversion script and therefore the ticket milestone is changed to "MSL with conversion script".

modelica-trac-importer, 2017-01-14:
**Changelog removed by otter on 9 Dec 2012 15:28 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

HansOlsson, 2019-03-20:
Regarding the function in conversion scripts:

- The only current solution for conversion scripts is to move current one to ObsoleteModelica4 - and introduce a new one. That works, but in case the argument is never zero-sized that is excessive - and in case they mistakenly thought it would work the other way it is problematic.
- Automatically detecting if the function argument may be zero-sized is both difficult and something that would require an advanced extension of conversion-script.
- However, it is also the basis for two blocks: And and Nand - I assume they should also be changed in the same way. In that case it seems as if we could handle the smart conversion with current conversions. There is also another issue for Nand.

PR #2916 (2019-05-28): Add andTrue (returning true on empty input vector)
This PR supersedes #2888 that was closed by the library officer.

Closes #682, #2860.
closed 2019-05-28 by beutlich