modelica/ModelicaStandardLibrary #694: Reference temperature in resistors

modelica-trac-importer, 2017-01-14:
**Reported by m&#46;ceraolo on 2 Feb 2012 08:04 UTC**
In Modelica.Electrical.Analog.Basic.Resistor the resitor temperature T_ref is set as default to 300.15 kelvin.
However in the ConditionalHeatPort it uses, the HeatPort temperature in case HeatPort=false, is set to 293.15 kelvin.

According to my tests, this never causes troubles; but it is still a minor glitch.
Is it possibile to choose one value (I suggest 293.15) for both variables?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/694

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 2 Feb 2012 09:17 UTC**
Christoph Clauss: 
 Thank you for reporting that. We will check it, that means find out the reasons for these different values.

modelica-trac-importer, 2017-01-14:
**Comment by clauss on 6 Jun 2012 07:01 UTC**
T_ref is generally set to 300.15 kelvin in Modelica.Electrical.Analog.
The reason is that SPICE uses 27 Celsius (300 kelvin) as default reference temperature.
To be precise, we use 300.15 kelvin.

From my point of view the ticket could be closed.

modelica-trac-importer, 2017-01-14:
**Modified by clauss on 6 Jun 2012 07:03 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 6 Jun 2012 09:01 UTC**
Fixed in aabf47fa9be790ac66e2cd5ce81b136c3d5b26a5.

modelica-trac-importer, 2017-01-14:
**Comment by ahaumer on 8 Jun 2012 07:16 UTC**
Please, for now leave it at 300.15 K for backwards compatibility not to break existing user models. As soon as a conversion script is used for a new version of MSL, please change the default to 293.15 K. Reference values are normally given at 20 degC, 27 degC is a source of errors! If Spice3 needs 27 degC as default, please use appropriate extends statements with (Tref=300.15).

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 8 Jun 2012 07:43 UTC**
Toni, well spotted. I reverted the change in 8d9f01b8b5d8c9a576475ad5123a61287547c74a and will reschedule the ticket to the next Milestone with conversion script (most likely [milestone:MSL4.0])

modelica-trac-importer, 2017-01-14:
**Comment by otter on 18 Jul 2013 12:11 UTC**
This is not a defect, there are just different opinions about a default reference value in different communities. Therefore changed to "discussion". Furthermore, the model is correct therefore changed Severity to "minor"

modelica-trac-importer, 2017-01-14:
**Changelog removed by otter on 18 Jul 2013 12:11 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

beutlich, 2017-01-16:
Point to @ceraolo.

christophclauss, 2019-08-04:
I'd like solve this issue in connection with #694, therefore postpone it beyond MSL4.0.0.