modelica/ModelicaStandardLibrary #712: Revise/revisit all SI-units listed in `Modelica.SIunits`

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 12 Mar 2012 12:10 UTC**
The current version of `Modelica.SIunits` is based on quote

> Library of type and unit definitions based on SI units according to ISO 31-1992

The problem is that this ISO norm is deprecated by [ ISO 80000-1:2009](http://www.iso.org/iso/iso_catalogue/catalogue_ics/catalogue_detail_ics.htm?csnumber=30669). 

In addition to that has the current `Modelica.SIunts` been steadily "enhanced" by other library officers with unit names that are not defined in the ISO norm but where just _convenient_ to use. There should be a proper rule on what can be added and what not.

Related tickets: #545 (Molar units missing), #688 (Add Complex Units (Magnetic))

I'm scheduling this already for [milestone:MSL3.2.1] so that we can evaluate the new units that have been added recently. The main overhaul should probably be done in connection with [milestone:MSL4.0]

Related tickets: #545 (Molar units missing), #688 (Add Complex Units (Magnetic))

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/712

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 17 Apr 2012 13:24 UTC**
To me it seems that if Modelica.SIunits is not based on the SI standard we should consider using another name. 

I know that one can argue that the other units are based on the SI-base units, but the same should be true for all models in the entire MSL, except for:
```
Modelica.SIunits.Conversions.NonSIunits
```
which does not seem like the most intuitive name.

The new standard is possibly not that important, I would assume it is just a slight revision with possibly some changed names etc; i.e. not a change with major impact.

modelica-trac-importer, 2017-01-14:
**Comment by wischhusen on 17 Apr 2012 14:19 UTC**
Perhaps one could stay with Modelica.SIunits but open another subpackage Modelica.SIunits.Derived to account for units that were derived from the standard unit system with multipliers. I think, it is useful to add somewhere in the Modelica Standard Library really commonly used units for many reasons (e. g. there is still no length unit in feet ... ). I know that this can (and is) already be done with local settings on each single machine but is that reasonable?

So what about that proposal:

Create a new subpackage Modelica.SIunits.Derived and put all those types inside which are really used by scientist and engineers :-).

modelica-trac-importer, 2017-01-14:
**Comment by mtiller on 17 Apr 2012 15:26 UTC**
The biggest issue I see here is with the number of potential imports that users are going to have to deal with here.  Consider for the moment that users are going to need to know, for a given unit, whether it is formally in the ISO specification or not.  Is this really practical?

Also, keep in mind that "SI" is not universally synonymous with the ISO specification.  In fact, the Wikipedia page (in English) doesn't even mention the ISO specification until half way through the article and it lists it under "Related systems".

It seems to me that the real requirement here is that we give users a way to determine whether a given type is part of the ISO standard or not.  As I said, my feeling is that most will not care.  But for those who do, it should be possible.

One could imagine including a descriptive string on each unit indicating "ISO" or "derived".  You could further add an annotation (which could even specify what version of an ISO standard it was pulled from and the name in that standard if it was different).  Frankly, it might be sufficient to just indicate derived units only.

Frankly, I'm not sure it is worth it to go through all this.  However, I do agree that we should have a process for determine what types go into the SIunits package and making sure that any additional units are properly documented.

modelica-trac-importer, 2017-01-14:
**Comment by wischhusen on 17 Apr 2012 15:54 UTC**
Replying to [comment:3 mtiller]:
> The biggest issue I see here is with the number of potential imports that users are going to have to deal with here.  Consider for the moment that users are going to need to know, for a given unit, whether it is formally in the ISO specification or not.  Is this really practical?

Right, this is not a good solution from that point of view.

> 
> Also, keep in mind that "SI" is not universally synonymous with the ISO specification.  In fact, the Wikipedia page (in English) doesn't even mention the ISO specification until half way through the article and it lists it under "Related systems".

Okay, so that means derived units could stay inside the Modelica.SIunits package. This would be fine for me.

> It seems to me that the real requirement here is that we give users a way to determine whether a given type is part of the ISO standard or not.  As I said, my feeling is that most will not care. [...]

I think units are the base of every model, parameter dialog and result plot. It's important to offer those units which are really accepted in Modelica using (and perhaps also future!) technical domains. The Modelcia Standard Library could improve here in my opinion. I agree absolutely that there should be a process to integrate new units. How should that be done? Potentially on the Design Meetings, right?

modelica-trac-importer, 2017-01-14:
**Comment by pharman on 18 Apr 2012 09:11 UTC**
Rather than have another package for derived units it would perhaps be neater to have a Units package in each subpackage (such as `Modelica.Mechanics.Translational.Units`) containing units that are useful in that particular domain.

A related issue that this would help with is that of nominal values. It is useful to use nominal values for scaling, but the default values are always 1, units in a domain-specific library can extend the base units and supply a nominal value that is closer to typical values (such as `type HydraulicPressure = SI.Pressure(nominal=1e5);`).

modelica-trac-importer, 2017-01-14:
**Comment by mtiller on 18 Apr 2012 09:43 UTC**
This idea of per-library Units packages is interesting.  But I don't think it is very practical.  We will end up importing lots of package contents from different places and we'll still have a lot of left over units that have to go somewhere (since I think we should include at least the ISO standard units, esoteric as they may be).

However, you raise an interesting point about the nominal values.  I think perhaps a good compromise might be to put everything in Modelica.SIunits but then create per library units packages for that library to use where domain specific attributes (min, max, start, nominal) could all be associated with it.  Then the library could extend from that type.

This would achieve the objective of having domain specific units used within the library while still preserving the aggregated units package.

modelica-trac-importer, 2017-01-14:
**Comment by pharman on 18 Apr 2012 10:06 UTC**
Replying to [comment:6 mtiller]:
> I think perhaps a good compromise might be to put everything in Modelica.SIunits but then create per library units packages for that library to use where domain specific attributes (min, max, start, nominal) could all be associated with it.  Then the library could extend from that type.

Yes I meant the base units would remain in SIunits, maybe it should be renamed Modelica.Units at some point (though the resulting update script might be quite big!).

modelica-trac-importer, 2017-01-14:
**Comment by fcasella on 18 Apr 2012 13:40 UTC**
Replying to [comment:6 mtiller]:
> This idea of per-library Units packages is interesting.  But I don't think it is very practical.  We will end up importing lots of package contents from different places and we'll still have a lot [...]

Note that a similar concept has already been implemented long time ago in Modelica.Media: each fluid model is described by a package, which has local unit definitions with appropriate min, max, start, and nominal attributes. These can also be modified locally when using the medium package in a specific modelling context.

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 18 Apr 2012 14:27 UTC**
Having per library types with specific min, max, start and nominal in **non-Media** parts in **MSL** does not seem ideal to me, since the nominal values are more application-specific than domain specific.

E.g. the translational library is used for 1d-motion; it could be an entire vehicle or it could the parts of the hydraulic system of a vehicle - the equations are the same, but the nominal values differ by several orders of magnitude.

Having the possibility to modify it locally as for the media-library does work; but it requires modifying packages which seems excessive for the other parts of MSL.
\--
Oh, and renaming "Modelica.SIunits" to "Modelica.Units" is just one line a conversion script.

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 17 Dec 2012 13:50 UTC**
Find a proper rule for units for MSL3.3 (decided at the MAP-LIB phone meeting 2012-12-17)

modelica-trac-importer, 2017-01-14:
**Changelog removed by dietmarw on 17 Dec 2012 13:50 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 29 Apr 2014 06:34 UTC**
@dietmarw: Should Milestone be set to MSL_withConversionScript?

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 29 Apr 2014 06:45 UTC**
Well the decision was to include it in MSL3.3 which means there will have to be a conversion script for that version. I'd like to leave this at MSL3.3 as an explicit decided statement.

modelica-trac-importer, 2017-01-14:
**Comment by beutlich on 29 Apr 2014 06:48 UTC**
But there is no Milestone MSL3.3.

modelica-trac-importer, 2017-01-14:
**Comment by otter on 13 Dec 2015 17:14 UTC**
Replying to [ticket:712 dietmarw]:
> The current version of `Modelica.SIunits` is based on quote
>
>
> The problem is that this ISO norm is deprecated by [ ISO [...]

I agree that this should be clarified. However, I have no time to inspect and compare the newest SI unit ISO norm with the older one. I therefore re-schedule to the next milestone.

> 
> In addition to that has the current `Modelica.SIunts` been steadily "enhanced" by other library officers with unit names that are not defined in the ISO norm but where just _convenient_ to use. There should be a proper rule on what can be added and what not.

I made this clearer in the documentation with 95ba212c5798c588ffb0884b627abc082f247e21:
 _Some of the types are derived SI units that are utilized in package Modelica (such as ComplexCurrent, which is a complex number where both the real and imaginary part have the SI unit Ampere)._ 

I quickly checked and did not found type definitions that are neither directly defined in the ISO norm nor are derived from it. Please, can you give concrete type definitions where you see an issue (the solution is then to either move this type to Non-SIunits or improve the description of the package). I do not really understand why the name of the package should be changed (e.g. this package does not contain non-SIunits like "foot" or "rev/min" and it contains only units that can be expressed, without conversion factors, by the seven SI base units).

modelica-trac-importer, 2017-01-14:
**Comment by hansolsson on 13 Dec 2015 18:43 UTC**
Note that this ticket is about 4 years old, and in the coming 3 years we are likely to get a revised SI-system. Even if it primarily is about the basic definition of the units (like kilogram) - they are also planning awareness campaigns about the new SI-system.

An _unrelated_ major change of Modelica.SIunits close in time to that seems confusing.

dietmarw, 2019-02-19:
Since we are now at a point where we can do proper clean-ups I like to bring the idea of reorganising the `SIUnits` package  to the surface again. 

1. **Name**: I'm still pretty much in favour to make a rename from `SIunits` to simply `Units`. Reason being that: 
   - having the nonSI unit conversions as part of the **SI**units package looks strange
   - it always bothered me that the camel-case convention in `SIunits` is kind of broken. Naturally I would write SI**U**nits. 
   - when domain/3rd-party libraries specify their own units I've only seen the package name `Units` for that. So this seems to be the more natural thing to call things. 
2. **Structure**: Finding the right unit without knowing the exact name in MSL beforehand makes it hard to find. Browsing is cumbersome because of the pure quantity. So having the units split up into sub-packages by chapters might help. What I don't like about that is that you then can not do a simple 
  ```mo
 import U=Modelica.Units;
```
and access __all__ units with just short syntax like `U.xxx` but rather have to type then `U.yyy.xxx`. Doing this via a special description string could help if the tool would allow filtering the component browser based on that. Which neither OMEdit nor Dymola (only ones I have access to) do currently.

3. **Type of unit**: By this I mean is it a derived or an SI unit. For this the use of a [description string was mentioned](https://github.com/modelica/ModelicaStandardLibrary/issues/712#issuecomment-272611755) which I think would be sufficient. 

4. **Difference SI and non-SI is fuzzy**: When looking at `Modelica.SIunits.Conversions.NonSIunits` it is not clear to me why these are not SI units but rather just scaled versions of SI. But then again all units are likely to be obtained by scaling. We even have already things like `Modelica.SIunits.Temp_C` (which in turn links down to non-SI). So all this really points to 1. and just calling it `Units` and marking the ones we are certain about being SI as SI or ISO or ... and others not.

HansOlsson, 2019-02-19:
Two additional comments:

- The Modelica.SIunits.* names do actually not seem to be clearly defined by the SI-standard as far as I know, and it is fact more complicated. The names correspond to the names of "Derived quantities" - but I can only see the SI-standard listing examples for that - and not a complete list; so we don't know if many of the units should be there or not. 
- Regarding NonSI: the derived quantity list in the SI-standard includes "Celsius temperature" (for temperature difference); "cm" is a prefixed SI-unit (part of the SI-standard), "deg" (for degree) and "h" (for hour) are non-SI-units accepted for use with the SI-standard according to the SI-standard - so even if the name is counterintuitive it makes sense for them.
- It is still unclear how to use the SI-types for connector-variables.

HansOlsson, 2019-02-20:
> So having the units split up into sub-packages by chapters might help. 

It might - or it might just cause confusion since many units are common for several domains; and thus the chapters might not be logical; and thus making it harder. I guess tool vendors could experiment with some advanced support for finding appropriate units - but assumedly after the conference.

dietmarw, 2019-02-20:
Nothing in this ticket is meant to be fixed by the conference ;-)

tobolar, 2019-02-20:
>    1. **Name**: I'm still pretty much in favour to make a rename from `SIunits` to simply `Units`.

Not sure if  `Units` is sufficiently clear. [BIMP](https://www.bipm.org/en/measurement-units/) develops a __system of units of measurement__.

beutlich, 2019-07-26:
@dietmarw @HansOlsson @tobolar Shall we call a web-meeting how to go on here?

dietmarw, 2019-08-06:
Since @christiankral and @MartinOtter are the library officers of `Modelica.SIunits` they should be included here. 

For MSL4.0.0 maybe we can address the main points:
 1. Update the name from `SIunits` to `Units` since the current class name structure includes strange constructs (see above https://github.com/modelica/ModelicaStandardLibrary/issues/712#issuecomment-272611734 ) which basically means the units are simply a collection of, well units. Some are SI some are not, and some are derived.
 2. Decided if an update to ISO 80000-1:2009 is relevant. If not at least update the documentation to state why MSL is still fine with using the deprecated ISO 31 norm.
 3. Resolve the related ticket: #545 (Molar units missing) at the same time.

MartinOtter, 2019-08-07:
There are several main unit systems:
- SI (International System of Units) (meter-kilogram-second-ampere-kelvin-mole-candela)
- FPS (foot-pound-second)
- MKS (meter-kilogram-second)
- CGS (centimeter-gram-second)
- EMU (Electromagnetic) (centimeter-gram-second-abampere)
- ESU (Electrostatic) (centimeter-gram-second-abcoulomb)

The purpose of Modelica.SIunits are units based on  (meter-kilogram-second-ampere-kelvin-mole-candela), not on one of the other unit systems. Using "Units" as package name would indicate that also units of the other unit systems could be used, but this is not the intention and should not be done.

Nearly every sub-library in package Modelica has transformers to another domain, e.g., Translational.Components.IdealGearR2T transforms rotational into translational motion. Although, such mixed-domain components are present, the library is called Translational and not TranslationalWithSomeRotationalParts. Why should SIunits be an exception? SIunits.Conversions contains conversions into non SI units and in order to do this, non SI units are defined here.

To conclude, the name SIunits should not be changed.

christiankral, 2019-08-07:
I agree, SIunits shall not be changed.

sjoelund, 2019-08-07:
Is it really that much of a change to add a single `.` compared to right now if `Units.SI` was used instead? I see `import SI = Modelica.SIunits` a lot anyway, so it seems to be what people want:
```
$ grep -R "import *SI *= *Modelica" ~/MODELICAPATH/ | wc -l
1137
```

dietmarw, 2019-08-07:
> To conclude, the name SIunits should not be changed.
Noted. 
So what about 2 and 3?

christiankral, 2019-08-07:
> Is it really that much of a change to add a single `.` compared to right now if `Units.SI` was used instead? I see `import SI = Modelica.SIunits` a lot anyway, so it seems to be what people want:
> 
> ```
> $ grep -R "import *SI *= *Modelica" ~/MODELICAPATH/ | wc -l
> 1137
> ```

https://github.com/modelica/ModelicaStandardLibrary/issues/2960#issuecomment-501223526

sjoelund, 2019-08-07:
Indeed. So if it was renamed to `Modelica.Units.SI`, you could do the recommended style `import Modelica.Units.SI` and use `SI.Voltage` like many people seem to prefer.

beutlich, 2019-08-07:
> 
> 
> > Is it really that much of a change to add a single `.` compared to right now if `Units.SI` was used instead? I see `import SI = Modelica.SIunits` a lot anyway, so it seems to be what people want:
> > ```
> > $ grep -R "import *SI *= *Modelica" ~/MODELICAPATH/ | wc -l
> > 1137
> > ```
> 
> [#2960 (comment)](https://github.com/modelica/ModelicaStandardLibrary/issues/2960#issuecomment-501223526)

Which perfectly fits to having `import Modelica.Units.SI`.

MartinOtter, 2019-08-07:
> Decided if an update to ISO 80000-1:2009 is relevant. If not at least update the documentation to state why MSL is still fine with using the deprecated ISO 31 norm.

I quickly inverstigated. There are about 14 chapters of ISO80000 (e.g. Part 4: Mechanics). Every part costs about 120 CHF, so the whole standard > 1000 Euro. I have no access to it. Even if someone has access, one would have to inspect > 500 definitions in Modelica.SIunits and ISO80000 and update correspondingly. This is a lot of work. Could be probably made by paid work (MA backoffice?). In my opinion, it is not worth it. The simplest approach is to change the documentation and just state that Modelica.SIunits is based on ISO80000 (assuming that the parts of ISO80000 that are relevant for Modelica.SIunits did not change, but that just new stuff was added that is missing in Modelica.SIunits).

MartinOtter, 2019-08-07:
> Update the name from SIunits to Units since the current class name structure includes strange constructs (see above #712 (comment) ) which basically means the units are simply a collection of, well units. Some are SI some are not, and some are derived.

Rethinking about various options:

1. Modelica.Units and in the description state that these are SI units and contains conversions to non-SI units.

2. Modelica.Units.SI: Drawbacks: We should not support other unit systems and therefore only one sub-package under Units, which seems to be over-structured. The Modelica package avoids abbreviations in package names and this would be an abbreviated package name (yes, SIunits also contains an abbrevation, which is not nice).

3. Modelica.SI: Would avoid one drawback of 2, but would be not so descriptive.

So to summarize: I changed my mind. For me it is also o.k. to just use Modelica.Units, if in the documentation it is stated that the units are based on SI-units of ISO 80000 (with some conversions to non SI-units).

Since this is a change that influences nearly every model, we could make a polly (Modelica.SIunits, Modelica.Units (but only SI units), Modelica.Units.SI, Modelica.SI).

dietmarw, 2019-08-13:
I'd like to discuss option 2 a bit further as I think that would actually allow us to solve issues like the ambiguity of the current characterisation of SI and nonSI and also of having nonSI units inside the current SIunits. OK let me explain how I think this can be solved rather elegantly and also "fixing" the import issue that came up.

Currently (as of 3.2.3) the strurcture looks as follows:

- Modelica
  - SIunits
    - UsersGuide
    - Icons
    - Conversions
      - NonSIunits

Taking option 2 as a starting point and doing some structual clean up at the same time I propose (let's call it option 2a):

- Modelica
  - Units
    - UsersGuide
    - Icons
    - SI (toplevel content of old  Modelica.SIunits but non of the sub-packages)
    - NonSI (old Modelica.SIunits.Conversions.NonSIunits) 
    - Conversions (same as before but with the NonSIunits removed)

The benefit of 2a is:
  1. We can use the recommended import:
  ```
  import Modelica.Units.SI;
  ```
  2. `Modelica.Units.SI` can continue to strictly follow is ISO standard
  3. Greater freedom of adding useful (and practically used) units under `Modelica.Units.NonSI` though it might be an idea to put those (and the ones inside the existing NonSI) simply under `Modelica.Units.Other`

christiankral, 2019-08-26:
Option 2a makes sense to me and I am in favor of it. I do, however, not understand what 

>`Modelica.Units.SI` can continue to strictly follow is ISO standard

means. As I have also no access to the ISO 80000 standard, I am not sure how I exactly stick to the standard. Or does that mean that we strictly keep only SI units in this package?

dietmarw, 2019-08-26:
That is of course a problem. In practice it probably means unless somebody with access to the ISO standard can document that a unit is SI then it can be added there otherwise it simply is added to `Modelica.Units.Other`. This probably means that the content of `Modelica.Unitis.SI` will stay as it is as per today.

MartinOtter, 2019-09-02:
Option 2a looks good and is fine for me.

However, since this influences practically every existing model, other people of the MAP lib group and the tool vendors should give their opinion (especially regarding issues with automatic conversion).

HansOlsson, 2019-09-03:
> That is of course a problem. In practice it probably means unless somebody with access to the ISO standard can document that a unit is SI then it can be added there otherwise it simply is added to Modelica.Units.Other. This probably means that the content of Modelica.Units.SI will stay as it is as per today.

I understand the general direction, but the concrete details seem doubly problematic:

- The existing Modelica.Units.SI likely contains some units that shouldn't be there (or with wrong name); having some special preference for them seems odd.
- Even if someone has access to the ISO 80000 standard I doubt that they are allowed to just copy every definition of the standard.

I'm also unsure if ISO 80000 really standard really defines the units in such an unambiguous way (the SI-standard itself doesn't really).

TL;DR: To me it doesn't make sense that we base something on something we haven't read.

The SI-standard itself is freely accessible.

dietmarw, 2019-09-03:
> The SI-standard itself is freely accessible.

@HansOlsson So you suggest to check the content of SIUnits against the freely accessible SI-Standard (do you have a link?) or what is it you are suggesting should be done?

tobolar, 2019-09-03:
@dietmarw Do you mean https://www.bipm.org/en/measurement-units/ ?

dietmarw, 2019-09-03:
I'm just trying to find a solution as to what our claim should be. Just the 7 Base SI units can not be the answer.

tobolar, 2019-09-03:
There are also some derived and non-SI units to be found in the [SI broschure](https://www.bipm.org/utils/common/pdf/si-brochure/SI-Brochure-9-EN.pdf). But I understand. We need more.

HansOlsson, 2019-09-04:
> There are also some derived and non-SI units to be found in the SI broschure. But I understand. We need more.

True, one possibility is that we allow other derived units as long as they are derived SI-units (without prefixes). That seems to be what we already do - but it is clearly ad-hoc.

henrikt-ma, 2019-09-04:
Option 2a+"Other" is fine with Wolfram (SystemModeler).

dietmarw, 2019-11-28:
So I started working on the ticket and finished already the structure of `Modelica.Units` itself. When then updating the references in MSL itself it became clear that I **really** need to wait with this until all currently outstanding PRs for 4.0.0 are merged. Otherwise the references need to be updated twice. So I hope we can get the open PRs done by Jan 2020 and then announce a kind of "week(s) of PR silence". 

[image]

dietmarw, 2020-01-09:
Some more thoughts on how I'm going to proceed. Currently we have a mixture of absolute class name use (i.e., `Modelica.SUunits.`) and making use of named imports (i.e., `SI.`). Now thanks to the new structure the later usage will no longer depend on named imports but can use the simple import:

```mo
import Modelica.Units.SI
```

So in order to keep the *changes* due to the restructure of the SIunits package within the MSL to a minimum (least classes being touched) I'll do it in these steps (after `SIunits` has been changed to `Units.`):
1. Rename/update all non-SI references
2. Rename all "SI"-`Modelica.SIunits.` to `SI.`
3. In order that those class references can work I will add **one** top-level import:
```mo
import Modelica.Units.SI
```
and at the same time getting rid of all (now unnecessary) lower level SI imports. 

So for the future when inside the MSL the developers can always be able to use `SI.` types without having to worry about adding another import statement.

HansOlsson, 2020-01-09:
> In order that those class references can work I will add one top-level import:
> 
> import Modelica.Units.SI
> and at the same time getting rid of all (now unnecessary) lower level SI imports.
> So for the future when inside the MSL the developers can always be able to use SI. types without having to worry about adding another import statement.

Three notes:

- Remember that this procedure will not work in encapsulated classes, as the import isn't found.
- Stylistically I'm not too keen on relying on such imports.
- Remember: if you want to copy an example for use outside of MSL use "New>Duplicate class" in Dymola (and something corresponding in other tools); don't just copy the text.

dietmarw, 2020-01-09:
>     * Remember that this procedure will not work in encapsulated classes, as the import isn't found.
Yes I'll look out for those.

>     * Stylistically I'm not too keen on relying on such imports.
I understand but since these kinds are already in place it makes more sense to keep those (i.e., no change) and just update the others to have one consistent way within the MSL.

>     * Remember: if you want to copy an example for use outside of MSL use "New>Duplicate class" in Dymola (and something corresponding in other tools); don't just copy the text.
I know but I think this is something that we can live with.

christiankral, 2020-01-09:
> > * Remember: if you want to copy an example for use outside of MSL use "New>Duplicate class" in Dymola (and something corresponding in other tools); don't just copy the text.
> 
> I know but I think this is something that we can live with.

Alternatively we could use 

```modelica
import Modelica.Units.SI
```

in each model where SI units are used. I know this implementation requires a lot more changes, but it is then more obvious to new Modelica users, what, e.g.,  `SI.Voltage` means. It also solves the problem of the missing `import Modelica.Units.SI` when duplicating a model.

dietmarw, 2020-01-09:
That would be the exact opposite than what I would consider a clean implementation. Having unnecessary code duplication goes against the whole idea of Modelica with its object oriented features. Taking care of the class rename is basically a tool issue.

christiankral, 2020-01-09:
The problem is that there are currently no tool features (recommended by the Modelica Specification) which allow the user to understand what imports are used in a specific model. This is the reason why I somehow dislike package specific imports, which I can never understand without knowing where the import is implemented in the code. 

However, I yet agree with your point of view, that it makes sense to use object oriented features and to avoid redundant code. So using one import in the top package makes sense to me -- even though it will the be difficult to new Modelica users to understand the code.

dietmarw, 2020-01-10:
I perfectly get your point but need to decide to either: 
 1. make the change of the unit structure as least invasive (least changes) as possible. This would mean leave all `SI.` type references as they are and only change the current `SIunits` references. 
 2. use absolute class references everywhere. This would basically change almost every single model. 

What I don't want is a mixture of both inside the MSL.

dietmarw, 2020-01-14:
I discussed this with @beutlich and will wait to create the PR until after feature freeze. Since this is not a new feature but a reorganisation and would otherwise disturb the current surge in feature PRs.

tobolar, 2020-01-31:
Regarding comment https://github.com/modelica/ModelicaStandardLibrary/issues/712#issuecomment-519014269
> We should not support other unit systems

`Temperature_degF` actually belongs to Imperial / US customary measurement systems. So we already do mix different unit systems.

(Siehe also comment on "Celsius temperature" in https://github.com/modelica/ModelicaStandardLibrary/issues/712#issuecomment-465222355.)

PR #3379 (2020-02-01): Reorganise SIunits into Units package
This fixes #712 by reorganising the SIunits package into a Units package

I checked the following libraries (Dymola 2020, non-pedantic mode, see #3378 why) :
- Complex.mo
- Modelica
- ModelicaReference
- ModelicaServices
- ModelicaTest
- ModelicaTestConversion4.mo
- ModelicaTestOverdetermined.mo
- ObsoleteModelica4.mo
closed 2020-02-01 by beutlich