modelica/ModelicaStandardLibrary #733: Rename `SynchronousInductionMachine` to `SynchronousMachines`

modelica-trac-importer, 2017-01-14:
**Reported by dietmarw on 30 Apr 2012 13:25 UTC**
There seems to be a major flaw with the naming convention. To my knowledge an electrical machine can be either of induction type **or** of synchronous type but not both at the same time. Therefore calling a machine "Synchronous Induction Machine" doesn't make sense and the naming convention should be changed as soon as a conversion script allows this.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/733

modelica-trac-importer, 2017-01-14:
**Comment by ahaumer on 1 May 2012 19:16 UTC**
Dear Dietmar,
this is a general naming issue, already discussed with Christian Kral; we tried to use:
"induction machine"                = German "Drehfeldmaschine"
"asynchronous (induction) machine" = German "Asynchronmaschine
"synchronous (induction) machine"  = German "Synchronmaschine"
"induction" means "voltage generation" German "Induktion" which holds for both machine types.
We're open to your suggestions ... (especially for an English term for German "Drehfeld" since "rotating (magnetic) field" sounds clumsy, but we have to observe backwards compatibility of naming until a conversion script is avaliable.
Regards, Toni

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 2 May 2012 06:49 UTC**
Toni, the problem is that in technical English inducton motor/machine is identical to asynchronous motors/machines. So in English if you use induction machine a person automatically assumes you are talking about an asynchronous machine and **not** about an synchronous one. Calling a machine Synchronous Induction Machine is simply confusing if not even plainly wrong. 

"Drehfeld" would translate as "Rotating magnetic field" (http://en.wikipedia.org/wiki/Rotating_magnetic_field) which is most likely a bit too cumbersome to use as a base class name I guess ;-).  Please not that also in that WP article they distinguish between synchronous and induction machines.

So renaming would make the structure of `Electrical.Machines` and `FundamentalWave` cleaner and more logical. Especially since you kind of applied the logic in the abbreviations you use. I.e., `AIM_...` and `SM_...` but not `SIM_...`. 

I know that the fix to this will have to wait for the next non-backward compatible version with conversion script. That's why I scheduled it for now for [[milestone:MSL4.0]].

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 10 Nov 2013 17:33 UTC**
Based on the discussion from comment 3 of #1348 how about a "induction"-less structure which also takes care of the new quasistatic components:

```
Modelica.Electrical.Machines.
  Transient
     AsynchronousMachines
     SynchronousMachines
     DCMachines
     Transformers
     Components
  QuasiStatic
     DCMachines
     (any future quasistatic sublibs)
```

Just an idea. 

A complete other approach would be rewriting the machine models in such a way that their behaviour, i.e., transient or quasistatic, can be made dependent on a parameter. So we would only have one DC  Machine model which can be made transient or quasistatic. But this should probably handled as a different enhancement in a separate ticket in case it is something you think is feasible.

modelica-trac-importer, 2017-01-14:
**Changelog removed by dietmarw on 10 Nov 2013 17:33 UTC**

modelica-trac-importer, 2017-01-14:
**Comment by dietmarw on 1 Aug 2015 21:39 UTC**
Milestone renamed

AHaumer, 2018-05-21:
Yes we should do the rename to the common English term, but also "AsynchronousInductionMachines" to "InductonMachines" whenever a conversion script is applicable.

dietmarw, 2018-05-21:
I would recommend to use 
`AsynchronousMachines` and `SynchronousMachines`  just like originally suggested. It simply reflects the physical nature best.

AHaumer, 2018-05-25:
No English native speaker understands "Asynchronous Machine", the use the term "Induction Machine".

dietmarw, 2018-05-25:
That's actually true. So `InductionMachine` it shall be then :-)

dietmarw, 2019-05-13:
Just to make sure we are on the same page, for the MSL4.0.0 conversion the structure of the machines should be(as discussed earlier):
```
Modelica.Electrical.Machines.
  Transient
     InductionMachines
     SynchronousMachines
     DCMachines
     Transformers
     Components
  QuasiStatic
     DCMachines
     (any future quasistatic sublibs)
```

But it turns out in the meantime we now also got now `Modelica.Electrical.QuasiStatic`. So the structure should probably be 
```
Modelica.Electrical.
  Machines.
    InductionMachines
    SynchronousMachines
    DCMachines
    Transformers
    Components
  QuasiStatic.
    Machines.
      DCMachines
      (any future quasistatic sublibs)
```

Looks a bit strange tough. Any suggestions to make the structure cleaner and more systematic @christiankral @AHaumer?

AHaumer, 2019-05-14:
There are no plans for other QuasiStatic Machines besides the exisiting DCMachines. 
QuasiStatic ACMachines are covered by Magnetic.QuasiStatic.FundamentalWave, that's sufficient.

PR #2993 (2019-07-13): Consistent spelling of electrical and magnetic terms
This is getting massive.

* polyphase
* single-phase
* three-phase
* quasi-static
* induction machine
* synchronous machine
* electromagnetic

close #733, close #2973
closed 2019-07-13 by beutlich